raspiblitz/raspiblitz #367: Display Update Notice To User

rootzoll, 2019-02-27:
Tho Nodl has a clever way to get some basic information if a update is available - without the node having to call home.

https://medium.com/@ketominer/how-the-nodl-polls-for-updates-without-us-knowing-and-other-privacy-concerns-ffa7871c8797

For the beginning (v1.1) it would make sense to have just a extra optione in the main menu "Check for Updates" that makes that check and shows some instructions just in case .. maybe even offering to prepare for update by making some extra backup data and turniung the device off. In the future also on the LCD should be some notice about an available update (maybe just turn the color of the version red or something like that).

nyxnor, 2021-04-16:
https://gitlab.lightning-solutions.eu/nodl/nodl-docs/-/blob/master/docs/guides/upgrade-nodl.md
https://gitlab.lightning-solutions.eu/nodl-private/nodl-admin-private/-/blob/nodl-dojo/bin/nodl-installer.php

They dont do through Tor. Ketominer explained some users with slow internet fail to update with Tor. They have other privacy measure that to not be biased, here is a compilation of the responses till this moment. Will update if new information arrives.

>**Ketominer** (inside parentesis are my nyxnor comments)
use the nodl-dojo branch it’s the most up to date
it’s not currently (not currently using Tor)
it’s just doing a git clone from gitlab.lightning-solutions.eu directly
this part happens here: https://gitlab.lightning-solutions.eu/nodl-private/nodl-admin-private/-/blob/nodl-dojo/share/update-nodl.sh
I want to switch to a torrent based upgrade later
**Me**
the band-aid is torsocks. A way is to just display version number on a page of a onion address, so 1 time per day it is queried from the onion address if the version you are using matches the displayed there.
**Ketominer**
for the version number it’s not needed as dns is not reaching our servers. For the git clone itself, torsocks would be too unreliable (I dont think so, I torsocks git almost everytime)
we already have users failing the clone just because they have slow internet
for pulling the version number, the dns mecanism is really very good - it’s signed with dnssec and totally distributed and stateless
I even pull the release notes this way now :)
if you run your local DoT resolver it’s even better

nyxnor, 2021-05-16:
https://github(dot)com/bitcoin/bitcoin/issues/2626 - Change (dot) to (.)
Update checker - Bitcoin core