raspiblitz/raspiblitz #368: Powered USB hub to avoid HDD data corruption

openoms, 2019-02-27:
As a part of the work with other SBC-s I did some research regarding how the Pi is capable of powering a HDD requiring 1Amp through an USB 2 port. Good comment here https://dietpi.com/phpbb/viewtopic.php?f=9&t=5115&start=10#p15042 : 

> SSDs with trashed performance is quite common when powered by SBC, the reason is usually undervoltage and not undercurrent (on almost all SBC USB ports there are current limiters in action exceeding the commonly known 500mA for USB2 or 900mA for USB3 -- ROCK64 allows 650mA on each USB2 port and 950 mA on the USB3 port but a RPi 3 uses 1.2A for all 4 USB2 ports combined or a NanoPi M4 for example has one global 2A current limiter for all 4 USB3 ports -- you always need to study schematics).
> 
> But as already said: usually it's undervoltage (cable/contact resistance between PSU and board and again between board and disk) causing the problems and not limited current (same on the RPi 3 where 1.2A are set by default -- compare with 'vcgencmd get_config int' output -- but due to Polyfuses and Micro USB powering the voltage available to USB peripherals often drops below 4.5V and then majority of external SSDs get in trouble, the majority of 2.5" HDD already has trouble with less than 4.75V)

See this one with the schematics explaining how the Pi is able to power up to 1.1Amp through the USB: https://raspberrypi.stackexchange.com/questions/340/how-much-power-can-be-provided-through-usb

I still don`t think it is reliable enough and might be a reason behind many data corruption issues. 
I propose we should use a powered USB hub to power the HDD and possibly the rPi as well. 
I am ordering this: https://www.amazon.co.uk/SuperSpeed-Individual-Switches-Charging-Splitter-USB-3-0-hub-15w-AC-adapter/dp/B07B2HJG8P/ref=sr_1_1?ie=UTF8&qid=1551263091&sr=8-1&keywords=powered+usb+hub Come with 3 Amp PSU and outpust up to 2.4A through the USB-s so should be able to power both the HDD and the Pi.

I found that no other SBC has this hack to output 1.1A through the USB, so a powered hub will be necessary.

rootzoll, 2019-02-27:
Thanks for the deep reserach on this ... a powered USB Hub could be one solution - let me know your experience.

On other SBCs (that dont share power over all USB ports) a y-cable (using two usb ports for power) like this one could be working, correct? https://www.amazon.de/Goobay-95746-Power-SuperSpeed-Schwarz/dp/B00DON261G

openoms, 2019-02-27:
In theory the HDD coulbed powered from 2 usb ports, but most of these low power devices are not designed to give even 500ma through USB. They are only supposed to power simple usb drives at ~100 mA. Also it is quite difficult to test for HDD undervoltages which are not causing immediate issues.
I think separating the power supplies of the HDD and the SBC does get rid of a big risk factor. The less number of connections and switches the current goes through the more stable it will be.

So far I was using this expansion board which is great for the Pi and the same form factor alternatives (like the Rock64) https://www.amazon.co.uk/gp/aw/d/B07LFZ5HZQ/
It is a USB to SATA adapter and built in 2.5 A USB power output. Proved to be very stable for >4 months.

rootzoll, 2019-02-27:
Yeah I think seperate power supplies are a way to go. An extension board is a way .. but to keep it cheap and simple it should be OK then to advertise a y-cable that is not using a second usb port of the SBC but power from a seconde power adapter.

openoms, 2019-02-27:
Agree and thanks for pointing it out that a Y cable is needed to get extra power from the second source let it be a powered hub or battery.
Here is one for the Uk list: https://www.amazon.co.uk/AKORD-Type-Male-Micro-Cable/dp/B00B5T3UTS/

Agree not using the expansion board as it is strongly associated to the Nodl now. I had it before they announced their product, but showing it around always resulted to questions if it is the Nodl :).

openoms, 2019-02-27:
I also use one of these: https://www.aliexpress.com/item/USB-3-0-to-SATA-Power-Adapter-Converter-Cable-USB3-0-Cable-Converter-for-Samsung-Seagate/32913703122.html
It has a great feature that it takes a standard barrel connector and can be powered with either 5V or 12V to use a 2.5" or 3.5" HDD.

rootzoll, 2019-02-27:
We are RaspiBlitz ... we go for the cheap solutions first - thats our signature move ;)

rootzoll, 2019-02-27:
TODO for v1.1 realease: Add info about an optional y-cable with second power source to README and to the shopping lists.

fluidvoice, 2019-03-30:
> Also it is quite difficult to test for HDD undervoltages which are not causing immediate issues.

see this about the `vcgencmd` command: https://github.com/rootzoll/raspiblitz/issues/474#issue-427212699

openoms, 2019-06-03:
> I am ordering this: https://www.amazon.co.uk/SuperSpeed-Individual-Switches-Charging-Splitter-USB-3-0-hub-15w-AC-adapter/dp/B07B2HJG8P/ref=sr_1_1?ie=UTF8&qid=1551263091&sr=8-1&keywords=powered+usb+hub Comes with a 3 Amp PSU and outputs up to 2.4A through the USB-s so should be able to power both the HDD and the Pi.
> 
> I found that no other SBC has this hack to output 1.1A through the USB, so a powered hub will be necessary.

Comes very handy for the AtomicPi with it`s single USB3: https://twitter.com/openoms/status/1135489673244160001

openoms, 2021-09-15:
Closing as the research was done for the RPI3 and HDDs (none of which is recommended now). Also the links are not available any more.

closed 2021-09-15 by openoms