raspiblitz/raspiblitz #369: LND Won't stay up!

OfficialBigAl, 2019-02-27:
Can anyone shed some light on why LND keeps failing?

I was in the process of trying to recover the LN state by opening connecting to peers and opening channels with new mBTC funds and one failed .  Since then LND won't stay connected, only occasionally do I see the the wallet having a positive value and then it drops. :(
Have tried recovery_window = 25000 and even greater but no joy.

Feb 27 19:27:17 BigAlsLNnode2 lnd[537]: 2019-02-27 19:27:17.237 [DBG] CNCT: ChannelArbitrator(c6d3a3d11ec92a0414aaea4cdd7ad67760ffe459eed7d9b2c4968b2f49ea9c67:1): checking chain actions at height=564915
Feb 27 19:27:17 BigAlsLNnode2 lnd[537]: 2019-02-27 19:27:17.237 [INF] DISC: Authenticated Gossiper is starting
Feb 27 19:27:17 BigAlsLNnode2 lnd[537]: 2019-02-27 19:27:17.237 [INF] BRAR: Starting contract observer, watching for breaches.
Feb 27 19:27:17 BigAlsLNnode2 lnd[537]: 2019-02-27 19:27:17.237 [INF] NTFN: New block epoch subscription
Feb 27 19:27:17 BigAlsLNnode2 lnd[537]: 2019-02-27 19:27:17.242 [INF] CRTR: FilteredChainView starting
Feb 27 19:27:17 BigAlsLNnode2 lnd[537]: 2019-02-27 19:27:17.245 [ERR] DISC: Unable to rebroadcast stale channels: unable to retrieve outgoing channels: Channel from self node has no policy
Feb 27 19:27:18 BigAlsLNnode2 lnd[537]: 2019-02-27 19:27:18.330 [INF] CRTR: Filtering chain using 22321 channels active
Feb 27 19:27:18 BigAlsLNnode2 lnd[537]: 2019-02-27 19:27:18.339 [INF] CRTR: Prune tip for Channel Graph: height=564915, hash=00000000000000000028dd360b6fa0aa840b7805398d3e28534cc0af7005d803
Feb 27 19:27:18 BigAlsLNnode2 lnd[537]: 2019-02-27 19:27:18.559 [DBG] LNWL: Marked address bc1qtm9rkcrn2pq9vt2lptysgsjq46vylrahe82ak0 used
Feb 27 19:27:18 BigAlsLNnode2 lnd[537]: 2019-02-27 19:27:18.602 [INF] NTFN: New block epoch subscription
Feb 27 19:27:18 BigAlsLNnode2 lnd[537]: 2019-02-27 19:27:18.602 [INF] NTFN: New confirmation subscription: txid=c6d3a3d11ec92a0414aaea4cdd7ad67760ffe459eed7d9b2c4968b2f49ea9c67, numconfs=3
Feb 27 19:27:18 BigAlsLNnode2 lnd[537]: 2019-02-27 19:27:18.603 [DBG] NTFN: Using height hint 564629 retrieved from cache for c6d3a3d11ec92a0414aaea4cdd7ad67760ffe459eed7d9b2c4968b2f49ea9c67
Feb 27 19:27:18 BigAlsLNnode2 lnd[537]: 2019-02-27 19:27:18.603 [DBG] NTFN: Dispatching historical rescan for txid=c6d3a3d11ec92a0414aaea4cdd7ad67760ffe459eed7d9b2c4968b2f49ea9c67
Feb 27 19:27:18 BigAlsLNnode2 lnd[537]: 2019-02-27 19:27:18.603 [INF] FNDG: Waiting for funding tx (c6d3a3d11ec92a0414aaea4cdd7ad67760ffe459eed7d9b2c4968b2f49ea9c67) to reach 3 confirmations
Feb 27 19:27:18 BigAlsLNnode2 lnd[537]: 2019-02-27 19:27:18.664 [INF] CMGR: Server listening on [::]:9735
Feb 27 19:27:18 BigAlsLNnode2 lnd[537]: 2019-02-27 19:27:18.731 [INF] LTND: Shutdown complete
Feb 27 19:27:18 BigAlsLNnode2 lnd[537]: panic: runtime error: invalid memory address or nil pointer dereference
Feb 27 19:27:18 BigAlsLNnode2 systemd[1]: lnd.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
Feb 27 19:27:18 BigAlsLNnode2 systemd[1]: lnd.service: Unit entered failed state.
Feb 27 19:27:18 BigAlsLNnode2 systemd[1]: lnd.service: Failed with result 'exit-code'.

OfficialBigAl, 2019-03-01:
Asking for a friend: https://github.com/lightningnetwork/lnd/issues/2722

OfficialBigAl, 2019-03-03:
It's a bit drastic I know but ...backed up the PI to second HDD.  
Shutdown and flashed a new image of V1.0 and rebuilt.  
Script worked a charm.  
All up and running fine.  
Even after reindexing found funded channels!
and able to close out ;)

Whilst not a resolve to the initial problem, proves that V1.0 is working (from me at least).

Cheers.

rootzoll, 2019-03-06:
I hope for the v1.1 of RaspiBlitz also a new release of LND is available. So my take away is here:

TODO: Check for v1.1 for new LND release.

kilrau, 2019-04-11:
It'll be. Already 0.6 rc3.

fluidvoice, 2019-04-11:
> It's a bit drastic I know but ...backed up the PI to second HDD.
> Shutdown and flashed a new image of V1.0 and rebuilt.
> Script worked a charm.
> All up and running fine.
> Even after reindexing found [...]

It's a know LND problem that is hopefully fixed in their next release (prob also in the next v1.2 Raspiblitz) so it is possible it will happen again, but hopefully not.

rootzoll, 2019-04-15:
TODO: Waiting for final LND 0.6 release
TODO: Final testing

rootzoll, 2019-04-16:
OK final LND 0.6 release is out:
https://github.com/lightningnetwork/lnd/releases/tag/v0.6-beta

rootzoll, 2019-04-17:
Looking good. LND v0.6-beta will be part of RaspiBlitz v1.2. Closing Issue.

closed 2019-04-17 by rootzoll