raspiblitz/raspiblitz #371: Replace PublicIP Discovery

rootzoll, 2019-02-28:
This might be a better way to determine the pubicIP:
https://www.cyberciti.biz/faq/how-to-find-my-public-ip-address-from-command-line-on-a-linux/

The following seems to run on RaspiBlitz quite well:
`host myip.opendns.com resolver1.opendns.com`

drunkenclam, 2019-03-02:
FYI this doesn't work here, output is:

Using domain server:
Name: resolver1.opendns.com
Address: 2620:119:35::35#53
Aliases:

Host myip.opendns.com not found: 3(NXDOMAIN)

(Perhaps it's because I use pihole for DNS, idk)

rootzoll, 2019-03-13:
TODO v1.1: sanatize return value from third party service and then replace in furture release

rootzoll, 2020-04-01:
Marked this idea for the Lightning HackSprint - details see: https://wiki.fulmo.org/index.php?title=Lightning_HackSprint

The tasks would be:
- Research what better (decentral, privacy preserving) ways there are to determine the public IP when the RaspiBlitz is behind a NAT.
- Implement and make a PR

frennkie, 2020-04-03:
being lazy (and not doing research) isn't this something that could be determined e.g. over the Bitcoin P2P network ..?

frennkie, 2020-04-03:
Something I have worked with in the past is using UPNP, NAT-PMP or PCP. This only talks to your local router/gateway (very good privacy characteristics). But there are many different implications out there.. so this might work or not ..  But could be a primary attempt before falling back to something else (e.g. doing outbound connections beyond your local network).

frennkie, 2020-04-04:
@rootzoll @openoms @Emzy can you guys try this on a RaspiBlitz which is running behind a NAT router:

```
sudo apt-get install miniupnpc
upnpc -s
```

For me it gives me:

```
$ upnpc -s
upnpc : miniupnpc library test client, version 2.1.
 (c) 2005-2018 Thomas Bernard.
Go to http://miniupnp.free.fr/ or https://miniupnp.tuxfamily.org/
for more information.
List of UPNP devices found on the network :
 desc: http://192.168.178.1:49000/igddesc.xml
 st: urn:schemas-upnp-org:device:InternetGatewayDevice:1

Found valid IGD : http://192.168.178.1:49000/igdupnp/control/WANIPConn1
Local LAN ip address : 192.168.178.48
Connection Type : IP_Routed
Status : Connected, uptime=61316s, LastConnectionError : ERROR_NONE
  Time started : Sat Apr  4 02:21:51 2020
MaxBitRateDown : 259361000 bps (259.3 Mbps)   MaxBitRateUp 41485000 bps (41.4 Mbps)
ExternalIPAddress = 83.XXX.YYY.13      <- redacted :-)
Bytes:   Sent: 2992632077       Recv: 2744679930
Packets: Sent: 17348198 Recv: 20744619
```

I have a AVM Fritz box and there is a setting fir UPNP (*"Statusinformationen über UPnP übertragen"*)  that is enabled (not sure what the default for this)

frennkie, 2020-04-04:
It would be interessting to hear from someone who has Apple (AirPort [Extreme]) device as a router.

openoms, 2020-04-05:
I am using a Netgear router flashed with DD-WRT (UPnP is usually off - probably by default too - but activated to test).
External IP was showing correctly (redacted).
```
admin@raspberrypi:~ $ upnpc -s
upnpc : miniupnpc library test client, version 2.1.
 (c) 2005-2018 Thomas Bernard.
Go to http://miniupnp.free.fr/ or https://miniupnp.tuxfamily.org/
for more information.
List of UPNP devices found on the network :
 desc: http://192.168.3.1:1780/InternetGatewayDevice.xml
 st: urn:schemas-upnp-org:device:InternetGatewayDevice:1

Found valid IGD : http://192.168.3.1:1780/control?WANIPConnection
Local LAN ip address : 192.168.3.236
Connection Type : IP_Routed
Status : Connected, uptime=504708s, LastConnectionError : ERROR_NONE
  Time started : Mon Mar 30 21:22:00 2020
MaxBitRateDown : 1000000000 bps (1000.0 Mbps)   MaxBitRateUp 1000000000 bps (1000.0 Mbps)
ExternalIPAddress = XX.XX.XXX.XX 
Bytes:   Sent: 1749450911	Recv: 549709003
Packets: Sent: 120393565	Recv: 120403965
```

rootzoll, 2020-04-14:
Asked on Twitter about someone with a Apple Airport router.
https://twitter.com/rootzoll/status/1250103888251883521?s=20

Marking this issue to be part of v1.6 release.

derPUPE, 2020-04-20:
> Asked on Twitter about someone with a Apple Airport router.
> https://twitter.com/rootzoll/status/1250103888251883521?s=20
> 
> Marking this issue to be part of v1.6 release.

raspiblitz 1.5rc1 connectet via cable to Apple AirPort Express - 1st Generatoin 
The AirPort Express ist WLAN connected to an AVM Routing Fritzbox:

command:

$ upnpc -s

result:

upnpc : miniupnpc library test client, version 2.1.
 (c) 2005-2018 Thomas Bernard.
Go to http://miniupnp.free.fr/ or https://miniupnp.tuxfamily.org/
for more information.
List of UPNP devices found on the network :
 desc: http://192.168.0.90:80/description.xml
 st: urn:schemas-upnp-org:device:basic:1

 desc: http://192.168.0.90:80/description.xml
 st: uuid:2f402f80-da50-11e1-9b23-00178820e78e

 desc: http://192.168.0.90:80/description.xml
 st: upnp:rootdevice

 desc: http://192.168.0.1:49000/igddesc.xml
 st: urn:schemas-upnp-org:device:InternetGatewayDevice:1

Found valid IGD : http://192.168.0.1:49000/igdupnp/control/WANIPConn1
Local LAN ip address : 192.168.0.62
Connection Type : IP_Routed
Status : Connected, uptime=70820s, LastConnectionError : ERROR_NONE
  Time started : Mon Apr 20 01:48:21 2020
MaxBitRateDown : 109307000 bps (109.3 Mbps)   MaxBitRateUp 36145000 bps (36.1 Mbps)
ExternalIPAddress = x.x.x.x.  <= was the correct IP
Bytes:   Sent: 2222759253	Recv: 3866455363
Packets: Sent:  4061897	Recv:  7378553

rootzoll, 2020-04-20:
@derPUPE much thx  👍 

@frennkie so we have a fritzbox, a DD-WRT & a Airport Express ... do you need more?

frennkie, 2020-05-09:
This is good news. So basically it should be pretty straight forward to implement a scripts that first tries  `upnpc` and only if this doesn't work check (one of the) online services (if the user activated this).

rootzoll, 2020-05-09:
I bought a "Apple Airport Extreme Base Station" from eBay - its a 2nd Generation A1143 Model from 2007. see https://en.wikipedia.org/wiki/AirPort_Extreme#Comparison_chart

When I run a RaspiBlitz behind that and run `upnpc -s` I get:
```
upnpc : miniupnpc library test client, version 2.1.
 (c) 2005-2018 Thomas Bernard.
Go to http://miniupnp.free.fr/ or https://miniupnp.tuxfamily.org/
for more information.
No IGD UPnP Device found on the network !
```

Note: Not sure if it would make a difference - but the Airport Extreme is a Router behind a Router and not connected to the ISP directly.

frennkie, 2020-05-09:
In theory they should (or at least could) support: NAT-PMP (NAT Port Mapping Protocol). Anything on the GUI regarding that?

rootzoll, 2020-05-09:
There is a check mark for "NAT-Port-Mapping-Protokol" it was already on ... but it looks like that in a "Double NAT" situation it will not respond.

frennkie, 2020-05-09:
I think that `miniupnpc` will not work with NAT PMP. There is a separate implementation for it: http://miniupnp.free.fr/libnatpmp.html

rootzoll, 2020-05-09:
How can I test this on my RaspiBlitz with the Airport Extreme?

frennkie, 2020-05-09:
Well.. I would try

```
apt-get install natpmpc
natpmpc -h
natpmpc -l (?!)
```

And see what it does. Obviously you would need to be connected to the LAN side of the Router.

rootzoll, 2020-05-09:
OK it was `natpmpc`- just without any further options - giving me this:
```
initnatpmp() returned 0 (SUCCESS)
using gateway : 10.0.1.1
sendpublicaddressrequest returned 2 (SUCCESS)
readnatpmpresponseorretry returned 0 (OK)
Public IP address : 192.168.178.121
epoch = 1263
closenatpmp() returned 0 (SUCCESS)
```

So it gave me the IP that was givin to it by the outer-router as the public IP. But If the AirPort Extreme would be my main ISP router we should see the publicIP there.

rootzoll, 2020-05-09:
So it looks to me that the strategy here would be ...

- to first try: `upnpc -s`
- if no result or a local IP comes back the fallback to `natpmpc`
- and if still no result or a just a local IP comes back then fallback to a central service

drunkenclam, 2020-05-09:
For everyone else reading this: Just use tor. It comes with raspiblitz, so it's easy to enable. It's a double win: Much much more privacy than clearnet, and you don't have to deal with this upnp crap that's working different on every router...

rootzoll, 2020-05-09:
Yes going to TOR by default is a path for RaspiBlitz .. but switching to clearIP will still be an option in the future and for this option the goal is to use local publicIP discovery instead of relying on some central workaround.

rootzoll, 2020-05-27:
With the aim of v1.7 to switch to TOR by default, the publicIP discovery should still be available for people defaulting to IP mode.

NilByte, 2020-12-13:
I like the idea on using STUN mentioned in this thread.

https://unix.stackexchange.com/questions/22615/how-can-i-get-my-external-ip-address-in-a-shell-script#