raspiblitz/raspiblitz #374: Dojo software by Samourai to be installed on the RaspiBlitz and paired with Samourai wallet

openoms, 2019-03-01:
Looking forward to this one a lot. Tor + Samourai + RaspiBlitz = 🖤 https://twitter.com/SamouraiWallet/status/1101372204686680064?s=19

openoms, 2019-06-10:
This is now imminent: https://twitter.com/SamouraiWallet/status/1135120352521179136?s=19

Got a couple helper scripts to look into. Would like to achieve a seamless integration.

https://gist.github.com/05nelsonm/d6cfaa8d706e4078250c66b5ae8ace1d
https://gist.github.com/05nelsonm/96ad8da7724de3a66f44d6c2d70b361c
https://gist.github.com/Engelberg/aac9da2f27b723683982d187bd4d5ea4

rootzoll, 2019-06-11:
How is the performace on RaspberryPi .. or is this more on extra for the port to more powerfull SBCs?

openoms, 2019-06-12:
What I could gather so far from the DOJO:
* it is a docker based system.
* can be memory hungry during the initial indexing phase when it builds a MySQL database pulling data from RPC. Recommended mem >4GB.
* Neither the publicly available Samourai Wallet nor the Sentinel app is able to connect so far.

Will be very interesting, especially when the mobile mixing will be in play.

bitcoinheiro, 2019-06-24:
Raspi 4 will make possible to run this smoothly... Nice.

openoms, 2019-06-25:
How to install using an existing full node. Likely the way to go. Due to test.
https://github.com/Samourai-Wallet/samourai-dojo/blob/develop/doc/DOCKER_advanced_setups.md#external-bitcoin-full-node

xavierfiechter, 2019-07-13:
https://twitter.com/SamouraiWallet/status/1150148015660904455

openoms, 2019-08-16:
Activated txindex=1 to play with Dojo. Synchronized in the background in ~7h on my RPi4 4GB+SSD. LND and other services ran without a problem during indexing.

grnqrtr, 2019-10-25:
> Looking forward to this one a lot. Tor + Samourai + RaspiBlitz

I'm looking forward to this too. I'm sure you are busy @openoms, but I'd be happy to test any scripts or progress on this.

At one point I was trying to install Dojo on my RaspiBlitz following [this guide](https://bitcoin-on-raspberry-pi-4.gitbook.io/workspace/) (I think), but ran into some problems that I couldn't figure out so gave up.

openoms, 2019-10-26:
I am more and more inclined to look into Joinmarket. It has no Javascript or .NET and has a graphical interface too. Worth a bit of work to set up: https://github.com/JoinMarket-Org/joinmarket-clientserver/blob/master/docs/JOINMARKET-QT-GUIDE.md

grnqrtr, 2019-10-26:
That would be cool too. Can you remotely connect to use the GUI though, or the raspberry pi itself needs to be setup with a display/GUI?

openoms, 2019-10-26:
As it seems Joinmarket can use the bitcoind of the Blitz remotely through RPC, but the GUI would need to run on a desktop.
```
[BLOCKCHAIN] rpc_user = yourusername-as-in-bitcoin.conf rpc_password = yourpassword-as-in-bitcoin.conf rpc_host = localhost #default usually correct rpc_port = 8332 # default for mainnet
```

MrSlumber, 2020-03-11:
Yes, can we please enjoy the freedom/privacy of Dojo/Whirlpool in the next Raspiblitz release? :-) 

Thank you very much in advance! :-) (just read that this just became a part of "mynode")

bitcoinheiro, 2020-03-11:
> Thank you very much in advance! :-) (just read that this just became a part of "mynode")

Yeah, been running it just for that feature and their implementation kinda sucks. Not working - a lot of support requests. Nothing like my raspiblitz... This repo rocks

openoms, 2020-03-11:
FYI the The Ronin Dojo project is working closest to the Samourai team so would recommend to run their implementation as a dedicated node for Dojo: 
https://code.samourai.io/ronindojo/RoninDojo/-/wikis/home

Similarly as with BTCPayServer bundling together more and more big services comes with tradeoffs especially regarding support.
It is the best to run the package maintained by the team and benefit from quickest, streamlined updates and dedicated support.

fulmofulmo, 2020-04-14:
Requests about adding [Whirlpool](https://github.com/Samourai-Wallet/Whirlpool) to the RaspiBlitz keep popping up every now and then. [[1]](https://twitter.com/SamouraiWallet/status/1249472292205334529) [[2]](https://twitter.com/InPHARMaticist/status/1249458343053103104) [[3]](https://twitter.com/zuko_tfl/status/1240044145000869888)

I think it's worth taking another look at the next #LightningHacksprint and label this issue as a hackathon challenge.

nyxnor, 2021-05-03:
[Nodl ](https://gitlab.lightning-solutions.eu/nodl-private/nodl-admin-private/-/tree/nodl-dojo)seems to have the most stable implementation of Dojo without docker.
Other node packages also host their [dojo branches separately](https://github.com/louneskmt/umbrel-samourai-dojo/tree/umbrel/docker/my-dojo) (Umbrel is not good at it for now, too much complaints in the Dojo telegram group, but keep an eye as it is GPL license), nodl host in their main branch as it has become intrinsic.

I still prefer JM over Whirlpool, so I tend to focus more in the dojo approach to serve a reusable payment code.

As far as I can see, this need to be done by someone that can keep an eye on [Ronin Dojo code](https://code.samourai.io/dojo/samourai-dojo) and solve issues when updating it on Blitz. I am not prepared to deep dive into this and commit entirely, but won't hurt to try. I will keep updating how the journey is going if there is something useful to say, or if someone wants to help or do it alone, also good.