raspiblitz/raspiblitz #375: 2nd node setup

Lit3block, 2019-03-01:
Hi guys,
I already have one RaspiBlitz and I want to run another one. how can I do? What ports should I forward? And the most important; should I change bitcoin.conf and lnd.conf right?

What other ports and what should I add to ufw settings?

Excuse me for the many questions, but I don't know where I should start

rootzoll, 2019-03-01:
Running twi puplic nodes in your network would require to set a different LND port on your RaspiBlitz and forward to that ... see this issue for details: https://github.com/rootzoll/raspiblitz/issues/100#issuecomment-465997126

Lit3block, 2019-03-01:
Thank you very much. I couldn't find the thread! I Close this issue :)

closed 2019-03-01 by Lit3block