raspiblitz/raspiblitz #377: After reboot will not connect to WIFI

gehrlein, 2019-03-02:
When running the Raspiblitz over WIFI it displays sometime after reboot something like "Can't connect to. Waiting for network".

After a simple reboot by unplugging the energy it works.

Cannot replicate the exact issue yet and also do not have the specific error message. Will update once I have more info. Since a simple restart by unplugging solves the issue, it is not so problematic, but still needs a fix in the future.

rootzoll, 2019-03-06:
This screen happen when the RaspiBlitz cannot determine its local ip with the following command:
`localip=$(ip addr | grep 'state UP' -A2 | tail -n1 | awk '{print $2}' | cut -f1 -d'/')`
But its a waiting loop and and will retry every x seconds. So normally it should continue as soon it gets a localIP. If you run again into that state, please try to SSH into the RaspiBlitz, CTRL+C to the terminal and post the output of the following two commands to this issue so that I can check if maybe that method is not flexible enough:

`ip addr | grep 'state UP' -A2 | tail -n1 | awk '{print $2}' | cut -f1 -d'/'`
`ip addr`

gehrlein, 2019-03-08:
Thanks for your answer, the issue is that I run the RaspiBlitz solely over WIFI and therefore I have no option to connect via ssh then. I think I also waited a decent time and it did not work.
Once I got the issue again I can check more things (like if the device is listed under my router) and maybe wait some more.

philbertw4, 2020-07-04:
For what it's worth, I have run into this as well multiple times, including just today. Probably anyone who runs their blitz over wifi will run into this if they forget to put a wpa_supplicant.conf file. I tried plugging in keyboard and pressed ctrl+c, ctrl+d, ctrl+z, and anything else I could think of. The pi was in fact registering key strokes (per what was displayed on the LCD), but nothing seemed to break it out.

The only solution that worked was for me to unplug the energy like @gehrlein did. Would be great to know if there is a better workaround.

rootzoll, 2020-07-04:
if you activate the touchscreen you should have shutdown & restart options - even when WIFI connection gets lost.

philbertw4, 2020-07-04:
Thanks for the quick reply @rootzoll -- I'm using (at the moment) 1.5.1 and this was on a brand new fresh install. I've since pulled the energy, added the wpa_supplicant.conf file to the sd card and started up again. However, for future reference, can you clarify what you mean by "activate the touchscreen?" It seemed non-responsive when I was touching it and I'm not sure how to activate it when it's stuck in that loop. Thanks again.

rootzoll, 2020-07-05:
Touchscreen is a feature that needs to be activated once setup is done. In v1.5 its MAINMENU > SERVIVES > Touchscreen ... in future versions it might be activated by default.

nyxnor, 2021-04-18:
please close this.

closed 2021-04-19 by rootzoll