raspiblitz/raspiblitz #378: Monitor progress and Error States on LND Scan

rootzoll, 2019-03-02:
Multiple reports indicate that LND often get stuck on the LND Scan screen. Either with a ? that indicates when a long time some kind of error. Or with a block height but making no progress for hours. It would make sense to improve the sanning and monitoring here. A reboot fixed it for a lot of people in a stuck situation. Maybe also consider switching to python script for more flexibility.

jay653, 2019-03-05:
after going through the setup of 1.0, it seemed that lnd wasn't installed correctly, because when it came to the lnd scan screen it showed a ? and a message about missing files in the lnd folder. the raspi didn't went through the full lnd setup it seems because I didn't set up password c and d in the progress and was directly redirected to the lnd scan screen. 

after I run the script mentioned in the other thread (can't find it anymore, I am on my mobile now), lnd folder was ok (no more errors when it came to lnd scan screen) but still the ? (although I got my passwords set now and got a seed).

now the raspi keeps asking for password C over and over again while on lnd scan screen but still no progress and the "?".

rootzoll, 2019-03-05:
The "?" is just a sign that the RaspiBlitz monitor script cannot filter a progress from the lnd debug logs.

I was noit able to reproduce this error - but you are not the only one reporting this, so rewriting the lnd-wallet creation and the lnd-scan for v1.1 is on my list.

To find out why lnd is stuck on your machine we would need more debug data. So when you SSH into the RaspiBlitz and it shows you that "? progress" screen. Press CTRL+C to get to the terminal. There enter the command `./XXdebugLogs.sh` and copy+past the output into this issue. Thanks.

vamp111, 2019-03-06:
*** RASPIBLITZ LOGS ***
blitzversion: 1.0
chainnetwork: bitcoin / main
 09:30:51 up 14:50,  2 users,  load average: 0.47, 0.38, 0.31

*** CHAINNETWORK SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2019-03-05 18:19:07 GMT; 15h ago
  Process: 755 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
 Main PID: 763 (bitcoind)
   CGroup: /system.slice/bitcoind.service
           └─763 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid

Warning: Journal has been rotated since unit was started. Log output is incomplete or unavailable.

*** LAST 20 CHAINNETWORK LOGS ***
2019-03-06T06:23:14Z UpdateTip: new best=00000000000000000006764506e7b2f1f5de4911149995cb20c80d578720dbc4 height=565867 version=0x20000000 log2_work=90.413031 tx=389421215 date='2019-03-06T06:22:27Z' progress=1.000000 cache=90.2MiB(810023txo) warning='35 of last 100 blocks have unexpected version'
2019-03-06T06:26:50Z UpdateTip: new best=00000000000000000005a82bde391cdbb4efbab4b60ba1b0a1b2e2b46d379ab2 height=565868 version=0x20000000 log2_work=90.413054 tx=389422414 date='2019-03-06T06:26:03Z' progress=1.000000 cache=90.4MiB(811800txo) warning='35 of last 100 blocks have unexpected version'
2019-03-06T06:41:37Z UpdateTip: new best=00000000000000000021dfbff51cdf4f6646b7fbc418bca4cbfd9489291abad9 height=565869 version=0x20000000 log2_work=90.413077 tx=389424294 date='2019-03-06T06:41:09Z' progress=1.000000 cache=91.0MiB(818297txo) warning='34 of last 100 blocks have unexpected version'
2019-03-06T06:44:48Z Pre-allocating up to position 0xa00000 in rev01551.dat
2019-03-06T06:44:48Z UpdateTip: new best=0000000000000000000d6f74c2d7f3a4b4dac5aa39b4089ee0a074a29269590f height=565870 version=0x20000000 log2_work=90.4131 tx=389425465 date='2019-03-06T06:44:24Z' progress=1.000000 cache=91.1MiB(818625txo) warning='34 of last 100 blocks have unexpected version'
2019-03-06T06:47:35Z socket recv error Connection reset by peer (104)
2019-03-06T06:52:58Z UpdateTip: new best=00000000000000000002ac38690ac27ee19a9fde34dde123fbaf1d16a49d5523 height=565871 version=0x20000000 log2_work=90.413123 tx=389428159 date='2019-03-06T06:52:23Z' progress=1.000000 cache=91.3MiB(820567txo) warning='33 of last 100 blocks have unexpected version'
2019-03-06T07:02:44Z UpdateTip: new best=000000000000000000199985017f5042c4c25867968fe124c3d546f0c11a8f40 height=565872 version=0x20000000 log2_work=90.413145 tx=389430131 date='2019-03-06T07:02:00Z' progress=1.000000 cache=91.3MiB(820874txo) warning='32 of last 100 blocks have unexpected version'
2019-03-06T07:12:58Z UpdateTip: new best=000000000000000000283219d0a04d0e43e2282248abbc1f8f792a467cf67739 height=565873 version=0x2fff8000 log2_work=90.413168 tx=389432917 date='2019-03-06T07:12:31Z' progress=1.000000 cache=91.6MiB(823561txo) warning='32 of last 100 blocks have unexpected version'
2019-03-06T07:25:59Z UpdateTip: new best=00000000000000000015cbe3af8a6c1dc3c82b381843cdc1d9b6a512a8e505a8 height=565874 version=0x20000000 log2_work=90.413191 tx=389435863 date='2019-03-06T07:25:31Z' progress=1.000000 cache=91.9MiB(826513txo) warning='32 of last 100 blocks have unexpected version'
2019-03-06T07:36:17Z UpdateTip: new best=0000000000000000000cf8f1e540295991289f08488e580ccb408b9f4b4a6f3c height=565875 version=0x2000e000 log2_work=90.413214 tx=389438371 date='2019-03-06T07:35:54Z' progress=1.000000 cache=92.1MiB(828537txo) warning='32 of last 100 blocks have unexpected version'
2019-03-06T08:08:45Z Potential stale tip detected, will try using extra outbound peer (last tip update: 1948 seconds ago)
2019-03-06T08:08:45Z New outbound peer connected: version: 70015, blocks=565875, peer=105
2019-03-06T08:10:55Z Pre-allocating up to position 0xb00000 in rev01551.dat
2019-03-06T08:10:55Z UpdateTip: new best=0000000000000000001862e6324e8b8ae9d383eaaa1bf3c93666bad15a0b0d27 height=565876 version=0x2000e000 log2_work=90.413237 tx=389440927 date='2019-03-06T08:11:06Z' progress=1.000000 cache=93.7MiB(843393txo) warning='33 of last 100 blocks have unexpected version'
2019-03-06T08:17:15Z Pre-allocating up to position 0x6000000 in blk01551.dat
2019-03-06T08:17:15Z UpdateTip: new best=0000000000000000000e57dca8122f3748d71b1ad57c40eee2c03b526d0355d8 height=565877 version=0x20000000 log2_work=90.41326 tx=389442710 date='2019-03-06T08:16:58Z' progress=1.000000 cache=94.1MiB(847222txo) warning='33 of last 100 blocks have unexpected version'
2019-03-06T08:25:08Z UpdateTip: new best=00000000000000000006e8394d936212d4265b783b517d260a93a81ec65403fe height=565878 version=0x20000000 log2_work=90.413282 tx=389443729 date='2019-03-06T08:24:46Z' progress=1.000000 cache=94.3MiB(848918txo) warning='32 of last 100 blocks have unexpected version'
2019-03-06T08:56:58Z UpdateTip: new best=000000000000000000014358180889c7fab7968fb95995ba3a5ed35a07147589 height=565879 version=0x2000e000 log2_work=90.413305 tx=389446218 date='2019-03-06T08:56:39Z' progress=1.000000 cache=95.9MiB(863429txo) warning='33 of last 100 blocks have unexpected version'
2019-03-06T09:24:53Z UpdateTip: new best=0000000000000000000a10718778b265d05bcdc65e6872645b37a8f554c83292 height=565880 version=0x20c00000 log2_work=90.413328 tx=389448840 date='2019-03-06T09:24:22Z' progress=1.000000 cache=4.7MiB( [... cut ...]
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2019-03-05 18:45:26 GMT; 14h ago
 Main PID: 3449 (lnd)
   CGroup: /system.slice/lnd.service
           └─3449 /usr/local/bin/lnd --externalip=78.130.148.218

Warning: Journal has been rotated since unit was started. Log output is incomplete or unavailable.

*** LAST 20 LND LOGS ***
-- No entries --

*** LAST 20 RTL LOGS ***
-- No entries --

vamp111, 2019-03-06:
[image]

vamp111, 2019-03-06:
after a 2nd reboot, all went live and working fine now... dunno

*my local IP has changed, why is that and can I prevent/change it back somehow?

rootzoll, 2019-03-06:
that there were no LND logs was strange ... I think it has something todo with LND, but I am not sure why yet. The whole lnd-wallet-creation and lnd-scan section is up for a rewrite for v1.1 anyway.

vamp111, 2019-03-06:
and is it a problem, that my router changes my IP

rootzoll, 2019-03-06:
@vamp111 I dont think that an publicIP change by the router is the problem here, at least not for the logs or getting stuck at that point - that is normally just relevant for being publicly reachable.

vamp111, 2019-03-07:
@rootzoll  i notices my pub. adress of the node is displayed in red on the LCD, so basicly unreachable from the outside, however a few new nodes were able to connect with me, dunno...

rootzoll, 2019-03-07:
The "public detection" is not perfect yet - will see if we find for v1.1 a better way to do it.

ctrlF4runner, 2019-03-07:
I'm getting the same issue where the LND Scan is displaying '?'. During the initial setup, everything finished. But when I shut the system down and restarted it, I'm always stuck here. I've been waiting for over 8 hours.

Any ideas what I should do?

Here's my log:
^Cadmin@LiteNode:~ $ ./XXdebugLogs.sh

*** RASPIBLITZ LOGS ***
blitzversion: 1.0
chainnetwork: bitcoin / main
 15:53:59 up 16:00,  2 users,  load average: 0.28, 0.60, 0.68

*** CHAINNETWORK SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2019-03-06 23:17:30 GMT; 16h ago
  Process: 781 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
 Main PID: 790 (bitcoind)
   CGroup: /system.slice/bitcoind.service
           └─790 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoi…id

Warning: Journal has been rotated since unit was started. Log output is incomplete or unavailable.

*** LAST 20 CHAINNETWORK LOGS ***
2019-03-07T14:03:37Z Potential stale tip detected, will try using extra outbound peer (last tip update: 2081 seconds ago)
2019-03-07T14:03:37Z socket recv error Connection reset by peer (104)
2019-03-07T14:03:38Z New outbound peer connected: version: 70015, blocks=566048, peer=92
2019-03-07T14:04:11Z UpdateTip: new best=0000000000000000000b13773c53f164a58efa5c48a95fa0718e4aa7c1102b7e height=566049 version=0x3fff0000 log2_work=90.41718 tx=389786319 date='2019-03-07T14:04:04Z' progress=1.000000 cache=4.1MiB(6213txo) warning='32 of last 100 blocks have unexpected version'
2019-03-07T14:06:23Z UpdateTip: new best=00000000000000000009347b87f06ec1d054b803ff7d5d3c5bc8d6f10930b65b height=566050 version=0x2fff4000 log2_work=90.417202 tx=389787970 date='2019-03-07T14:06:13Z' progress=1.000000 cache=4.8MiB(12859txo) warning='32 of last 100 blocks have unexpected version'
2019-03-07T14:16:05Z Leaving block file 1552: CBlockFileInfo(blocks=135, size=134105422, heights=565912...566050, time=2019-03-06...2019-03-07)
2019-03-07T14:16:05Z Pre-allocating up to position 0x1000000 in blk01553.dat
2019-03-07T14:16:06Z Pre-allocating up to position 0x100000 in rev01553.dat
2019-03-07T14:16:06Z UpdateTip: new best=0000000000000000001cf9dcb6fdea0f6744d95649eea10253c094f9729f9af0 height=566051 version=0x2fff4000 log2_work=90.417225 tx=389790731 date='2019-03-07T14:15:33Z' progress=1.000000 cache=6.2MiB(25536txo) warning='33 of last 100 blocks have unexpected version'
2019-03-07T14:37:04Z socket recv error Connection reset by peer (104)
2019-03-07T14:44:23Z UpdateTip: new best=00000000000000000005ed5f777154c758e326ada99599b760a5c7ef96dfb9cd height=566052 version=0x20000000 log2_work=90.417248 tx=389793346 date='2019-03-07T14:44:12Z' progress=1.000000 cache=9.0MiB(52292txo) warning='33 of last 100 blocks have unexpected version'
2019-03-07T14:45:37Z UpdateTip: new best=00000000000000000016ea7a358e193cb869e0f14e44f2b902ca3799ac9f2adf height=566053 version=0x20000000 log2_work=90.417271 tx=389795378 date='2019-03-07T14:45:22Z' progress=1.000000 cache=9.6MiB(57224txo) warning='32 of last 100 blocks have unexpected version'
2019-03-07T15:02:24Z socket recv error Connection reset by peer (104)
2019-03-07T15:17:10Z Potential stale tip detected, will try using extra outbound peer (last tip update: 1893 seconds ago)
2019-03-07T15:17:11Z New outbound peer connected: version: 70015, blocks=566053, peer=102
2019-03-07T15:20:55Z UpdateTip: new best=000000000000000000206e204b97bf5a4d98f72603a3d7a8592b80f03743670f height=566054 version=0x20000000 log2_work=90.417293 tx=389798068 date='2019-03-07T15:20:23Z' progress=1.000000 cache=11.5MiB(75219txo) warning='32 of last 100 blocks have unexpected version'
2019-03-07T15:22:07Z socket recv error Connection reset by peer (104)
2019-03-07T15:42:34Z UpdateTip: new best=00000000000000000005aa8f23234339789960a153af543bd68466e226fb82a7 height=566055 version=0x20c00000 log2_work=90.417316 tx=389800540 date='2019-03-07T15:42:25Z' progress=1.000000 cache=12.5MiB(84373txo) warning='33 of last 100 blocks have unexpected version'
2019-03-07T15:47:17Z Pre-allocating up to position 0x200000 in rev01553.dat
2019-03-07T15:47:18Z UpdateTip: new best=00000000000000000016a773e89257912de720d4fdf5c4599fb48b2bb167f3fe height=566056 version=0x20c00000 log2_work=90.417339 tx=389803531 date='2019-03-07T15:46:44Z' progress=1.000000 cache=13.1MiB(90005txo) warning='34 of last 100 blocks have unexpected version'

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2019-03-06 23:17:30 GMT; 16h ago
 Main PID: 791 (lnd)
   CGroup: /system.slice/lnd.service
           └─791 /usr/local/bin/lnd --externalip=75.168.103.87

Warning: Journal has been rotated since unit was started. Log output is incomplete or unavailable.

*** LAST 20 LND LOGS ***
-- No entries --

- RTL is OFF by config
admin@LiteNode:~ $

ctrlF4runner, 2019-03-07:
It finally completed. I guess it wasn't hung up and it was just taking a long time to sync.

rootzoll, 2019-03-07:
This is confusing people a lot .. better scan and error detection at this point of the setup is on the list for v1.1. release.

jay653, 2019-03-10:
after been stuck for >1day in ln scan i decided to make a clean reset of the pi and flashed an untouched img. now it's working without any issues and i was not able to reproduce the error.

I did not change anything compared to my first try so i still dont know what the trigger was.

vamp111, 2019-03-14:
sometimes my screen just shows the same scan  ?/5******* stucks on this for a few seconds/minutes and autoresumes normal behaviour

rootzoll, 2019-03-14:
@vamp111 The screen gets displayed if the LND gets out of sync. The actual version of scan script is very primitive and has just a limited scope of analysing. Its not critical (LND just needs some minutes sometime to get back on track) but the displaying of the RaspiBlitz is to be improved in the future.

vamp111, 2019-03-21:
ver. 1.0 i get this error after reboot
[image]

vamp111, 2019-03-21:
here the log:
*** RASPIBLITZ LOGS ***
blitzversion: 1.0
chainnetwork: bitcoin / main
 20:01:11 up 6 min,  2 users,  load average: 3.58, 2.26, 1.02

*** CHAINNETWORK SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2019-03-21 19:55:11 GMT; 6min ago
  Process: 735 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
 Main PID: 743 (bitcoind)
   CGroup: /system.slice/bitcoind.service
           └─743 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoi…id

Mar 21 19:55:10 FreeTradeBulgaria systemd[1]: Starting Bitcoin daemon...
Mar 21 19:55:11 FreeTradeBulgaria systemd[1]: Started Bitcoin daemon.

*** LAST 20 CHAINNETWORK LOGS ***
2019-03-21T19:57:31Z net thread start
2019-03-21T19:57:31Z dnsseed thread start
2019-03-21T19:57:31Z addcon thread start
2019-03-21T19:57:31Z opencon thread start
2019-03-21T19:57:31Z msghand thread start
2019-03-21T19:57:31Z init message: Done loading
2019-03-21T19:57:32Z New outbound peer connected: version: 70015, blocks=568172, peer=0
2019-03-21T19:57:42Z Loading addresses from DNS seeds (could take a while)
2019-03-21T19:57:44Z 204 addresses found from DNS seeds
2019-03-21T19:57:44Z dnsseed thread exit
2019-03-21T19:59:21Z UpdateTip: new best=000000000000000000093cd7d3168c85d059966204564f706b1e1e933abaf065 height=568171 version=0x3c444000 log2_work=90.464668 tx=393772408 date='2019-03-21T20:01:17Z' progress=1.000001 cache=1.3MiB(11433txo) warning='35 of last 100 blocks have unexpected version'
2019-03-21T20:00:02Z UpdateTip: new best=0000000000000000000f11dcf424c3ddc973242f0879459d8215bb8c1abf487e height=568172 version=0x20000000 log2_work=90.46469 tx=393773174 date='2019-03-21T20:03:07Z' progress=1.000001 cache=1.7MiB(14988txo) warning='35 of last 100 blocks have unexpected version'
2019-03-21T20:00:03Z New outbound peer connected: version: 70015, blocks=568172, peer=2
2019-03-21T20:00:04Z New outbound peer connected: version: 70015, blocks=568172, peer=3
2019-03-21T20:00:16Z New outbound peer connected: version: 70015, blocks=568172, peer=4
2019-03-21T20:00:16Z New outbound peer connected: version: 70015, blocks=568172, peer=5
2019-03-21T20:00:22Z New outbound peer connected: version: 70015, blocks=568172, peer=6
2019-03-21T20:00:24Z New outbound peer connected: version: 70015, blocks=568172, peer=7
2019-03-21T20:00:32Z New outbound peer connected: version: 70015, blocks=568172, peer=8
2019-03-21T20:00:36Z Imported mempool transactions from disk: 157 succeeded, 2696 failed, 0 expired, 0 already there

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2019-03-21 19:57:40 GMT; 3min 30s ago
 Main PID: 2755 (lnd)
   CGroup: /system.slice/lnd.service
           └─2755 /usr/local/bin/lnd --externalip=78.130.148.218

Mar 21 20:01:03 FreeTradeBulgaria lnd[2755]: 2019-03-21 20:01:03.934 [DBG] RPCS: [walletbalance] Total balance=0…084 BTC
Mar 21 20:01:04 FreeTradeBulgaria lnd[2755]: 2019-03-21 20:01:04.795 [INF] RPCS: [listchannels] fetched 20 chann…from DB
Hint: Some lines were ellipsized, use -l to show in full.

*** LAST 20 LND LOGS ***
-- Logs begin at Thu 2019-03-21 19:54:50 GMT, end at Thu 2019-03-21 20:01:11 GMT. --
Mar 21 20:00:23 FreeTradeBulgaria lnd[2755]: 2019-03-21 20:00:23.300 [DBG] CNCT: ChannelArbitrator(8086fe45e010fd0ef2c44f8cd26b38e1f41316d489d99566f60d8f43e74a1102:0): checking chain actions at height=568172
Mar 21 20:00:23 FreeTradeBulgaria lnd[2755]: 2019-03-21 20:00:23.352 [INF] DISC: Authenticated Gossiper is starting
Mar 21 20:00:23 FreeTradeBulgaria lnd[2755]: 2019-03-21 20:00:23.352 [INF] BRAR: Starting contract observer, watching for breaches.
Mar 21 20:00:23 FreeTradeBulgaria lnd[2755]: 2019-03-21 20:00:23.353 [INF] NTFN: New block epoch subscription
Mar 21 20:00:23 FreeTradeBulgaria lnd[2755]: 2019-03-21 20:00:23.405 [INF] CRTR: FilteredChainView starting
Mar 21 20:00:28 FreeTradeBulgaria lnd[2755]: 2019-03-21 20:00:28.639 [DBG] DISC: Skipping retransmission of channel without AuthProof: 624555589996969984
Mar 21 20:00:34 FreeTradeBulgaria lnd[2755]: 2019-03-21 20:00:34.613 [DBG] RPCS: [walletbalance] Total balance=0.00432084 BTC
Mar 21 20:00:34 FreeTradeBulgaria lnd[2755]: 2019-03-21 20:00:34.741 [DBG] RPCS: [walletbalance] Total balance=0.00432084 BTC
Mar 21 20:00:35 FreeTradeBulgaria lnd[2755]: 2019-03-21 20:00:35.665 [INF] RPCS: [listchannels] fetched 20 channels from DB
Mar 21 20:00:36 FreeTradeBulgaria lnd[2755]: 2019-03-21 20:00:36.764 [DBG] RPCS: [walletbalance] Total balance=0.00432084 BTC
Mar 21 20:00:36 FreeTradeBulgaria lnd[2755]: 2019-03-21 20:00:36.896 [DBG] RPCS: [walletbalance] Total balance=0.00432084 BTC
Mar 21 20:00:37 FreeTradeBulgaria lnd[2755]: 2019-03-21 20:00:37.910 [INF] RPCS: [listchannels] fetched 20 channels from DB
Mar 21 20:00:48 FreeTradeBulgaria lnd[2755]: 2019-03-21 20:00:48.120 [DBG] RPCS: [walletbalance] Total balance=0.00432084 BTC
Mar 21 20:00:48 FreeTradeBulgaria lnd[2755]: 2019-03-21 20:00:48.223 [DBG] RPCS: [walletbalance] Total balance=0.00432084 BTC
Mar 21 20:00:49 FreeTradeBulgaria lnd[2755]: 2019-03-21 20:00:49.185 [INF] RPCS: [listchannels] fetched 20 channels from DB
[... cut ...]
Mar 21 20:01:03 FreeTradeBulgaria lnd[2755]: 2019-03-21 20:01:03.934 [DBG] RPCS: [walletbalance] Total balance=0.00432084 BTC
Mar 21 20:01:04 FreeTradeBulgaria lnd[2755]: 2019-03-21 20:01:04.795 [INF] RPCS: [listchannels] fetched 20 channels from DB

*** LAST 20 RTL LOGS ***
-- Logs begin at Thu 2019-03-21 19:54:50 GMT, end at Thu 2019-03-21 20:01:11 GMT. --
Mar 21 19:55:11 FreeTradeBulgaria systemd[1]: Started RTL daemon.
Mar 21 19:55:16 FreeTradeBulgaria node[745]: Server is up and running, please open the UI at http://localhost:3000

fluidvoice, 2019-03-21:
I would upgrade to v1.1

vamp111, 2019-03-22:
sure, its what I planned for the weekend, thx

nyxnor, 2021-04-18:
pls close this.

closed 2021-04-19 by rootzoll