raspiblitz/raspiblitz #381: [lncli] rpc error after recovering old .lnd folder

CommanderPoe, 2019-03-02:
Hey all, This morning i proceded and recovered my old channel state replacing the whole .lnd folder except for the lnd.conf which i kept. After doing this, took a while catching up but in the end it went smooth, lnd seem to be working smoothly now that its finally updated and the logs seem to be as well printing the correct info... Now, whenever i try to run any command that includes `lncli` it prints this error: _[lncli] rpc error: code = Unavailable desc = all SubConns are in TransientFailure, latest connection error: connection error: desc = "transport: authentication handshake failed: x509: certificate signed by unknown authority"_ 

Worth saying that i used the bitcoin user to replace the .lnd files via SFTP and i unlocked the wallet with no issues. 
[image]

CommanderPoe, 2019-03-02:
Seems to be an issue with certificates and owners cuz i can run all commands with the bitcoin user.... By the way i recovered the info of my old channel state but they all seem to be inactive, altho i can see both the onchain and offchain balance i had before, so it actually works. 

Now if you guys can help me out with the owner that be much appreciated.

CommanderPoe, 2019-03-02:
Did something stupid with the certs and macaroon but in the end i managed to fix it... it was all a matter of given the right certs to the admin user

closed 2019-03-02 by CommanderPoe

malixg, 2019-03-08:
getting this issue too.. exactly how did you resolve it?

rootzoll, 2019-03-09:
@malixg can you try if this FAQ info fixes your problem: "How to solve a "signature mismatch after caveat verification" error?"
https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#how-to-solve-a-signature-mismatch-after-caveat-verification-error

rytisbitcoin, 2019-05-16:
For anyone trying to resolve this issue, do NOT choose the "RESET Macaroons & TLS" option, because this will delete your recovered macaroons and create new ones, which is this case if not what you would want. I am trying to solve the same problem now, and if I find the solution I will update this comment with it.