raspiblitz/raspiblitz #382: paymentrequest timeout after 1min

kangoala, 2019-03-03:
After setting up RaspiBlitz 1.0 I tried to pay on SatoshisPlace. I copied the payment request to the RTL-web-interface (Payments) but couldn't pay the service (timeout after 1min).
There are 4 peers connected and I have already opened two channels. However, bitcoinID and node adress are both displayed in red. Connection to my internet provider is by IPv6 and I can't set portforwarding on my router.
Is IPv6 the problem in my case? What settings can I change to become able to pay with my RaspiBlitz?

rootzoll, 2019-03-06:
I dont think thats a IPv6 problem. If you have channels open and active that should show it worked with your network settings. I had also on a new node problems to find a route to SatoshisPlace - tried some other channels and waited a day to let the gossip protocol to build up the network map and then got it working.

kangoala, 2019-03-06:
Ok, today it worked fine. After approx. 30 sec. and 6 hops the payment request was sent succesfully.
Thanks, great work!!!

rootzoll, 2019-03-06:
I have not verified yet, but I really thing it has todo that the fresh LND node needs to gather info about the network thru the gossip protocol first before its able to find good routes thru the network - and that can take some time.

closed 2019-03-06 by rootzoll

kangoala, 2019-03-06:
Ok, after paying SatoshisPlace I have a small amount of Satoshis on my remote balance. However I can not send an even smaller amount from my mobile eclair wallet back to the raspi node. I also can't open a channel to my raspi blitz. Does anybody know the reason for that? What else do I have to do? (bitcoinID and node adress are still both displayed in red)

kangoala, 2019-03-06:
Another test to send some satoshi to SatoshisPlace also ended with a timeout. The routing succes ist still quite uncertain. I will now try to open some more channels to see whether that helps.