raspiblitz/raspiblitz #384: Can not get valid macaroon

StifflersMoM69, 2019-03-03:
Hi!
I am trying to connect my mobile wallet (shango) and i always get this error after scanning QR Code.

rootzoll, 2019-03-06:
Shango changed the content of there QR code - will adapt in v1.1 release.

rootzoll, 2019-03-13:
There was already a commit on master to fix the QR code: https://github.com/rootzoll/raspiblitz/commit/e754d531c748af70f7344a4ac8c2a66fb0c2d7a2

Check if that is working with the lastest version of Shango.

StifflersMoM69, 2019-03-13:
I can't get it work. Now the code is not beeing recognized. Nothing happens.

rootzoll, 2019-03-13:
In version 1.0 there is still that old connect QR code that is not working. Please try again once v1.1 is released and you made the update.

rootzoll, 2019-03-17:
OK v1.1 tested fine with the latest Shango app. Closing issue for v1.1 release.

closed 2019-03-17 by rootzoll