raspiblitz/raspiblitz #385: qrencode: command not found

gehrlein, 2019-03-03:
Trying to set up a mobile wallet with Shango it wants me to press ENTER to generate ("generare") the QR code. When I press Enter it gives:

*** STEP 2 : SCAN MACAROON (make whole QR code fill camera) ***
*** STEP 2 : SCAN MACAROON (make whole QR code fill camera) ***
./97addMobileWalletShango.sh: line 46: qrencode: command not found
(To shrink QR code: OSX->CMD- / LINUX-> CTRL-) Press ENTER when finished.

rootzoll, 2019-03-06:
Everytime that script is run it should execute `sudo apt-get install qrencode -y ` .. you could try it again. If it still not working, try the apt-get-command directly on terminal - see if there is any error installing it. 

BTW shango changes the content of there qr code and the script needs update. Will be fixed with v1.1

frennkie, 2020-08-08:
Please try on current version (V1.6) and re-open is issue persists.

closed 2020-08-08 by frennkie