raspiblitz/raspiblitz #386: Only change externalip / port

LightningK0ala, 2019-03-03:
I have multiple lightning nodes running behind the same router.
To expose them publicly, I port forward non-default ports in the router, i.e. external 9745 maps to an internal 9735.

The only thing left is to set externalip to announce the public port (9745).

I ran into a gotcha because I tried to set it in the lnd.conf file, but hadn't realised it's being included in the systemd command.

Perhaps there could also be a config script similar to lnd.setport.sh but lnd.setexternalip.sh?
Or at least some documentation for this use-case?

Just a suggestion in case someone else has a similar requirement.

rootzoll, 2019-03-05:
Yes setting a externalIP should get more easier with next version. Had a similar request here: https://github.com/rootzoll/raspiblitz/issues/401 .. putting this in the v1.1 milestone.

rootzoll, 2019-04-03:
test of wirking with IP and Domain

rootzoll, 2019-04-27:
This will be part of v1.2 - see https://github.com/rootzoll/raspiblitz/issues/401
closing this issue

closed 2019-04-27 by rootzoll