raspiblitz/raspiblitz #387: Lightning fails to start

hopitas, 2019-03-03:
After power failure I've got this message when LND tries to start:

`/home/admin $ lnd
2019-03-03 13:53:45.545 [INF] LTND: Version: 0.5.2-beta commit=v0.5.2-beta, build=production, logging=default
2019-03-03 13:53:45.545 [INF] LTND: Active chain: Bitcoin (network=mainnet)
2019-03-03 13:53:45.546 [INF] CHDB: Checking for schema update: latest_version=7, db_version=7
2019-03-03 13:53:45.551 [INF] RPCS: password RPC server listening on [::]:10009
2019-03-03 13:53:45.552 [INF] RPCS: password gRPC proxy started at [::]:8080
2019-03-03 13:53:45.553 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.
2019-03-03 13:54:23.152 [INF] LNWL: Opened wallet
2019-03-03 13:54:23.650 [INF] LTND: Primary chain is set to: bitcoin
2019-03-03 13:54:23.696 [INF] LNWL: Started listening for bitcoind block notifications via ZMQ on tcp://*:28332
2019-03-03 13:54:23.696 [INF] LTND: Initializing bitcoind backed fee estimator
2019-03-03 13:54:23.697 [INF] LNWL: Started listening for bitcoind transaction notifications via ZMQ on tcp://*:28333
2019-03-03 13:54:23.702 [DBG] LNWL: Using minimum fee rate of 253 sat/kw
2019-03-03 13:54:30.125 [INF] LNWL: The wallet has been unlocked without a time limit
2019-03-03 13:54:30.777 [INF] LTND: LightningWallet opened
2019-03-03 13:54:30.777 [DBG] LNWL: Birthday block has already been verified: height=537678, hash=000000000000000000279fd8582a9d3ee830e6031f34eccdc597a03e60cbe7b4
2019-03-03 13:54:30.964 [INF] HSWC: Restoring in-memory circuit state from disk
2019-03-03 13:54:30.995 [INF] HSWC: Payment circuits loaded: num_pending=0, num_open=0
2019-03-03 13:54:31.031 [INF] HSWC: Trimming open circuits for chan_id=564058:186:1, start_htlc_id=3
2019-03-03 13:54:31.032 [INF] HSWC: Trimming open circuits for chan_id=564064:308:0, start_htlc_id=0
2019-03-03 13:54:31.067 [INF] LTND: Shutdown complete
panic: page 16577 already freed

goroutine 1 [running]:
github.com/coreos/bbolt.(*freelist).free(0x25bcee0, 0x4659e6, 0x0, 0x62be6000)
        /Users/roasbeef/gocode/pkg/mod/github.com/coreos/bbolt@v0.0.0-20180223184059-7ee3ded59d4835e10f3e7d0f7603c42aa5e83820/freelist.go:143 +0x468
github.com/coreos/bbolt.(*node).spill(0x25b8980, 0x24a4600, 0xc5c930)
        /Users/roasbeef/gocode/pkg/mod/github.com/coreos/bbolt@v0.0.0-20180223184059-7ee3ded59d4835e10f3e7d0f7603c42aa5e83820/node.go:363 +0x1f0
github.com/coreos/bbolt.(*node).spill(0x25b87c0, 0x24a45f0, 0xc5c930)
        /Users/roasbeef/gocode/pkg/mod/github.com/coreos/bbolt@v0.0.0-20180223184059-7ee3ded59d4835e10f3e7d0f7603c42aa5e83820/node.go:350 +0x98
github.com/coreos/bbolt.(*node).spill(0x25b8780, 0x28b13b4, 0x5)
        /Users/roasbeef/gocode/pkg/mod/github.com/coreos/bbolt@v0.0.0-20180223184059-7ee3ded59d4835e10f3e7d0f7603c42aa5e83820/node.go:350 +0x98
github.com/coreos/bbolt.(*Bucket).spill(0x264abe0, 0x264ab00, 0x28b15fc)
        /Users/roasbeef/gocode/pkg/mod/github.com/coreos/bbolt@v0.0.0-20180223184059-7ee3ded59d4835e10f3e7d0f7603c42aa5e83820/bucket.go:568 +0x384
github.com/coreos/bbolt.(*Bucket).spill(0x24a020c, 0xbf171505, 0xa9fd3e57)
        /Users/roasbeef/gocode/pkg/mod/github.com/coreos/bbolt@v0.0.0-20180223184059-7ee3ded59d4835e10f3e7d0f7603c42aa5e83820/bucket.go:535 +0x2e4
github.com/coreos/bbolt.(*Tx).Commit(0x24a0200, 0x0, 0x0)
        /Users/roasbeef/gocode/pkg/mod/github.com/coreos/bbolt@v0.0.0-20180223184059-7ee3ded59d4835e10f3e7d0f7603c42aa5e83820/tx.go:160 +0xb8
github.com/coreos/bbolt.(*DB).Update(0x2870240, 0x28b16ec, 0x0, 0x0)
        /Users/roasbeef/gocode/pkg/mod/github.com/coreos/bbolt@v0.0.0-20180223184059-7ee3ded59d4835e10f3e7d0f7603c42aa5e83820/db.go:677 +0xd4
github.com/lightningnetwork/lnd/channeldb.(*ChannelGraph).SetSourceNode(0x264c198, 0x24a0180, 0x47, 0xac5fb5fc)
        /Users/roasbeef/gocode/src/github.com/lightningnetwork/lnd/channeldb/graph.go:329 +0x5c
main.newServer(0x25be5f8, 0x1, 0x1, 0x2802fb0, 0x263c000, 0x264a3c0, 0x0, 0x0, 0x0)
        /Users/roasbeef/gocode/src/github.com/lightningnetwork/lnd/server.go:510 +0x1400
main.lndMain(0x0, 0x0)
        /Users/roasbeef/gocode/src/github.com/lightningnetwork/lnd/lnd.go:307 +0x960
main.main()
        /Users/roasbeef/gocode/src/github.com/lightningnetwork/lnd/lnd.go:449 +0x14
`

Any idea what should I do to try to resolve the issue?

hopitas, 2019-03-04:
I googled a bit and I think that my on channel coins are lost:

https://github.com/lightningnetwork/lnd/issues/1107

Before there is reliable backups for lnd state, there is a chance that money will be lost on power failure.

rootzoll, 2019-03-05:
Yes - thats still a problem and lnd state backups are much needed. Hope your loss of funds are not hard ... its a soft reminder not to get too "reckless" at this point of deleopment.

closed 2019-03-05 by rootzoll