raspiblitz/raspiblitz #388: html Status Page

raumi75, 2019-03-03:
Discussing #88 I posted a command to use a browser to render a dynamic web page on the LCD of the RaspiBlitz without using too much resources.

This weekend I've toyed around with that idea and made some svg-graphics that should render on the pi. The html-code so far is only a mockup, but it's possible to breathe life into this. @frennkie has already prepared python code to make the info-page more flexible.

I'd love to hear feedback from everyone. It's okay to shoot me down, if you think this is not the RaspiBlitz-Way of doing it. I won't take it personal.

You can take a look at it right here: https://jsfiddle.net/ygph17wk/1/ (click run to see the result) 

If you want to change this mockup, please use this link: https://jsfiddle.net/ygph17wk/1/#&togetherjs=JM64iTIekQ

raumi75, 2019-03-03:
This would be a dark mode

https://jsfiddle.net/raumi75/g8zae1tL/31/

rootzoll, 2019-03-05:
Great work. i hope you already got some feedback on twitter. Let me know if you have further revisions ready - I will share on twitter to get more feedback on it.

rootzoll, 2019-03-05:
I am putting this on the v1.1 release milstone. But no pressure and no promise - the main focus of the 1.1 release will be making maintenace and recover easier - but every cool improvement otherwise that is ready by then (targeting (late) April) I love to include :)

jay653, 2019-03-07:
i like the darkmode! will look good with a fancy case for the raspi.

frennkie, 2020-05-17:
@raumi75 I'm making progress on this.. do want to complete the mockup...?! :-D

rootzoll, 2022-05-09:
HTML status page will get replaced with WebUI login on v1.8.0 release. Closing issue.

cstenglein, 2023-06-07:
> Closing issue.

Wasn't closed until now. Closing.

closed 2023-06-07 by cstenglein