raspiblitz/raspiblitz #389: Trouble understanding Channels

eliddell1, 2019-03-03:
when i list my open channels i see channels, but don't understand the color coding and icons.

2 channels are red, and one of those has a blue dot by it.  The other two are white.

I tried looking for documentation on this but nothing.. please help me understand the display and what any potential problems/ fixes there are.

CommanderPoe, 2019-03-04:
as far as im concerned red channels mean Inactive on the raspiblitz, furthermore you can always check on the cli with `lnlci listchannels` and see the info it prints for certain kind of channels.

eliddell1, 2019-03-04:
Yeah i gathered that, but does inactive mean the channel is dead, or just that there is  no activity..

CommanderPoe, 2019-03-04:
> Yeah i gathered that, but does inactive mean the channel is dead, or just that there is no activity..

if not yours.. maybe the other node's offline. Some nodes go offline from times to times due to public ip changes or some other minor issue, that can cause innactivity on that channel for a period of time. Im talking here from empiric experience not an actual reason.

eliddell1, 2019-03-04:
fair and good to know.. thanks..

closed 2019-03-05 by rootzoll