raspiblitz/raspiblitz #391: Invalid transfer for blockchain data

DariusParvin, 2019-03-04:
I am getting an error when doing the "Copy" option ("Blockchaindata from another node with SCP") for transferring blockchain data from my btc core node on my laptop connected to the raspberrypi through an ethernet cable.

It takes about 4 hours to finish doing the SCP and this is what it the two windows look like:

[image]

Then on the ssh terminal on the right, I hit enter and get this message 

[image]

followed by this screen

[image]

I'm going to try one of the other methods for now, but could you help me figure out what's going wrong? Thanks

CommanderPoe, 2019-03-04:
Did u stop the client in your laptop before copying the files? Did you set added the _txindex=1_ on the bitcoin.conf of your laptop's client?

DariusParvin, 2019-03-04:
I opened up the bitcoin.conf file and added _txindex=1_ . The rest of the bitcoin.conf file should empty right?

If you're saying mine was fast then im guessing perhaps it hadn't finished. The screenshot i took above was when I hit enter. It looked like it had stopped based on the lack of activity on the terminal but maybe it was still going? Should there be a 'completed' message when doing SCP? Cheers

CommanderPoe, 2019-03-04:
Yes, the rest is ok blank.

Regarding the copy i had the same issue as you, first time it did not work and it was pretty annoying not knowing the progress of the copy... i ended up using Cyberduck (which is the same as SCP for windows) and transfered all the files manually using SFTP. At least there i could see the progress, when it finished i just pressed enter and it worked like a charm.

rootzoll, 2019-03-05:
Using SCP as a transfere command was a first shot on offering that COPY option. And yes SCP is not giving a "finished" message - thats a bummer. The community already suggested to optimization on that in switching to rsync that afaik should give better feedback on the transfer. 

See: https://github.com/rootzoll/raspiblitz/issues/397

So I hope with version v1.1 this will get a better user experience.
I am closing this issue - but feel free to open up again if still questions are open.

closed 2019-03-05 by rootzoll