raspiblitz/raspiblitz #392: RESEARCH: Channel Rebalancing Script

rootzoll, 2019-03-04:
Take a look at this tweet:
https://twitter.com/ockeghem7/status/1102358521100808194 based on the works of:
https://github.com/C-Otto/rebalance-lnd

It would ne an awesome idea to add such rebalancing scripts as a service you can switch on/off set with some parameters on the RaspiBlitz.

This thread is for everybody wants to experiment with that, share experience and maybe even implement it as a service.

For switchinhg on/off think about a script like `/home/admin/config.scripts/service.rebalance.sh` - take a look how the other config scripts in that directory are written. Also the RaspiBlitz has a background-script `/home/admin/_background.sh` that can (when service is signaled on in the `/mnt/hdd/raspiblitz.conf`) check if a running of that rebalancing script is necessary and execute it.

bitsam, 2019-06-05:
@rootzoll . The twitter account is now disabled. I was just reading today about the C-Otto rebalance script and would like to give it a try.

rootzoll, 2019-06-05:
@bitsam maybe also take a look at: https://github.com/bitromortac/lndmanage

scarolan, 2022-01-24:
This looks awesome. I would love to have something like this auto-rebalance channels for me.

rootzoll, 2022-01-24:
Re-adding lndmanage is on our roadmap for v1.8: see #2314

openoms, 2022-01-25:
> This looks awesome. I would love to have something like this auto-rebalance channels for me.

`bos rebalance` is working very well
Install `Balance of Satoshis` from SERVICES for LND.
C-lightning has `rebalance` plugin and also CLBOSS achieves balanced channels.