raspiblitz/raspiblitz #393: Connection to ZAP mobile wallet IOS

CommanderPoe, 2019-03-04:
Do any of you guys mind sharing with me your love story with ZAP for IOS? I still cant connect, when scanning the QR on the mobile wallet it says "Unable to connect" Port 10009 is opened and node working like a charm, tho I have not used DNS service ( does it really matter ? ) since i truly dont know how that works .

Jantokil, 2019-03-04:
Actually I succeeded to connect ZAP (iOS) in version 0.98. Used the freeDNS service as they suggested. Although I repeatedly got errors when I tried to send funds. 

Now with the update to version 1.0 I have the same problem as you describe. Can't connect to the server by scanning the QR code. I have the feeling it has something to do with the DDNS though.

rootzoll, 2019-03-05:
I will add the ZAP connection to a recheck for v1.1 But with v1.0 just 5 days ago I connected a friend of mine with ZAP iOS to his new RaspiBlitz. But also ZAP iOS has a rapid release cycle, its possible that something changed.

rootzoll, 2019-03-05:
BTW make sure to be on the same local network. Also make sure that the WLAN has not IP-Isolation turned on, preventing WLAN clients to talk to other local network clients.

CommanderPoe, 2019-03-06:
> I will add the ZAP connection to a recheck for v1.1 But with v1.0 just 5 days ago I connected a friend of mine with ZAP iOS to his new RaspiBlitz. But also ZAP iOS has a rapid release cycle, its possible that something changed.

Is it compulsory to run the DNS service in order to connect to the ZAP mobile wallet? ithought i could simply connect by just opening the port on my router.

CommanderPoe, 2019-03-06:
>I have the feeling it has something to do with the DDNS though.

Prolly this, cuz i have tried almost everything except for connecting to the DDNS service and i have nor managed to make it happen.

rootzoll, 2019-03-06:
If the DDNS is OFF its the scenario to just use ZAP mobile wallet while you are connected to the same local network. The script puts the localIP of the raspiblitz into the QR code (assuming it will be stable). So there needs to be no port opening on thr router for that. If you have problems check if the mobile phone can reach the raspiblitz in the local network .. for example activate the RTL and open the URL on the mobile phone.

If the DDNS is ON the script will put your DDNS-domain into the QR code. You then need to have the port 10009 forwarded from your router - that should make the connection even when you are on another network with your mobile.

CommanderPoe, 2019-03-08:
The DDNS server simply works like a charm... It is a waste of time to try anything else... Thanks

closed 2019-03-08 by CommanderPoe

walletofsatoshi, 2019-03-21:
It worked great before zap released newer updates and raspiblitz released version 1 anyone working on the issue?

rootzoll, 2019-03-22:
@walletofsatoshi did you tried version 1.1 of raspiblitz. A fix was part of that release.

walletofsatoshi, 2019-03-22:
No I was not aware of v1.1. Will try it out

walletofsatoshi, 2019-03-31:
It’s still not working even with v 1.1 can’t comnect zap wallet

openoms, 2019-03-31:
see this with the same problem: https://github.com/rootzoll/raspiblitz/issues/476