raspiblitz/raspiblitz #395: INFO: How to get Channel Liquidity fast?

rootzoll, 2019-03-04:
Now that people have their RaspiBlitz setup the question is how to get it connected well with good incomming channel liquidity?

Best of course is if you meet other lightning node operators and ask them to open a channel with you. Beside that groundwork - lets collect some info what other ways there are to get Channel Liquidity fast.

BEWARE: Those services listed here are just a collection what is out there - not saying its the best way or that services are safe to use. Use at your own risk.

rootzoll, 2019-03-04:
Lightning Singles is a directory service for channel liquidity matchmaking on the Bitcoin Lightning Network. Submit your node information and ask the community to connect.

https://github.com/shocknet/singles-api

rootzoll, 2019-03-04:
Thor: Lightning Channel-Opening Service
Buy incomming capacity.

https://www.bitrefill.com/thor-lightning-network-channels

rootzoll, 2019-03-04:
lightningto.me will provide you with a 0.02 BTC incomming bandwidth to get started.
You need at least have 10 channels open to use this service.

https://lightningto.me

rootzoll, 2019-03-04:
lightninghood.com - buy incomming capacity, also for TOR nodes

https://lightninghood.com/lightning-hood-liquidity/

rootzoll, 2019-03-04:
lightningpowerusers.com - Open a channel with https://lightningpowerusers.com/home/ and I will open one in your direction, so you have capacity to receive BTC on the LN.

https://lightningpowerusers.com/home/

vmavromatis, 2019-03-04:
> Need an inbound channel? $3.54 USD for 2m sat

https://yalls.org/about/

rootzoll, 2019-03-04:
Also just showed up on open bazzar:
Lightning Network Inbound Channel for Liquidity 

https://openbazaar.com/store/QmQbk7yu9RGiAv3eM3B92AW4pTNeM5rPGhLaQTf3u9GKCK/lightning-network-inbound-channel-for-liquidity

openoms, 2019-05-22:
Aiming to update this with anything significant, comments and PR-s are also welcome: https://github.com/openoms/lightning-node-management

thephizzz, 2019-10-31:
I have a real noobie question about balancing. I have two channels - both to the same PubKey, different channel ID - one with  a local balance and no remote balance and the other with a remote balance and no local.
Is this a correct balancing? Or does it need to be on only one channel?
See picture>
Thanks.
[image]

rootzoll, 2019-11-16:
The idea of balancing is that a ONE channel with a another can route maximum into both directions ... this is then its 1/2 local and 1/2 remote.

When you have TWO channels with the same other node to maximise into both directions your strategy looks best balanced to me.

openoms, 2021-11-08:
closing this due to inactivity. 
See a more up-to-date list: https://github.com/openoms/lightning-node-management/blob/en/createinboundliquidity.md

closed 2021-11-08 by openoms