raspiblitz/raspiblitz #396: can you split Raspiblitz_standard.stl into two files

eliddell1, 2019-03-04:
I have a small monoprice mini, and am using freecad/Cura to print.  I will admit i am not an expert with freecad, but i can't fit Raspiblitz_standard.stl as one print on my little guy.  and I can't figure out how to separate the bottom and middle pieces from that stl file.  Is it possible to deliver them as two separate stl files?

rootzoll, 2019-03-04:
I am not into 3D printing myself .. I will post your request on twitter - maybe someone is joining in :)

rootzoll, 2019-03-04:
@eliddell1 join in on twitter: https://twitter.com/JasonEbacher/status/1102699568846856193

CryptoCloaks, 2019-03-05:
Let us know if this works!

[RaspiBlitz Standard - 2 Pieces.zip](https://github.com/rootzoll/raspiblitz/files/2928685/RaspiBlitz.Standard.-.2.Pieces.zip)

CryptoCloaks, 2019-03-05:
For future reference to build on your 3D printing skills.
Take a look at a free program called Fusion360, it's an amazing program for 3D modeling etc.

It will help take your 3D printing to the next level!

rootzoll, 2019-03-05:
@eliddell1 OK - I added the two splitted files to the GitHub master branch.
@CryptoCloaks much thanks for helping out :D

closed 2019-03-05 by rootzoll