raspiblitz/raspiblitz #397: Use rsync on COPY blockchain data option

rootzoll, 2019-03-04:
See tweet: https://twitter.com/xavierfiechter/status/1102657860415176704

Check if rsync is available on win10 (when SSH tools are activated) - if yes: replave scp by rsync. If not give both options for users to choose.

rootzoll, 2019-03-04:
rsync seems hard to use on Win10 - see:
https://twitter.com/pwkad/status/1102673684379054080

rootzoll, 2019-03-04:
Maybe even ask the user which OS the other computer has and then be able to give more detailed instructions.

rootzoll, 2019-03-04:
Also check out: https://unix.stackexchange.com/questions/238152/why-is-scp-so-slow-and-how-to-make-it-faster/238207#238207

cryptomulde, 2019-03-06:
Idea: You could add backup script service that backups LND's "data" folder to an external usb stick.

`mount /dev/sdXXX /mnt/lndbackup`
`tar -cPvf /mnt/lndbackup/lnddata.tar /mnt/hdd/lnd/data/` 

In combination with an hourly cronjob it could be a nice way to easily backup lnd data. I know there are risks in restoring data, but it's better than having no backups at all.

rootzoll, 2019-03-06:
@cryptomulde I think backuping the LND data is another (this thread is about the COPY option to get the blockchain data). But there is a working issue about moving LND data to external USB storage for better protection: https://github.com/rootzoll/raspiblitz/issues/329

cryptomulde, 2019-03-06:
Oh ok. Thx. Misunderstood.

rootzoll, 2019-03-17:
Running tests with TORRENT and COPY options.

rootzoll, 2019-03-17:
MacOS rsync does not support the `--info=progress2` to show total progress, so I moved back to simple single file progress to be compatible for all UNIX. Running latest tests now.

fluidvoice, 2019-03-17:
> MacOS rsync does not support the `--info=progress2` to show total progress, so I moved back to simple single file progress to be compatible for all UNIX. Running latest tests now.

That doesn't make much sense. Isn't it compiled from the same source?  
Did you try it with `--info=progress2 --no-inc-recursive`

rootzoll, 2019-03-18:
I tried, not working on MacOS ... sadly MacOS has just rsync version 2.6.9 preinstalled - on debian its 3.1.2 ... the old version does not have the total progress info, just classic per file percent and number of files to go

rootzoll, 2019-03-18:
OK test were good with using the rsync on repair copy. Running last rest on normal setup.

rootzoll, 2019-03-19:
Worked :) closing issue.

closed 2019-03-19 by rootzoll