raspiblitz/raspiblitz #398: Displaying QR code on the LCD display

rootzoll, 2019-03-04:
see https://github.com/rootzoll/raspiblitz/pull/334 merged into dev

test, maybe clean up and also check what to do for people that run the RaspiBlitz without LCD.

raumi75, 2019-03-05:
Thanks for merging. People without LCD still have the option to display the QR the regular way via ssh.

openoms, 2021-03-04:
@rootzoll I found that `ANSIUTF8` format QRcode is way smaller then`ANSI256`.
Could this solve the scaling issues or might lead some problems in different terminals? 

Please try in different terminals and OS-es:
```
qrencode -t ANSIUTF8 "https://github.com/rootzoll/raspiblitz/issues/398"
qrencode -t ANSI256 "https://github.com/rootzoll/raspiblitz/issues/398"
```
[image]

closed 2021-04-27 by rootzoll