raspiblitz/raspiblitz #399: VirtualBox

finndev, 2019-03-04:
Hello,

Has anyone managed to run this on VirtualBox on OSX Host? I have downloaded the .img file and converted it to a .vdi file using:

```
$ VBoxManage convertdd raspiblitz-v1.0-2019-02-18.img raspiblitz.vdi
```

I've then set it as disk and booted it, however screen appears to stay black.

Any help?

cedricwalter, 2019-03-05:
VirtualBox lets you run x86 virtual machines on an x86 processor. Raspbian is a distribution for ARM processors. Raspbian cannot run in VirtualBox.

Raspbian is essentially Debian with binaries compiled to match the Pi's processor more closely than the official Debian binaries. So if you're going to your system in a virtual machine rather than on Pi hardware, there is very little reason to run Raspbian and not Debian.

If you really want to run Raspbian in a virtual machine on your PC (again, for most purposes, you might as well run Debian x86), you need one that emulates the hardware, such as QEmu.

closed 2019-03-05 by finndev

rootzoll, 2019-03-05:
@finndev cedricwalter is correct. But if you feel very experimental you could check out the `dietpi` port of raspiblitz -> see https://github.com/rootzoll/raspiblitz/tree/master/dietpi

DietPi is more flexible to processor architecture and they list "VirtualBox" on their website.