raspiblitz/raspiblitz #400: UPNP open port in router

cedricwalter, 2019-03-05:
Hi

Would you be interested by a PR to automate the opening of port in router using upnp? it seems that http://people.ds.cam.ac.uk/ssb22/setup/upnp.html . allow to do that.

rootzoll, 2019-03-05:
The LND has a `AutoNAT` feature. Last time i tried my RaspiBlitz was not starting up correctly. So I removed it from the service menu. But the config-script to turn that feature on and off is still there. You can give it a try. 

Go to terminal and you can call: `/home/admin/config.scripts/lnd.autonat.sh [on|off]`

Let me know if it worked for you.

cedricwalter, 2019-03-13:
i was thinking built it in for newbies who struggle to forward their port in their router. Even if LND would work, what about bitcoind? 

i was thinking about using upnp dependencies directly not through LND to keep it simpler and be able to open bitcoin 8333 and LND 9735

rootzoll, 2019-03-13:
@cedricwalter feels free to experiment on this. If it results in a small script that can be added to the RaspiBlitz that would be an idea.

fluidvoice, 2019-03-13:
@cedricwalter @rootzoll this needs a bit more thought/research.  
Do all routers have this?
Some claim having UPnP enabled opens security holes and depending upon a routers age and firmware level may or may not be fixed.  I have a relatively old and cheap router and have no idea if past exploits have been fixed so I disabled UPnP which, scary as it is, was enabled by default!
Given this is a bitcoin app, better to error on the side of caution and security and have people do this stuff manually... AND.... perhaps document for people to disable UPnP on their router unless they KNOW it is secure.

rootzoll, 2019-03-19:
The UPnP is not available on all routers. Last time I tryed the LND Auto-NAT feature it didnt worked with my router and it freezed LND so i had todo a manual reset of my RaspiBlitz.

@cedricwalter I agree a with @fluidvoice that opening ports on your router the user should at least be aware of what is happening and that it adds more risk. So I would not add it to the default setup process. But having a script that helps you open ports if you really want it (Service Settings) that would be interesting. Let me know if you want to further in that direction and have some code to look at.

rootzoll, 2019-05-22:
LND 0.6.1 made improvements on NAT and not being needed to restart on auto-NAT IP change. Try again for v1.3

EricK-nl, 2019-05-23:
I have been running a fullnode for months. One needs to open/forward port 8333 to accept incomming peers and fully support the bitcoin-blockchain. I think it is more safe that BitcoinCore opens (ànd closes) that port instead of me setting the port open all the time. So Why not enable UPNP by default? And so for LND...

DaReaLDeviL, 2019-06-08:
>so let's make things easy and enable UPNP by default? And so for LND...

So we are back to TOR on default if you want this - easy and anonymously

https://github.com/rootzoll/raspiblitz/issues/592

rootzoll, 2019-06-13:
OK. With v1.3 there will be under SERVICES the options for LND and BTC to switch on UPnP.
Implemented and ready for final testing.

rootzoll, 2019-07-01:
OK its working with my FritzBox - keep in mind, that on some routers you need to allow a device to open up ports itself first. Will be part of 1.3 release.

closed 2019-07-01 by rootzoll