raspiblitz/raspiblitz #401: Defining ip when reverse tunneling from Raspberry to Google Cloud

cryptomulde, 2019-03-05:
Hello,

I am running my Raspiblitz from home and I tunnel ports 1833, 8333 and 9735 to a Google Cloud instance. I don't have static IP from my ISP and I don't want my public-home-ip to be announced.

Is there a way to configure Raspiblitz to use my Gcloud static IP?

I tried several ways but none of them lead to success. (maybe I just did it wrong)

1. setting DYNDNS to my static IP (1ml.com still shows my home address and not my DNS nor GCloud IP)
2. changing raspiblitz.info in /mnt/hdd/raspiblitz.info If i changed that manually raspiblitz info screen showed me my static Gcloud IP, but after rebooting changes were lost again.

Is there a way to either add my DNS link or my GCloud static IP to be shown on 1ml.com?

I think the easiest way would be just to set my GoogleCloud IP in the right config file and I would be good.

I don't know how to do it, that I can restart plus it gets announced the 1ml.com the right way.

Appreciate your help!

CommanderPoe, 2019-03-05:
Isnt easier to run your node on tor?

cryptomulde, 2019-03-05:
I always taught I will only be able to connect to TOR nodes if I use tor.

I wrote a tutorial on that approach and I wanted to use my method.
https://medium.com/@cryptomulde/free-static-ip-for-btc-lightning-node-b1eef6b592cd

cryptomulde, 2019-03-05:
I want to be publicly reachable, but not with my home-ip, hence I tweaked with google cloud static ip.

rootzoll, 2019-03-05:
@cryptomulde you are the first requesting this - setting a static IP is not a standard feature on the RaspiBlitz. So my take aways ...

TODO: for v1.1 make it easy to set a static ip - witch config script `lnd.staticip.sh`. Maybe add an option in raspiblitz.conf like `staticIP=1` and then dont reset publicIP anymore.

To make it possible in v1.0 some manual steps are needed. You can try to set static in in the lnd.service from systemd. Do you know how to edit a systemd service?

You need to change this line:
`ExecStart=/usr/local/bin/lnd --externalip=${publicIP}`
replace `${publicIP}` with your IP string. 

Just one note on this `hack` - when changing some config/services on the raspiblitz it might overwrite the lnd.service again (dyndns, etc.).

cryptomulde, 2019-03-06:
Thx for your help!
I edited the lnd.service but it does not seem to work.
Replacing "PublicIP" with my numerical IP gives me this:
2019-03-06 01:23:32.733 [ERR] SRVR: unable to create server: can't create router: unexpected EOF
Replacing "PublicIP" with my numerical DynDNS gives me this:
 unable to create server: can't create router: ReadVarBytes: sig is larger than the max allowed size [count 166, max 80]

I have auto unlock activated. LND shuts down after the error and is trapped in unlocking loop.

cryptomulde, 2019-03-06:
2019-03-06 01:36:00.922 [ERR] SRVR: unable to create server: can't create router: unexpected EOF

Also getting that error with "PublicIP" now will have to find out tomorrow. Need to sleep :P

cryptomulde, 2019-03-06:
I played around a little more and I found out something new to me.
I can simply change raspiblitz.conf by typing `sudo nano /mnt/hdd/raspiblitz.conf`
I edit `PublicIP=` with either my numerical static IP or the DNS pointing to my IP.
Seemes easier to me than editing lnd.service - also editing lnd.service was giving my issues and failed starting after I tried editing it.

rootzoll, 2019-03-06:
Editing the raspiblitz.conf could be a way for the future. But at the moment the publicIP is checked at every startup and even in time intervalls in the background .. so the RaspiBlitz would overwrite that edit in his config again in v1.0 ... thats why editing the service is the safest way at the moment before this becomes a feature in v1.1

rootzoll, 2019-04-09:
V1.2 will habe a config script to set a fixed domain or IP for the RaspiBlitz (to be used by LND).

```
/home/admin/config.scripts/internet.dyndomain.sh on [IP or DOMAIN]
```

TODO: FInal testing before release.

rootzoll, 2019-04-28:
OK. Works on v1.2 :) Closing issue.

closed 2019-04-28 by rootzoll