raspiblitz/raspiblitz #402: syslog tor warnings

AlverGant, 2019-03-05:
First of all congratulations on this great project!

I am seeing warnings on syslog regarding the tor user and I would like to ask if there are security implications to it.

[warn] /run/tor is not owned by this user (bitcoin, 1002) but by debian-tor (111). Perhaps you are running Tor as the wrong user?
[warn] Before Tor can create a control socket in "/run/tor/control", the directory "/run/tor" needs to exist, and to be accessible only by the user and group account that is running Tor.  (On some Unix systems, anybody who can list a socket can connect to it, so Tor is being careful.)

AlverGant, 2019-03-05:
It's strange, looking at internet.tor.sh script, /var/run/tor is given bitcoin user ownership during installation:
sudo chown -R bitcoin:bitcoin /var/run/tor/
but that is being overriden after reboot:
drwxr-sr-x  2 debian-tor debian-tor   40 Mar  5 13:52 tor
Is it normal?
The version installed is
tor/stable,now 0.3.5.8-1~d90.stretch+1 armhf [installed]

rootzoll, 2019-03-05:
Thanks for checking that out. I think I saw that kind of error during implementation but thought I fixed it with the code you quoted - but I havent seen (maybe overlooked) that overwrite after reboot.

TODO: check for v1.1 release with test systems if this overwrite on reboot is reproducable.

nyxnor, 2021-04-16:
Close this. Tor [WARN] fixes above is the recent discussion about this.

closed 2021-04-16 by rootzoll