raspiblitz/raspiblitz #403: unable to make lightning payments over tor

AlverGant, 2019-03-05:
I have build raspiblitz v1.0, building my onw SD card as per installation instructions but after two days trying I am still unable to make small (15 satoshis) payments, lncli payinvoice just hangs forever.

I had no problems funding raspiblitz, opening payment channels manually, receiving network information through gossip and seeing my node on 1ML.

I am routing through tor however, are there anyone out there with the same problem?

AlverGant, 2019-03-05:
In order to figure out the problem, I reconfigured raspiblitz to disable tor and set up a Dyn DNS forwarding the relevant ports and rebooting the raspberry. Had to setup new channels and now I am able to make payments. It appears to me that something is wrong with tor routing. 
By the way even after disabling tor under service menu, it is still running "/usr/bin/tor --defaults-torrc /usr/share/tor/tor-service-defaults-torrc -f /etc/tor/torrc --RunAsDaemon 0"

rootzoll, 2019-03-05:
Thanks for the report. First of all I am adding a task ..

TODO: for v1.1 realease - check on the TOR disable

Normally making payments when the node is running on TOR should be possible. That the LND request "just hangs" and is not giving any kind of error message is not a good experience. Please feel free to let the people in LND Dev Slack know - so that can be improved: https://dev.lightning.community

nyxnor, 2021-04-16:
close this pls.

closed 2021-04-16 by rootzoll