raspiblitz/raspiblitz #404: Never got mnemonic seed... can I redo that part and get a new one?

malixg, 2019-03-05:
I got to the point in the installation where I typed in my Password D, and it said to write down the seed phrase, but I never saw one and the installation just continued. Do I need to completely start from scratch or is there a way to just redo that part and get a new seed? Thanks

jay653, 2019-03-05:
I had to scroll up a little in the terminal window on my laptop to get my seed.

malixg, 2019-03-05:
It's too late for that

cryptomulde, 2019-03-05:
Check this: https://github.com/rootzoll/raspiblitz/blob/master/home.admin/70initLND.sh

Try 
`./70initLND.sh`

malixg, 2019-03-05:
when can I type this in?

cryptomulde, 2019-03-05:
After establishing ssh connection to RPI type CTRL + C to cancel and be able to input commands yourself.

malixg, 2019-03-05:
I tried this and it went through some scripts, asked me for password C again, and went back to synching node, never prompting me to recreate password D or giving me my seen. https://pastebin.com/XzmBQk3G here is the output from the command.

cryptomulde, 2019-03-05:
you could also try creating a new one by typing 'lncli create' if testnet you have to add --network=testnet 'lncli --network=testnet create'

malixg, 2019-03-05:
that will let me create a new seed?

cryptomulde, 2019-03-05:
yes

rootzoll, 2019-03-05:
Thanks. This is the second report that the installation did somehow skipped parts of the wallet creation. My take away to improve is ...

TODO: For v1.1 implement the wallet creation script in python, do checks that wallet was really created and make single UI steps more clearly.

malixg, 2019-03-05:
Close but still not successful. 

~ $ lncli create
Input wallet password:
Confirm wallet password:

Do you have an existing cipher seed mnemonic you want to use? (Enter y/n): n

Your cipher seed can optionally be encrypted.
Input your passphrase if you wish to encrypt it (or press enter to proceed without a cipher seed passphrase):
Confirm cipher seed passphrase:

Generating fresh cipher seed...

[lncli] unable to generate seed: rpc error: code = Unimplemented desc = unknown service lnrpc.WalletUnlocker

malixg, 2019-03-05:
I restarted pi and tried again. Got this error now:
[lncli] unable to generate seed: rpc error: code = Unknown desc = wallet already exists

Is there a way to delete wallet so I can create a new one?

rootzoll, 2019-03-05:
> [lncli] unable to generate seed: rpc error: code = Unimplemented desc = unknown service lnrpc.WalletUnlocker

Thats unusual ... first time this got reported. 

@malixg can you head over to LND Dev Slack and get details on that error and report back? https://dev.lightning.community

rootzoll, 2019-03-05:
@malixg check the directory `/mnt/hdd/lnd` to delete the lnd data

malixg, 2019-03-05:
i see data, lnd.conf, logs, tls.cert, tls.key

just delete it all?

rootzoll, 2019-03-05:
you can be bold and delete all of it - as long as you never funded your wallet, you should be safe todo so. After delete try running `./70initLND.sh` again from your `/home/admin` directory.

malixg, 2019-03-07:
ok, I deleted everything in /mnt/hdd/lnd and then did lncli create, and successfully got a seed. Then I ran ./70initLND.sh and my the Block Chain synced. I made it to the main menu. Thank you

rootzoll, 2019-03-13:
TODO: implement the wallet creation script in python, do checks that wallet was really created and make single UI steps more clearly.

cricktor, 2021-06-25:
Does the LND seed follow BIP-39 rules?

I remember me writing down very carefully the seed of the LND wallet of my RaspiBlitz 1.7.0 as I'm aware of the impartance of a seed mnemonic words and order. I funded my LND wallet with some amount. Fortunately I had the idea to check with my Electrum wallet if I can restore the LND wallet with Electrum in case that's needed for some emergancy.
Dooh, I was puzzled to see that my LND seed fails the BIP-39 checksum in Electrum, but also fails with an offline script from iancoleman.io. OK, it may be my fault, but I can't explain it to myself. To be safe I defunded the LND wallet until I can sort this out (need to figure out if the seed is hidden in some way in the LND wallet, of course likely in some encrypted way; I'd really like to find out what kind of transcript error I made). Any hints for that?

rootzoll, 2021-07-01:
@cricktor see https://github.com/lightningnetwork/lnd/issues/2735