raspiblitz/raspiblitz #405: double input of first character in CLI [Not Critical]

malixg, 2019-03-05:
When I type in the FIRST character into the CLI, that character appears twice. If I ignore that it's there twice. the program continues as though it was only there once. This happened when I named my node - I typed the name, and noticed the first character was there twice, so I deleted the extra character. Then later I noticed my node was missing the first character in its name. In CLI I typed "help" but it came out as "hhelp". When I try to delete all characters, without pressing ENTER, the extra character can not be deleted. This doesn't affect anything else as far as I can tell.

rootzoll, 2019-03-05:
To get a clear picture: That is happening on the terminal when using `lncli` or is that happening when you are in the raspiblitz menus/dialogs?

nyxnor, 2021-04-18:
please close this. Did not notice this as of today.

closed 2021-04-19 by rootzoll