raspiblitz/raspiblitz #406: 2 nodes on one network

Elmit2015, 2019-03-06:
On my desktop I am running a node, so it uses the ports already on my internal network.
Adding Rapiblitz needs now to use another internal IP (same external IP) but with different ports.
How can I set up this? 

My external IP 123.123.123.123
My internal IP 192.168.1.12 (desktop)
My internal IP 192.168.1.15 (raspibitz)

vamp111, 2019-03-06:
On your desktop the BTC node uses port 8333, LN uses 9735, same public IP, but on different local IP, I have the same thing at home, not an issue at all. However someone else can elaborate

fluidvoice, 2019-03-06:
See the bitcoind and LND docs (and/or use --help command line parameter) to see how to start them listening on different ports.

rootzoll, 2019-03-06:
About setting alternative ports for LND see here if needed: https://github.com/rootzoll/raspiblitz/issues/100#issuecomment-465997126

closed 2019-03-13 by rootzoll