raspiblitz/raspiblitz #407: Progress freezes at "Lightning Filtering Blockchain"

MerlinMason, 2019-03-07:
During setup, after the blockchain sync is complete progress seems to freeze at "Lightning Filtering Blockchain".

[image]

I've tried turning it off and on again, it re-syncs but gets stuck again at the same point.

I also noticed an error when trying to run the `getinfo` command, not sure if that's expected at this stage of setup?

```
admin@RaspiBlitz:~ $ lncli getinfo
[lncli] open /home/admin/.lnd/tls.cert: no such file or directory
```

rootzoll, 2019-03-07:
Just for everyone reading this ... some people reported, that a restart solved this:
SSH into the RaspiBlitz - if you see the screen above press `CRTL+C` to get to the terminal and run the command to restart: `sudo shutdown -r now`.

@MerlinMason you already tried the restart and you are still stuck. So we need to get some more debug information:
SSH into the RaspiBlitz - if you see the screen above press `CRTL+C` to get to the terminal and run the command to restart: `/home/admin/XXdebugLogs.sh`
That will output some logs ... please copy and paste those logs to this issue, so that we can take a look.

MerlinMason, 2019-03-08:
Thanks for your help, here's the output of my logs:

```
*** RASPIBLITZ LOGS ***
blitzversion: 1.0
chainnetwork: bitcoin / main
 17:20:58 up 11 min,  2 users,  load average: 2.03, 2.29, 1.40

*** CHAINNETWORK SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2019-03-08 17:09:34 GMT; 11min ago
  Process: 686 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
 Main PID: 695 (bitcoind)
   CGroup: /system.slice/bitcoind.service
           └─695 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid

Mar 08 17:09:33 RaspiBlitz systemd[1]: Starting Bitcoin daemon...
Mar 08 17:09:34 RaspiBlitz systemd[1]: Started Bitcoin daemon.

*** LAST 20 CHAINNETWORK LOGS ***
2019-03-08T17:11:31Z opencon thread start
2019-03-08T17:11:31Z init message: Done loading
2019-03-08T17:11:31Z net thread start
2019-03-08T17:11:31Z msghand thread start
2019-03-08T17:11:37Z New outbound peer connected: version: 70015, blocks=566208, peer=0
2019-03-08T17:11:38Z New outbound peer connected: version: 70015, blocks=566208, peer=1
2019-03-08T17:11:39Z txindex is enabled at height 566208
2019-03-08T17:11:39Z txindex thread exit
2019-03-08T17:11:42Z P2P peers available. Skipped DNS seeding.
2019-03-08T17:11:42Z dnsseed thread exit
2019-03-08T17:11:46Z New outbound peer connected: version: 70015, blocks=566208, peer=2
2019-03-08T17:11:47Z New outbound peer connected: version: 70015, blocks=566208, peer=3
2019-03-08T17:11:48Z New outbound peer connected: version: 70015, blocks=566208, peer=4
2019-03-08T17:11:50Z New outbound peer connected: version: 70015, blocks=566208, peer=5
2019-03-08T17:11:50Z New outbound peer connected: version: 70015, blocks=566208, peer=6
2019-03-08T17:11:56Z New outbound peer connected: version: 70015, blocks=566208, peer=7
2019-03-08T17:12:25Z New outbound peer connected: version: 70015, blocks=566208, peer=8
2019-03-08T17:18:08Z New outbound peer connected: version: 70015, blocks=566208, peer=10
2019-03-08T17:19:28Z UpdateTip: new best=00000000000000000002bb1c0ee088f228761be2d01960c2061cec3647c0b650 height=566209 version=0x20400000 log2_work=90.420817 tx=390126094 date='2019-03-08T17:18:42Z' progress=1.000000 cache=3.7MiB(32991txo) warning='35 of last 100 blocks have unexpected version'
2019-03-08T17:19:32Z Imported mempool transactions from disk: 8029 succeeded, 4 failed, 0 expired, 19 already there

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2019-03-08 17:09:34 GMT; 11min ago
 Main PID: 696 (lnd)
   CGroup: /system.slice/lnd.service
           └─696 /usr/local/bin/lnd --externalip=83.244.179.2

Mar 08 17:09:35 RaspiBlitz lnd[696]: 2019-03-08 17:09:35.404 [INF] RPCS: password gRPC proxy started at [::]:8080
Mar 08 17:09:35 RaspiBlitz lnd[696]: 2019-03-08 17:09:35.405 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of…g wallet and unlock it.
Hint: Some lines were ellipsized, use -l to show in full.

*** LAST 20 LND LOGS ***
-- Logs begin at Fri 2019-03-08 17:09:04 GMT, end at Fri 2019-03-08 17:20:58 GMT. --
Mar 08 17:09:34 RaspiBlitz systemd[1]: Started LND Lightning Daemon.
Mar 08 17:09:35 RaspiBlitz lnd[696]: 2019-03-08 17:09:35.363 [INF] LTND: Version: 0.5.2-beta commit=v0.5.2-beta, build=production, logging=default
Mar 08 17:09:35 RaspiBlitz lnd[696]: 2019-03-08 17:09:35.364 [INF] LTND: Active chain: Bitcoin (network=mainnet)
Mar 08 17:09:35 RaspiBlitz lnd[696]: 2019-03-08 17:09:35.396 [INF] CHDB: Checking for schema update: latest_version=7, db_version=7
Mar 08 17:09:35 RaspiBlitz lnd[696]: 2019-03-08 17:09:35.404 [INF] RPCS: password RPC server listening on [::]:10009
Mar 08 17:09:35 RaspiBlitz lnd[696]: 2019-03-08 17:09:35.404 [INF] RPCS: password gRPC proxy started at [::]:8080
Mar 08 17:09:35 RaspiBlitz lnd[696]: 2019-03-08 17:09:35.405 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.

- RTL is OFF by config
```

fluidvoice, 2019-03-08:
from an SSH shell terminal try running this:
`./AAunlockLND.sh`

osilloh, 2019-03-09:
I have the same issue: so I did a  sudo shutdown -r now. Still stuck.

[image]

here is my log:

admin@RaspiBlitz:~ $ /home/admin/XXdebugLogs.sh

*** RASPIBLITZ LOGS ***
blitzversion: 1.0
chainnetwork: bitcoin / main
 20:48:31 up 2 min,  2 users,  load average: 1.55, 0.75, 0.29

*** CHAINNETWORK SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2019-03-09 20:46:01 GMT; 2min 30s ago
  Process: 706 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
 Main PID: 714 (bitcoind)
   CGroup: /system.slice/bitcoind.service
           └─714 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=…id

Mar 09 20:46:00 RaspiBlitz systemd[1]: Starting Bitcoin daemon...
Mar 09 20:46:01 RaspiBlitz systemd[1]: Started Bitcoin daemon.

*** LAST 20 CHAINNETWORK LOGS ***
2019-03-09T20:47:35Z init message: Starting network threads...
2019-03-09T20:47:35Z net thread start
2019-03-09T20:47:35Z addcon thread start
2019-03-09T20:47:35Z opencon thread start
2019-03-09T20:47:35Z msghand thread start
2019-03-09T20:47:35Z init message: Done loading
2019-03-09T20:47:35Z dnsseed thread start
2019-03-09T20:47:35Z New outbound peer connected: version: 70015, blocks=566369, peer=0
2019-03-09T20:47:37Z New outbound peer connected: version: 70015, blocks=566369, peer=1
2019-03-09T20:47:38Z New outbound peer connected: version: 70015, blocks=566369, peer=2
2019-03-09T20:47:44Z New outbound peer connected: version: 70015, blocks=566369, peer=3
2019-03-09T20:47:46Z P2P peers available. Skipped DNS seeding.
2019-03-09T20:47:46Z dnsseed thread exit
2019-03-09T20:47:52Z Imported mempool transactions from disk: 1191 succeeded, 0 failed, 0 expired, 0 already there
2019-03-09T20:47:52Z New outbound peer connected: version: 70015, blocks=566369, peer=4
2019-03-09T20:47:54Z New outbound peer connected: version: 70015, blocks=566369, peer=6
2019-03-09T20:47:54Z New outbound peer connected: version: 70015, blocks=566369, peer=7
2019-03-09T20:47:54Z New outbound peer connected: version: 70015, blocks=566369, peer=5
2019-03-09T20:47:55Z txindex is enabled at height 566369
2019-03-09T20:47:55Z txindex thread exit

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2019-03-09 20:46:01 GMT; 2min 30s ago
 Main PID: 715 (lnd)
   CGroup: /system.slice/lnd.service
           └─715 /usr/local/bin/lnd --externalip=xxx.xxx.xxx.xxx <redacted>

Mar 09 20:46:02 RaspiBlitz lnd[715]: 2019-03-09 20:46:02.337 [INF] RPCS: password gRPC proxy…]:8080
Mar 09 20:46:02 RaspiBlitz lnd[715]: 2019-03-09 20:46:02.338 [INF] LTND: Waiting for wallet …ck it.
Hint: Some lines were ellipsized, use -l to show in full.

*** LAST 20 LND LOGS ***
-- Logs begin at Sat 2019-03-09 20:45:43 GMT, end at Sat 2019-03-09 20:48:31 GMT. --
Mar 09 20:46:01 RaspiBlitz systemd[1]: Started LND Lightning Daemon.
Mar 09 20:46:02 RaspiBlitz lnd[715]: 2019-03-09 20:46:02.258 [INF] LTND: Version: 0.5.2-beta commit=v0.5.2-beta, build=production, logging=default
Mar 09 20:46:02 RaspiBlitz lnd[715]: 2019-03-09 20:46:02.265 [INF] LTND: Active chain: Bitcoin (network=mainnet)
Mar 09 20:46:02 RaspiBlitz lnd[715]: 2019-03-09 20:46:02.321 [INF] CHDB: Checking for schema update: latest_version=7, db_version=7
Mar 09 20:46:02 RaspiBlitz lnd[715]: 2019-03-09 20:46:02.331 [INF] RPCS: password RPC server listening on [::]:10009
Mar 09 20:46:02 RaspiBlitz lnd[715]: 2019-03-09 20:46:02.337 [INF] RPCS: password gRPC proxy started at [::]:8080
Mar 09 20:46:02 RaspiBlitz lnd[715]: 2019-03-09 20:46:02.338 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.

- RTL is OFF by config
admin@RaspiBlitz:~ $ 

================

then ran script: 

 ./AAunlockLND.sh

which tells me the wallet needs to be created.

****************************************************************************
Unlock LND Wallet --> lncli --chain=bitcoin unlock
****************************************************************************
HELP: Enter your PASSWORD C
You may wait some seconds until you get asked for password.
****************************************************************************
Input wallet password: 
[lncli] rpc error: code = Unknown desc = wallet not found

network(bitcoin) chain(main)
2019-03-09 20:46:02.338 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.
Wallet still locked - please try again or Cancel with CTRL+C
Input wallet password: 
[lncli] rpc error: code = Unknown desc = wallet not found

admin@RaspiBlitz:~ $ lncli create
[lncli] open /home/admin/.lnd/tls.cert: no such file or directory

I do not see a /home/admin/.lnd directory did I miss a step somewhere?

thank you.

osilloh, 2019-03-09:
I took a chance and ran 70initLND.sh, created my wallet. rebooted, now its continuing!

[image]

osilloh, 2019-03-10:
I think I got it, thanks :) 

[image]

codeoholic, 2019-03-10:
Mine is also stuck.

Here are the logs.

*** RASPIBLITZ LOGS ***
blitzversion: 1.0
chainnetwork: bitcoin / main
 01:29:04 up  1:12,  2 users,  load average: 0.68, 0.82, 1.39

*** CHAINNETWORK SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2019-03-10 00:17:26 GMT; 1h 11min ago
  Process: 752 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
 Main PID: 761 (bitcoind)
   CGroup: /system.slice/bitcoind.service
           └─761 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid

Mar 10 00:17:26 codeoholic systemd[1]: Starting Bitcoin daemon...
Mar 10 00:17:26 codeoholic systemd[1]: Started Bitcoin daemon.

*** LAST 20 CHAINNETWORK LOGS ***
2019-03-10T01:06:41Z UpdateTip: new best=0000000000000000000d733b630b8b122c3ecb81905c6c39098f57894163c438 height=566398 version=0x3fffe000 log2_work=90.425101 tx=390466068 date='2019-03-10T01:39:59Z' progress=1.000012 cache=74.1MiB(660264txo) warning='43 of last 100 blocks have unexpected version'
2019-03-10T01:06:50Z UpdateTip: new best=0000000000000000002e29f3541291bb26b72bd2cbe98773e48111a417c21076 height=566399 version=0x20c00000 log2_work=90.425124 tx=390469292 date='2019-03-10T02:07:20Z' progress=1.000022 cache=74.6MiB(664442txo) warning='44 of last 100 blocks have unexpected version'
2019-03-10T01:07:10Z UpdateTip: new best=00000000000000000004f41ff706827e736479a4863ae3a5454ed7d02e0aacc6 height=566400 version=0x20400000 log2_work=90.425146 tx=390471263 date='2019-03-10T02:18:44Z' progress=1.000026 cache=75.1MiB(669567txo) warning='45 of last 100 blocks have unexpected version'
2019-03-10T01:07:11Z New outbound peer connected: version: 70015, blocks=566403, peer=201
2019-03-10T01:07:20Z UpdateTip: new best=00000000000000000028754ed70f6a7987d6aaabaac25f578be78a92ba83132d height=566401 version=0x20000000 log2_work=90.425169 tx=390473057 date='2019-03-10T02:33:23Z' progress=1.000032 cache=75.3MiB(671743txo) warning='45 of last 100 blocks have unexpected version'
2019-03-10T01:07:20Z UpdateTip: new best=0000000000000000001faed98f887e439f32b79885fc5271afed1d0889f79c6e height=566402 version=0x20800000 log2_work=90.425192 tx=390473277 date='2019-03-10T02:34:31Z' progress=1.000032 cache=75.3MiB(671853txo) warning='45 of last 100 blocks have unexpected version'
2019-03-10T01:07:21Z New outbound peer connected: version: 70015, blocks=566403, peer=203
2019-03-10T01:07:23Z UpdateTip: new best=0000000000000000000966247eaf729c96ca1c3e38e416510a5d481bbc6732bf height=566403 version=0x20800000 log2_work=90.425214 tx=390473750 date='2019-03-10T02:37:16Z' progress=1.000033 cache=75.4MiB(672419txo) warning='45 of last 100 blocks have unexpected version'
2019-03-10T01:07:23Z New outbound peer connected: version: 70015, blocks=566403, peer=206
2019-03-10T01:07:27Z socket recv error Connection reset by peer (104)
2019-03-10T01:07:53Z Imported mempool transactions from disk: 11131 succeeded, 1 failed, 0 expired, 0 already there
2019-03-10T01:10:45Z UpdateTip: new best=00000000000000000020a74a553358e9251746b0e773d01f68c7d5839b852d4e height=566404 version=0x3fff0000 log2_work=90.425237 tx=390475156 date='2019-03-10T02:47:14Z' progress=1.000036 cache=75.6MiB(674113txo) warning='46 of last 100 blocks have unexpected version'
2019-03-10T01:13:29Z UpdateTip: new best=00000000000000000000e701f8b711ae9456f59bf8e2ff754a8b272d68ed676c height=566405 version=0x20000000 log2_work=90.42526 tx=390475415 date='2019-03-10T02:48:57Z' progress=1.000035 cache=75.6MiB(674418txo) warning='45 of last 100 blocks have unexpected version'
2019-03-10T01:13:52Z socket recv error Connection reset by peer (104)
2019-03-10T01:25:58Z socket recv error Connection reset by peer (104)
2019-03-10T01:26:51Z UpdateTip: new best=00000000000000000012585610106dca4df323aecff5023c3a163af0fd9d82d3 height=566406 version=0x20000000 log2_work=90.425282 tx=390477248 date='2019-03-10T03:02:46Z' progress=1.000035 cache=77.1MiB(687772txo) warning='45 of last 100 blocks have unexpected version'
2019-03-10T01:27:21Z ping timeout: 1200.016128s
2019-03-10T01:27:47Z socket send error Broken pipe (32)
2019-03-10T01:28:09Z Pre-allocating up to position 0xb00000 in rev01555.dat
2019-03-10T01:28:09Z UpdateTip: new best=0000000000000000000cbc8446f2c0455e4ff5625b51158331d751f1ab63c7c0 height=566407 version=0x20800000 log2_work=90.425305 tx=390477527 date='2019-03-10T03:04:41Z' progress=1.000036 cache=77.1MiB(688304txo) warning='45 of last 100 blocks have unexpected version'

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2019-03-10 00:22:27 GMT; 1h 6min ago
 Main PID: 5752 (lnd)
   CGroup: /system.slice/lnd.service
           └─5752 /usr/local/bin/lnd --externalip=61.246.39.159

Mar 10 01:06:50 codeoholic lnd[5752]: 2019-03-10 01:06:50.843 [DBG] LNWL: Possible reorg at block 0000000000000000000d733b630b8b…894163c438
Mar 10 01:09:32 codeoholic lnd[5752]: 2019-03-10 01:09:32.729 [DBG] LNWL: Disconnecting block 566223 (000000000000000000162c6d86…10b89e537)
Hint: Some lines were ellipsized, use -l to show in full.

*** LAST 20 LND LOGS ***
[... cut ...]
Mar 10 00:38:02 codeoholic lnd[5752]: 2019-03-10 00:38:02.342 [INF] RPCS: RPC server listening on [::]:10009
Mar 10 00:38:26 codeoholic lnd[5752]: 2019-03-10 00:38:26.881 [INF] LTND: Waiting for chain backend to finish sync, start_height=566252
Mar 10 01:06:50 codeoholic lnd[5752]: 2019-03-10 01:06:50.843 [DBG] LNWL: Possible reorg at block 0000000000000000000d733b630b8b122c3ecb81905c6c39098f57894163c438
Mar 10 01:09:32 codeoholic lnd[5752]: 2019-03-10 01:09:32.729 [DBG] LNWL: Disconnecting block 566223 (000000000000000000162c6d8635fa59b4cf6db57c127711cca61fd10b89e537)

- RTL is OFF by config

fluidvoice, 2019-03-10:
> On reboot. I see this.
> 
> [image]
> [image]

Your config files are not right for some reason.  They are not indicating **main**net.
what are the contents...
`cat /home/admin/raspiblitz.info`
`cat /mnt/hdd/raspiblitz.conf`

MerlinMason, 2019-03-11:
I also tried running `./70initLND.sh` and all is looking good :)

codeoholic, 2019-03-12:
Out of no where, I checked on the disk space. Came out it was full of logs, that's why it was unable to sync the chain.

fluidvoice, 2019-03-12:
> Out of no where, I checked on the disk space. Came out it was full of logs, that's why it was unable to sync the chain.

until v1.1 comes out the fix for this is change the logrotate config documented here: https://github.com/rootzoll/raspiblitz/issues/394#issuecomment-471760212

fluidvoice, 2019-03-12:
@rootzoll this problem is hitting enough people, and breaking their nodes, that it would prob be good to push out this one fix for the logrotate problem.

rootzoll, 2019-03-12:
I will start to work on a fix release tomorrow and hopefully have it ready before the weekend.

rootzoll, 2019-03-14:
I have done the following measures to fix this issue for the v1.1 release ... is that enough?

1. LND logging level is set to `info`

2. Added config to `/etc/init.d/rsyslog`:
```
/var/log/daemon.log
{
        rotate 4
        size=100M
        missingok
        notifempty
        compress
        delaycompress
        sharedscripts
        postrotate
                invoke-rc.d rsyslog rotate > /dev/null
        endscript
}

3. If on boot log dir is detected to be >1GB it all gets emergency deleted:
`sudo rm -r /var/log/*`
```

So v1.1 will reduce the log load, will keep daemon.log small and has a safe guard on boot to flush the logs if they get out of control again. Please let me know if you see more things to do on this.

This is now in testing and will be released as a new image end of week for everybody to easy update.

rootzoll, 2019-03-17:
Looks good so far ... closing the issue for v1.1 release.

closed 2019-03-17 by rootzoll

walletofsatoshi, 2019-04-15:
The issue is still there. Happened to me due to power failure and when I restarted got stuck at this point. Then ran the script ./70initLND.sh which seems to fix it .

bereska, 2019-04-21:
I had the same problem the other night on 1.1 release. Had to run ./70initLND.sh which fixed it. It took some time to restore channels

fluidvoice, 2019-04-21:
It would help if you provided logging rather than just reporting you saw the same screen output.  This screen shows this when LND is not running correctly - which can happen for any number of reasons.

bereska, 2019-04-21:
You are right. My bad. Too late for logs. I must mention I am running BTCPay server on my Raspiblitz as well. It might be too much for rpi 3b+ to pull it all off sometimes. I am thinking to further optimizing bitcoind to use less memory as it's the hungriest process. Right now it looks better after ./70initLND.sh reset:
admin@bereskaLND:~ $ sudo free -h
              total        used        free      shared  buff/cache   available
Mem:           927M        512M         37M         26M        377M        333M
Swap:          1.0G        185M        838M
before it was about 37M free and 120M available
I want to configure bitcoind like this:
-blocksonly -dbcache=20 -maxsigcachesize=4 -maxconnections=4 -rpcthreads=1
what do you think?

fluidvoice, 2019-04-21:
I think over time LND ends up using the most memory but I'm not sure. 
Note: do not enter your problem in an existing issue. Open your own issue.  But... I'm not sure you will get a lot of help here about BTCPayServer, you should open an issue with them/that project.

bereska, 2019-04-21:
It was not btcpayserver that froze, but raspiblitz. Anyways, thanks for your feedback

fluidvoice, 2019-04-22:
I'm not sure it matters.  Again, you'll get better results if you open your own issue and give more details than "I had the same problem".  Next time.  cheers.

rootzoll, 2019-04-28:
Version v1.2 release tries to improve on this part in the setup process. If someone still experiences problems at this point of setup - please report by opening a new issue.

bereska, 2019-04-29:
@rootzoll, thank you for the Raspiblitz. I have a BTCPay Server running with it on the rpi thanks to this great guide https://gist.github.com/normandmickey/6d6e74ebb6b3ddbd317bc1450f48f08e. I think it is so far the most affordable way to host your own bitcoin node, lightning node and btcpayserver 3 in 1.
Than you, mates

rootzoll, 2019-04-30:
@bereska addinf BTCPay server as a service option is part of this issue - hope we will get to it in a future release :) https://github.com/rootzoll/raspiblitz/issues/214

bereska, 2019-04-30:
@rootzoll, I hope BTCPay Server is not a problem. The hungriest processes are bitcoind and lnd of course. I will watch this set-up and report any issues. In the meantime I was wondering if bitcoind could be optimized even more with the following after the full network sync without breaking anything in Raspiblitz: -blocksonly -dbcache=20 -maxsigcachesize=4 -maxconnections=4 -rpcthreads=1
Any thoughts on lnd optimization for rpi as well?

fluidvoice, 2019-04-30:
> Any thoughts on lnd optimization for rpi as well?

The LND and/or c-lighting peeps could prob tell us more.  Maybe open an info-request issue with them.

bereska, 2019-05-01:
fyi 
https://github.com/lightningnetwork/lnd/issues/3030

fluidvoice, 2019-05-01:
> fyi
> [lightningnetwork/lnd#3030](https://github.com/lightningnetwork/lnd/issues/3030)

@roasbeef responded and closed:

> There's nothing atm, however if you upgrade to lnd 0.6-beta, then you'll find that there've been a number of significant performance improvements (both memory+CPU). 0.6.1 includes even more optimizations as well.

rootzoll, 2019-05-13:
@bereska thanks for the recommendation to optimize bitcoind. I added it to try it in a future release. See issue: https://github.com/rootzoll/raspiblitz/issues/603

bereska, 2019-05-13:
@rootzoll no problem, mate. Thank you for what you're doing. Gonna upgrade to bitcoin 0.18 tonight to see if these configs play well

bereska, 2019-05-13:
BTW, upgrading to lnd 0.6-beta did lower the memory and cpu usage. So my little rpi is still pulling it off with BTCPay Server onboard

yahooshua, 2019-05-19:
> 
> 
> I also tried running `./70initLND.sh` and all is looking good :)

This resolved the issue for me.
Thank you all who helped!

RiccardoMasutti, 2019-06-24:
@bereska @fluidvoice @rootzoll Probably the new Raspberry Pi 4 solves the problem:
https://techcrunch.com/2019/06/23/the-raspberry-pi-foundation-unveils-the-raspberry-pi-4/

bereska, 2019-06-24:
@RiccardoMasutti, yes, for sure, can't wait to test one)
for those of you who still has this problem from time to time here's one more remedy that works for me:

sudo systemctl restart lnd