raspiblitz/raspiblitz #409: Unable to fund wallet (rpc error: code = Unknown desc = verification failed: signature mismatch after caveat verification)

ctrlF4runner, 2019-03-08:
I'm having an issue where I'm unable to generate an address to fund the wallet. I turned on the RTL WebInterface (not sure if this would do anything), but I've turned it off and am still having the same issue. 

I ran the lncli getinfo command but I'm getting te same rpc error below.

```
******************************
Fund your Blockchain Wallet
******************************

COMMAND LINE:
lncli --chain=bitcoin --network=mainnet newaddress np2wkh

RESULT:
[lncli] rpc error: code = Unknown desc = verification failed: signature mismatch after caveat verification
FAIL PRECHECK - lncli getinfo shows 'synced_to_chain': false - wait until chain is sync

******************************
TODO
******************************
Send REAL Bitcoin to address -->
Whats next? --> Wait for confirmations. You can use info on LCD to check if funds have arrived.
If you want your lighting node to open channels automatically, activate the 'Autopilot' under 'Activate/Deactivate Services'

Press ENTER to return to main menu.
```
```
*** RASPIBLITZ LOGS ***
blitzversion: 1.0
chainnetwork: bitcoin / main
 03:51:32 up 11 min,  3 users,  load average: 0.34, 1.09, 0.87

*** CHAINNETWORK SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2019-03-08 03:40:07 GMT; 11min ago
  Process: 748 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
 Main PID: 756 (bitcoind)
   CGroup: /system.slice/bitcoind.service
           └─756 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoi…id

Mar 08 03:40:07 LiteNode systemd[1]: Starting Bitcoin daemon...
Mar 08 03:40:07 LiteNode systemd[1]: Started Bitcoin daemon.

*** LAST 20 CHAINNETWORK LOGS ***
2019-03-08T03:41:20Z net thread start
2019-03-08T03:41:20Z dnsseed thread start
2019-03-08T03:41:20Z addcon thread start
2019-03-08T03:41:20Z msghand thread start
2019-03-08T03:41:20Z opencon thread start
2019-03-08T03:41:20Z init message: Done loading
2019-03-08T03:41:24Z txindex is enabled at height 566138
2019-03-08T03:41:24Z txindex thread exit
2019-03-08T03:41:26Z Imported mempool transactions from disk: 141 succeeded, 0 failed, 0 expired, 0 already there
2019-03-08T03:41:26Z New outbound peer connected: version: 70015, blocks=566138, peer=0
2019-03-08T03:41:27Z New outbound peer connected: version: 70015, blocks=566138, peer=1
2019-03-08T03:41:31Z P2P peers available. Skipped DNS seeding.
2019-03-08T03:41:31Z dnsseed thread exit
2019-03-08T03:41:35Z New outbound peer connected: version: 70015, blocks=566138, peer=2
2019-03-08T03:41:35Z New outbound peer connected: version: 70015, blocks=566137, peer=3
2019-03-08T03:41:41Z New outbound peer connected: version: 70015, blocks=566138, peer=4
2019-03-08T03:41:42Z New outbound peer connected: version: 70015, blocks=566138, peer=5
2019-03-08T03:41:43Z New outbound peer connected: version: 70015, blocks=566138, peer=6
2019-03-08T03:42:00Z New outbound peer connected: version: 70015, blocks=566138, peer=7
2019-03-08T03:49:29Z socket recv error Connection reset by peer (104)

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2019-03-08 03:40:07 GMT; 11min ago
 Main PID: 757 (lnd)
[... 7 lines cut ...]
*** LAST 20 LND LOGS ***
-- Logs begin at Fri 2019-03-08 03:39:39 GMT, end at Fri 2019-03-08 03:51:32 GMT. --
Mar 08 03:51:32 LiteNode lnd[757]: 2019-03-08 03:51:31.981 [DBG] PEER: Received ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=620822748040462336, mflags=00000001, cflags=00000010, update_time=2019-03-08 03:50:51 +0000 GMT) from 66.44.56.44:9735
Mar 08 03:51:32 LiteNode lnd[757]: 2019-03-08 03:51:31.981 [DBG] PEER: Received ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=620028900609949697, mflags=00000001, cflags=00000001, update_time=2019-03-08 03:50:43 +0000 GMT) from 66.44.56.44:9735
Mar 08 03:51:32 LiteNode lnd[757]: 2019-03-08 03:51:31.982 [DBG] PEER: Received ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=622321382331252736, mflags=00000001, cflags=00000010, update_time=2019-03-08 03:50:58 +0000 GMT) from 66.44.56.44:9735
Mar 08 03:51:32 LiteNode lnd[757]: 2019-03-08 03:51:32.024 [DBG] PEER: Received ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=609845223975026688, mflags=00000001, cflags=00000000, update_time=2019-03-08 03:50:56 +0000 GMT) from 66.44.56.44:9735
Mar 08 03:51:32 LiteNode lnd[757]: 2019-03-08 03:51:32.025 [DBG] PEER: Received ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=615935418866270208, mflags=00000001, cflags=00000010, update_time=2019-03-08 03:49:56 +0000 GMT) from 66.44.56.44:9735
Mar 08 03:51:32 LiteNode lnd[757]: 2019-03-08 03:51:32.025 [DBG] PEER: Received ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=616298257716215808, mflags=00000001, cflags=00000011, update_time=2019-03-08 03:50:58 +0000 GMT) from 66.44.56.44:9735
Mar 08 03:51:32 LiteNode lnd[757]: 2019-03-08 03:51:32.026 [DBG] PEER: Received ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=622323581351690240, mflags=00000001, cflags=00000000, update_time=2019-03-08 03:51:02 +0000 GMT) from 66.44.56.44:9735
Mar 08 03:51:32 LiteNode lnd[757]: 2019-03-08 03:51:32.026 [DBG] PEER: Received ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=615619858987548673, mflags=00000001, cflags=00000000, update_time=2019-03-08 03:50:16 +0000 GMT) from 66.44.56.44:9735
Mar 08 03:51:32 LiteNode lnd[757]: 2019-03-08 03:51:32.026 [DBG] PEER: Received ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=620028900609884161, mflags=00000001, cflags=00000001, update_time=2019-03-08 03:50:43 +0000 GMT) from 66.44.56.44:9735
Mar 08 03:51:32 LiteNode lnd[757]: 2019-03-08 03:51:32.027 [DBG] PEER: Received ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=621620993494351873, mflags=00000001, cflags=00000000, update_time=2019-03-08 03:50:37 +0000 GMT) from 66.44.56.44:9735
Mar 08 03:51:32 LiteNode lnd[757]: 2019-03-08 03:51:32.027 [DBG] PEER: Received ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=620972281638879232, mflags=00000001, cflags=00000000, update_time=2019-03-08 03:51:03 +0000 GMT) from 66.44.56.44:9735
Mar 08 03:51:32 LiteNode lnd[757]: 2019-03-08 03:51:32.028 [DBG] PEER: Received ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=616638006828728321, mflags=00000001, cflags=00000000, update_time=2019-03-08 03:50:27 +0000 GMT) from 66.44.56.44:9735
Mar 08 03:51:32 LiteNode lnd[757]: 2019-03-08 03:51:32.028 [DBG] PEER: Received ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=620684209614880768, mflags=00000001, cflags=00000010, update_time=2019-03-08 03:50:34 +0000 GMT) from 66.44.56.44:9735
Mar 08 03:51:32 LiteNode lnd[757]: 2019-03-08 03:51:32.028 [DBG] PEER: Received ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=618906299227897856, mflags=00000001, cflags=00000001, update_time=2019-03-08 03:50:56 +0000 GMT) from 66.44.56.44:9735
Mar 08 03:51:32 LiteNode lnd[757]: 2019-03-08 03:51:32.029 [DBG] PEER: Received ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=621656177876402176, mflags=00000000, cflags=00000010, update_time=2019-03-08 03:51:00 +0000 GMT) from 66.44.56.44:9735
Mar 08 03:51:32 LiteNode lnd[757]: 2019-03-08 03:51:32.029 [DBG] PEER: Received ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=606898532778901504, mflags=00000001, cflags=00000000, update_time=2019-03-08 03:50:13 +0000 GMT) from 66.44.56.44:9735
[... cut ...]
Mar 08 03:51:32 LiteNode lnd[757]: 2019-03-08 03:51:32.037 [DBG] DISC: Got ChannelUpdate for edge not found in graph(shortChanID=620028900609949697), saving for reprocessing later
Mar 08 03:51:32 LiteNode lnd[757]: 2019-03-08 03:51:32.043 [DBG] DISC: Got ChannelUpdate for edge not found in graph(shortChanID=616298257716215808), saving for reprocessing later
Mar 08 03:51:32 LiteNode lnd[757]: 2019-03-08 03:51:32.045 [DBG] DISC: Got ChannelUpdate for edge not found in graph(shortChanID=620028900609884161), saving for reprocessing later

- RTL is OFF by config
```

rootzoll, 2019-03-08:
Hi there. Please check if the new FAQ chapter "How to solve a "signature mismatch after caveat verification" error?" helps you to fix this: https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#how-to-solve-a-signature-mismatch-after-caveat-verification-error

frennkie, 2020-05-30:
I just installed a new node (on `testnet`) and also got the error message:

```
signature mismatch after caveat verification 
```

The cause was that the macaroon's in `~/.lnd/` had not been updated (I assume when switching from `mainnet` to `testnet`. I was able to easily solve this from the new menu option `LNDCREDS` > `SYNC` 

Posting this here if other people encounter this.

@rootzoll I believe this issue can be closed..?

closed 2020-06-02 by rootzoll

6110110072, 2022-01-12:
I found this error on ubuntu how to fix it ?
"[lncli] rpc error: code = Unknown desc = verification failed: signature mismatch after caveat verification"