raspiblitz/raspiblitz #411: So far no instant payments possible

kangoala, 2019-03-08:
First of all many thanks to rootzoll for his brilliant support.
I finally reached to see my first incoming Sats. Very nice. :-) :-) :-)
However, sometimes the transactions takes quite some time and often they also get canceled.
I did most transactions by RTL Web-Interface and set up an Eclair-Wallet on my smart phone. On both I set up a channel to satoshis place and also to approx. three other peers. Anyway it was difficult to send payments to satoshis place or from the smart phone to the raspi blitz. Sometimes it took several minutes, but in most cases the transaction gets canceled due to routing problems or other issues. Also the channel status of the established channel is changing regularly from Active to Inactive and again back to Active.

Does anybody has an idea how to make the transaction process more reliable and more instant?
Probably this isn't a sole raspi-blitz problem but with that kind of performance LN would still need significant improvements.

fluidvoice, 2019-03-10:
you need to open channels with good/reliable/well-connected nodes...When you sent payment to Satoshis place did you already have a channel open with them?  It takes a while to open a channel. Only after that will subsequent payments be instant.

kangoala, 2019-03-10:
Also having a direct channel opened with SatoshisPlace, it took quite a while. Sometimes it wasn't even possible to pay by this direct channel. I don't know, maybe it is also a problem by SatoshisPlace.

Just a few days ago the channel with SatoshisPlace got forced closed (LOCAL_FORCE_CLOSE). I have no idea what caused this. I still have three other channels open. Sometimes I am lucky and a transaction get passed. But if so, I have to wait mostly 30 - 60 seconds. 

Can you recommend a reliable and well-connected node?

kangoala, 2019-03-14:
I'd like to set up a channel from my mobile eclair wallet with my raspiblitz. However, node and bitcoin id are displayed in red. I can't set up a channel from my eclair wallet. Has anybody an idea what can be the reason therefore?

rootzoll, 2019-03-19:
This issue is not on a specific RaspiBlitz issue - so I am closing it.

@kangoala about the red adresses: https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#why-is-my-bitcoin-ip-on-the-display-red

closed 2019-03-19 by rootzoll