raspiblitz/raspiblitz #412: Status Page doesn't show on LCD screen

pxsocs, 2019-03-08:
Everything seems to be working fine. I can SSH into the Raspberry Pi, blockchain was all synched. But the LCD screen attached to the pi (through HDMI) freezes as the attached image during reboot. 

I've tried attaching a keyboard to CTRL+C but it doesn't work. It's a minor issue since everything works fine through SSH and RPC but not sure why the Pi Display doesn't show the status page. 

[image]

Dropin16, 2019-03-08:
Did you read the README file in the LCD-show folder and followed the instructions?

rootzoll, 2019-03-08:
There is a longer thread trying to get behind that problem:
https://github.com/rootzoll/raspiblitz/issues/195

Last findings: That just the display gets stuck (some kind of broken pipe or driver bug) but the RaspiBlitz is cintinue to run in the background.

If this is persistent (same on next boot) and you haver time to experiment, you could try to deinstall and reinstall the LCD driver - see if that fixes it: see Waveshare LCD Hat documentation: https://www.waveshare.com/wiki/3.5inch_RPi_LCD_(A)

pxsocs, 2019-03-08:
Refered to issue #195

closed 2019-03-08 by pxsocs