raspiblitz/raspiblitz #413: Enhancement: Moving chainstate for improved speed

fluidvoice, 2019-03-08:
I found splitting the blockchain Data directory (moving the chainstate) as described in the link below **significantly** speeds up re-syncing the blockchain - when plugging in the HDD to my notebook and running bitcoind from there.  This requires moving the directory and creating a file link for that directory that points to a location on a solid-state drive. 
[https://en.bitcoin.it/wiki/Splitting_the_data_directory](https://en.bitcoin.it/wiki/Splitting_the_data_directory)
The chainstate directory is ~2.2GB in size.  **This begs the question if the Pi system performance in general would be improved if the chainstate dir location was made permanently on the SD card.**
Perhaps even better would be putting the SD card contents as well as the chainstate on a USB flash drive. Supposedly Pi's can be booted/run from USB as well as SD and there are various benefits to doing this.  Increased speed, reliability, easy backups, ability to test different versions by swapping USB drives, etc. (ie, my setup has a case with no access to the SD card without opening it up.)  
FURTHER RESEARCH NEEDED HERE.

fluidvoice, 2019-03-08:
For example the [32GB Samsung MUF-32AB/AM FIT Plus USB](https://www.amazon.com/Samsung-MUF-32AB-AM-Plus-32GB/dp/B07D7SX9NS/ref=dp_ob_title_ce) drive is US$10.  Yes it's USB 3.1 the Pi cannot take advantage of but it's good for burning and backing up. And I'm thinking is prob faster and more reliable than using a Pi SD card.

rootzoll, 2019-03-08:
Putting the chainestate onto a USB flash drive is an interesting idea. Maybe this fits in with this research ticket: 2 USB Sticks Raid for LND data storage -> https://github.com/rootzoll/raspiblitz/issues/329 - so when the switches to this 2-USB-Thumbdrives Raid, also the Chainstate directory gets moved.

I would like to keep the OS on the sd card - because thats the update strategy at the moment:  you just flash the new image to the sd card and you are ready to go. Having the OS on a USB would make that more complicated. But some research to better see what performance is to gain is always welcome.

fluidvoice, 2019-03-09:
I read some comments that the USB 2.0 bus will only handle 25-30MBps and so RAID won't speed things up over one good/fast USB drive.  All the USB ports are on the same bus, no?

fluidvoice, 2019-03-10:
Btw, I've found rampant fake USB's being sold all over, even on Amazon.  Here is how to see if connected USB drives are USB 2.0, 3.0, 3.1 etc.  One of mine made by Samsung actually was not from Samsung but some uknown Taiwan company!

`lsusb -vvv | grep -i -B5 -A5 bcdUSB`

openoms, 2021-11-08:
using SSD by default now

closed 2021-11-08 by openoms