raspiblitz/raspiblitz #414: Newly setup Raspblitz - easy way to setup solo mining on my own node with spare asic miners?

Luddite1337, 2019-03-09:
I really like the tutorial and setting up the Raspiblitz was easier than I thought!  Thank you.
I have a few s9 asic miners that I would like to have solo mine using my newly setup node.  (just for fun not expecting to mine a block)  How would I go about doing that or is it even possible?

rootzoll, 2019-03-09:
You really want to run the bitcoin full stack :) I am not exactly sure, but you should find on the google titorials about connection those asic miners with your full node. You have a bitcoind an a raspbian/debian linux in the local network .. sound not to exotic. Would appriciate if you share the resuts of your experiment.

Luddite1337, 2019-03-09:
I found this explanation but I am not sure how to do it with Raspiblitz?  I don't want to mess up my cool new node lol  Any advice or suggestions would be great!

"
Bitcoin Core doesn't directly allow for stratum miners to connect, you need a middleman proxy to translate the two mining methods used by Bitcoin Core and cgminer on the Antminer S3. This is simplest with bfgminer.

Run BFGminer on the same computer as Bitcoin Core with the following configuation, it won't need any further settings altered or configured (it gets the RPC password automatically on launch).

bfgminer --stratum-port=3333
BFGminer will launch, and retrieve the information needed for solo mining.

Now using coinbase address 1PkNBPBGEmvoMFbwRT5ky3XXVNstJnJ4c1, provided by pool 0
Network difficulty changed to 49.4G (353.6P)
Pool 0 http://localhost:8332/#getcbaddr#allblocks alive
Long-polling activated for http://localhost:8332/#getcbaddr#allblocks (getblocktemplate)
Added local bitcoin RPC server on port 8332 as pool 0
Probing for an alive pool
Pool 0 http://localhost:8332/#getcbaddr#allblocks alive  
                                                                        

You're now ready to solo mine against Bitcoin Core, you just have to point your antminer at the IP address of the computer running BFGminer, and the miner should show up in the top section of BFGminer with its hashrate showing as "PXY" (stratum proxy). Any blocks will be paid out to the coinbase address shown in BFGminer, which is a new one generated from your wallet.

shareimprove this answer
answered Jul 9 '15 at 23:46

Anonymous
8,88211028
This simply doesn't work -it makes sense that it would but something is missing. – user46824 Mar 2 '17 at 19:12
add a comment

0

**The previous method will work on Windows 10 as well but the missing thing is a stratum proxy. You must run a proxy for the stratum to connect with the rpc.**
"

rootzoll, 2019-03-10:
I think to give BFG miner a try is an idea. You may need to open port 3333 on the ufw firewall so that your miners cann connect to the BFG on your RaspiBlitz.

Xorell, 2019-12-11:
I've managed to solo mine by use of cgminer. I installed it as followed:

```
mkdir -p mining/newpac; cd mining/newpac
git clone https://github.com/vthoang/cgminer.git
cd cgminer

CFLAGS="-O2" ./autogen.sh --enable-gekko --enable-icarus
make -j 2
```

Once installed i run cgminer by used of the following command (in a screen session)

`sudo screen -S -d -m ./cgminer -o http://localhost:8332 -u raspibolt -p 7YIcI5Mt9bEF3KaTcFN2 --suggest-diff 90 --btc-address 15tf4HVVDy9kZVuNa6v1Mtv3z1vq7ZuyY8 --gekko-newpac-freq 400`

You can monitor the miner by entering `sudo screen -x miner`

I get the following messages in the cgminer screen session:

 [2019-12-11 22:27:21.273] Solo mining to valid address: 15tf4HVVDy9kZVuNa6v1Mtv3z1vq7ZuyY8
 [2019-12-11 22:27:26.145] Solo mining to valid address: 15tf4HVVDy9kZVuNa6v1Mtv3z1vq7ZuyY8
 [2019-12-11 22:27:31.255] Solo mining to valid address: 15tf4HVVDy9kZVuNa6v1Mtv3z1vq7ZuyY8
 [2019-12-11 22:27:36.043] Solo mining to valid address: 15tf4HVVDy9kZVuNa6v1Mtv3z1vq7ZuyY8
 [2019-12-11 22:27:41.208] Solo mining to valid address: 15tf4HVVDy9kZVuNa6v1Mtv3z1vq7ZuyY8
 [2019-12-11 22:27:46.129] Solo mining to valid address: 15tf4HVVDy9kZVuNa6v1Mtv3z1vq7ZuyY8
 [2019-12-11 22:27:51.178] Solo mining to valid address: 15tf4HVVDy9kZVuNa6v1Mtv3z1vq7ZuyY8
 [2019-12-11 22:27:56.084] Solo mining to valid address: 15tf4HVVDy9kZVuNa6v1Mtv3z1vq7ZuyY8
 [2019-12-11 22:28:01.243] Solo mining to valid address: 15tf4HVVDy9kZVuNa6v1Mtv3z1vq7ZuyY8
 [2019-12-11 22:28:06.043] Solo mining to valid address: 15tf4HVVDy9kZVuNa6v1Mtv3z1vq7ZuyY8

This is different compared to mining to a pool (e.g. solo.ckpool.org). Another benefit to ckpool is that you can read the status from outside your network. 

The cool thing is, by using it in this way that you are using all of the raspiblitz services.

metime17, 2020-01-18:
@Xorell how did you set up bitcoin core? I'm having issues getting CGMiner to connect to bitcoind

Xorell, 2020-01-18:
@metime17 I installed the CGMiner on the Raspiblitz. In this case you connect to http://localhost:8332 together with the username and password.

In case you are trying to connect with CGMiner from another target, then you need to make sure the firewall (i geuss UFW command) is allowing access toward the bitcoin deamon.

metime17, 2020-01-18:
@Xorell can you give me an example of your bitcoin.conf? also what version of bitcoin core are you using? im using version 19, and am not sure if it's having issues. i have UFW off so it shouldnt be that. im using localhost ofcourse.

Xorell, 2020-01-19:
@metime17 , I am using the default bicoin.conf frome the Raspiblitz:
```
# bitcoind configuration
# mainnet/testnet
testnet=0
# Bitcoind options
server=1
daemon=1
txindex=0
disablewallet=1
# Connection settings
rpcuser=raspibolt
rpcpassword=7YIcI5Mt9bEF3KaTcFN2
rpcport=8332
rpcallowip=127.0.0.1
rpcbind=127.0.0.1:8332
zmqpubrawblock=tcp://127.0.0.1:28332
zmqpubrawtx=tcp://127.0.0.1:28333
```

version is:
`Bitcoin Core Daemon version v0.18.1`

metime17, 2020-01-19:
@Xorell THANK YOU VERY MUCH!! idk why but my config was not working but yours did im guessing it was the zmqpubrawblock=tcp://127.0.0.1:28332 & zmqpubrawtx=tcp://127.0.0.1:28333 commands.
but thank you!!!

rootzoll, 2020-04-01:
Marked this idea for the Lightning HackSprint - details see: https://wiki.fulmo.org/index.php?title=Lightning_HackSprint

Tasks would be: See if its possible to use the RaspiBlitz to control mining hardware. If there would be the possibility to add it too a pool with supporting Stratum V2 with "job negotiation" and the RaspiBlitz could make the rules which transactions get into the Block - that would be awesome :) For background see: https://www.coindesk.com/a-plan-to-decentralize-bitcoin-mining-again-is-gaining-ground

rootzoll, 2020-08-25:
Also maybe look into the possibility to connect to a Mining Pool that paysout with lightning. Collecting info on that with this tweet: https://twitter.com/rootzoll/status/1298198278887084032?s=20

rootzoll, 2020-10-08:
Also checkout BrainsOS ... it allows to customize the own blocktemplate as part of BetterHash-style StratumV2 improvement and supports S9 miners https://braiins-os.com/plus

wihug, 2022-04-25:
Sure, already installed some packages including bfgminer. Testing and let you know when I achieve something. thank you.  this [link ](https://notgrubles.medium.com/solo-mining-with-blockstream-satellite-9303fdfc5090) was my starting point.

hpx127, 2022-07-21:
"Raspiblitz" mine block itself or use(require) third party miners ...? if third party miners require then how to connect each other raspiblitz and third party miner.....?