raspiblitz/raspiblitz #416: Question about LND transactions

zuardi, 2019-03-10:
Guys, 
I've finally set my node up and I've been trying to open some channels. But every time I fund my wallet with on-chain BTC, a significant part of it is sent elsewhere and is never added to Total Balance. Autopilot is OFF. Am I being robbed?
[image]

closed 2019-03-10 by zuardi

rootzoll, 2019-03-10:
@zuardi its a quirk with LND not counting the "change outputs" of a BTC transaction (to fund a channel) until they are confirmend. So far it just needs time to have the balance available again.

zuardi, 2019-03-11:
thanks for your response! I had figured it out and closed it.