raspiblitz/raspiblitz #417: Syncing blockchain taking ages?

pravsa, 2019-03-10:
Hey,

I downloaded the blockchain via Torrent it took about 10 hours. I guess that's pretty fast but now I have 20.87% of final blockchain sync done after 70 hours. 
Is this normal? It seems like it's not even moving anymore.

Thanks a lot for your answer.

Best regards

fluidvoice, 2019-03-10:
SSH --> `sudo tail -n 50 /mnt/hdd/bitcoin/debug.log`

pravsa, 2019-03-10:
> SSH --> `sudo tail -n 50 /mnt/hdd/bitcoin/debug.log`

Thanks for helping me. Seems like i can't ssh anymore. It says:

> Starting main menu...
> checking bitcoind - please wait...
> can take longer if device was off or first time 

btw: i never turned it off..

fluidvoice, 2019-03-10:
It says that where, on the LCD or when you do `ssh admin@your_node` ?
If the later hit ctrl+c to get out of the main menu to the command prompt, then run:
`sudo tail -n 50 /mnt/hdd/bitcoin/debug.log`

rootzoll, 2019-03-10:
@pravsa when sync is below 90% after the torrent download something went wrong using that torrent data and it looks like your node is syncing from scratch.

pravsa, 2019-03-10:
@fluidvoice the latter was the case ind it worked now but nothing happened.

@rootzoll Yeah it seems like because it is really slow. 3 hours later and 21,15% now...
What would you suggest? Flashing the sd card new and start from begin? What do I have to consider?

rootzoll, 2019-03-10:
@pravsa normally I would be interested why that happend - why does the torrent download got ignored and how to fix this for future releases. But thats always hard to debug in a remote situation. At least I will add a:

TODO v1.1: make the torrent download check better and maybe lets remember the setup that it had to torrent and when bitcoind is starts syncing fresh add some warning.

To be practical about it ... you can try to restart the torrent with: SSH in, use CTRL+c to get to the terminal and then run the command: `/home/admin/50torrentHDD.sh`

pravsa, 2019-03-10:
@rootzoll As I remember correctly I think my laptop went into standby mode during the step "Setup Lightning" as I was AFK. For that reason, I had to SSH in again. Maybe that helps by finding the problem...

I tried it but now it downloaded on torrent in SSH but the display still showed the syncing. I was not sure so I am currently formatting everything again and I will give it another try. 

Thanks a lot for your effort - I'll let you know if i made it ;)

rootzoll, 2019-03-11:
TODO v1.1: make setup scripts not depending on open ssh anymore - make SSH observer and controller, not actor - so that it a broken pipe will not mess up things. That is needed to run setup thru a browser in the future anyway.

pravsa, 2019-03-13:
So i tried to download it again via FTP.
After 2 1/2 days i tried to ssh in (yesterday evening it said 66%). and it says 
"The download failed or is not complete. Do you want keep already downloaded data for next try?"
no matter I say yes or no it asks if i want to continue with the setup. Then i choose yes.
After that, it asks how i want to download the blockchain -> 5 options
I pick FTP again and I start from the begin:
"The download failed or is not complete. Do you want keep already downloaded data for next try?"

Never ending loop...

fluidvoice, 2019-03-13:
to download it manually you can do the following (repeatedly if it fails), then reboot once it has completed...
`sudo wget -c -r -P /mnt/hdd/download/ --show-progress ftp://f00f39c4:download@w0189aba.kasserver.com/`

pravsa, 2019-04-06:
Btw: I tried to download it via FTP several times but I always had different error messages.
After setting up everything new, I downloaded it via torrent and it finally worked!

Now I am struggling with the port forwarding ;)

nyxnor, 2021-04-18:
please close this.

closed 2021-04-19 by rootzoll