raspiblitz/raspiblitz #418: SD card Ran completely out of space.

CommanderPoe, 2019-03-10:
I just realised today that my node has ZERO free GB on sd card. 
[image]

Today ZAP was failing on me and i decided to restart it (reload the qr code connecting via the node), when i did this message appeared up: _"Cannot write to 'go1.11.linux-armv6l.tar.gz' (No space left on device)."_ see img below.
[image]

I checked the issues #15  and #16  but they all have to do with Exapnding the files which they are all supposed to be embeded within the latest configuration of the Raspiblitz... I have no idea what's going on from here...

I also noticed couple of days ago that my node runs out of FREE MEMORY as well from times to times when i back up the lnd files, i tried 2 different methods, one using `scp -r ... ` from my desktop pc and using cyberduck to copy the whole directory, both times after the copy it did ran out of FREE MEMORY. I dont even know if this helps for the space issue tbh. Any help could be more than welcome.

fluidvoice, 2019-03-10:
http://ask.xmodulo.com/find-biggest-files-directories-linux.html
https://www.makeuseof.com/tag/increase-storage-raspberry-pi/

CommanderPoe, 2019-03-10:
Nothing seems particularly weird... 
[image]

i tried to expand the files system but im getting this in return: _There was an error running option A1 Expand Filesystem_

CommanderPoe, 2019-03-10:
I ran the same commands with the bitcoin user and after lot of permission denied i got this.
[image]

The root user doesnt show anything weird as well. it the bitcoin user.

fluidvoice, 2019-03-10:
there are 3 users for Raspiblitz (admin, bitcoin, and pi) You'll have to run the commands for all three and that's why you get denied messages.  so you have to...
sudu -u pi ...
sudo -u admin ...
sudo -u bitcoin ...
probably for expanding the SD card partition you'll have to use pi user.
how big is your card?
`sudo -u pi lsblk`

fluidvoice, 2019-03-10:
better is `sudo du -h /var/log`

CommanderPoe, 2019-03-10:
admin -
[image]

bitcoin -
[image]

Pi -
[image]

`sudo ls -alS /var/log`
[image]
`sudo du -h /var/log`
[image]

@fluidvoice ... Defo var/log is punishing me... why?

CommanderPoe, 2019-03-10:
i have to mention something... When i recovered all my channels, i deleted the old LOG that was on that lnd directory in hopes a new one be created for a friend to analyse it.. could that be the reason?

fluidvoice, 2019-03-10:
well it looks like you have over 10GB of log files in /var/log. From your `df` output looks like your SD card is 16GB? so that's prob the root of the problem. These files need to be cleaned out. The daemon.log file alone is 7.4GB. We need to see how to limit this growth.

CommanderPoe, 2019-03-10:
> well it looks like you have over 10GB of log files in /var/log. From your `df` output looks like your SD card is 16GB? 
Yes it is.

>so that's prob the root of the problem. These files need to be cleaned out. The **daemon.log** file alone is 7.4GB. We need to see how to limit this growth.
Howcome no body else has experienced issues like this? Maybe bigger space in SD card and maybe they are all bound to face issues like these sooner or later? How can i clean out that space?  Does my argument about deleting the logs from the lnd folder when i copied got anything to do with it?

fluidvoice, 2019-03-10:
> > well it looks like you have over 10GB of log files in /var/log. From your `df` output looks like your SD card is 16GB?
> 
> Yes it is.
> 
> > so that's prob the root of the problem. These files need to be cleaned out. The **daemon.log** file alone is 7.4GB. We need to see how to limit this growth.
> 
> Howcome no body else has experienced issues like this? Maybe bigger space in SD card and maybe they are all bound to face issues like these sooner or later? How can i clean out that space? Does my argument about deleting the logs from the lnd folder when i copied got anything to do with it? Why are those logs so big?

I don't know yet. I'm learning as I go. Not an expert on Linux logging.

fluidvoice, 2019-03-10:
I would delete the daemon.log* and syslog* files in this directory then reboot.
You might have to use `sudo -u root` to delete them.

fluidvoice, 2019-03-10:
apparently this is supposed to be controlled by "logrotate" on Debian/Raspbian such that they get cleaned up every 7 days. Perhaps this needs a different configuration. Logrotate can be used to clean out the files also.  Needs research.

rootzoll, 2019-03-10:
This is already on the v.1.1 todo list with issue: https://github.com/rootzoll/raspiblitz/issues/394

fluidvoice, 2019-03-10:
@CommanderPoe **correction!:** read up on `logrotate` before deleting those files. Dunno if they get recreated properly if not "rotated" or managed by `logrotate`.

fluidvoice, 2019-03-10:
> This is already on the v.1.1 todo list with issue: #394

Is there a list somewhere showing v1.1 todo list?  so I can check next time :)

fluidvoice, 2019-03-10:
Config file info: it's confusing which one is used, this one or the one in my subsequent post

admin@THUNDA:~ $ `cat /etc/rsyslog.conf `
#  /etc/rsyslog.conf	Configuration file for rsyslog.
#
#			For more information see
#			/usr/share/doc/rsyslog-doc/html/rsyslog_conf.html **(DOESN'T EXIST!)**

#################
#### MODULES ####
#################

module(load="imuxsock") # provides support for local system logging
module(load="imklog")   # provides kernel logging support
#module(load="immark")  # provides --MARK-- message capability

# provides UDP syslog reception
#module(load="imudp")
#input(type="imudp" port="514")

# provides TCP syslog reception
#module(load="imtcp")
#input(type="imtcp" port="514")

###########################
#### GLOBAL DIRECTIVES ####
###########################

#
# Use traditional timestamp format.
# To enable high precision timestamps, comment out the following line.
#
$ActionFileDefaultTemplate RSYSLOG_TraditionalFileFormat

#
# Set the default permissions for all log files.
#
$FileOwner root
$FileGroup adm
$FileCreateMode 0640
$DirCreateMode 0755
$Umask 0022

#
# Where to place spool and state files
#
$WorkDirectory /var/spool/rsyslog

#
# Include all config files in /etc/rsyslog.d/
#
$IncludeConfig /etc/rsyslog.d/*.conf

###############
#### RULES ####
###############

#
# First some standard log files.  Log by facility.
#
auth,authpriv.*			/var/log/auth.log
*.*;auth,authpriv.none		-/var/log/syslog
#cron.*				/var/log/cron.log
daemon.*			-/var/log/daemon.log
kern.*				-/var/log/kern.log
lpr.*				-/var/log/lpr.log
mail.*				-/var/log/mail.log
user.*				-/var/log/user.log

#
# Logging for the mail system.  Split it up so that
# it is easy to write scripts to parse these files.
#
mail.info			-/var/log/mail.info
mail.warn			-/var/log/mail.warn
mail.err			/var/log/mail.err

#
# Some "catch-all" log files.
#
*.=debug;\
	auth,authpriv.none;\
	news.none;mail.none	-/var/log/debug
*.=info;*.=notice;*.=warn;\
	auth,authpriv.none;\
	cron,daemon.none;\
	mail,news.none		-/var/log/messages

#
# Emergencies are sent to everybody logged in.
#
*.emerg				:omusrmsg:*

fluidvoice, 2019-03-10:
admin@THUNDA:~ $ `less /usr/share/doc/rsyslog/README.Debian `
admin@THUNDA:~ $ `cat /etc/logrotate.d/rsyslog`
/var/log/syslog
{
	rotate 7
	daily
	missingok
	notifempty
	delaycompress
	compress
	postrotate
		invoke-rc.d rsyslog rotate > /dev/null
	endscript
}

/var/log/mail.info
/var/log/mail.warn
/var/log/mail.err
/var/log/mail.log
/var/log/daemon.log
/var/log/kern.log
/var/log/auth.log
/var/log/user.log
/var/log/lpr.log
/var/log/cron.log
/var/log/debug
/var/log/messages
{
	rotate 4
	weekly
	missingok
	notifempty
	compress
	delaycompress
	sharedscripts
	postrotate
		invoke-rc.d rsyslog rotate > /dev/null
	endscript
}

CommanderPoe, 2019-03-11:
Well... I deleted the first four on my list, daemon.log daemon.log.1 syslog syslog.1 that were occupying a big chunk of the sd card, around 60%. 

When i rebooted it asked me to unlock lnd and it tootk sometime with some error that had never appeared but in the end when the blockchain synched lnd started working properlyu and this morning i have all channels back in place. Maybe i rushed to do that, but it seems to be working properly atm. I will check all that when i get home, thanks @fluidvoice

fluidvoice, 2019-03-11:
> Well... I deleted the first four on my list, daemon.log daemon.log.1 syslog syslog.1 (some of them were 1 week old) that were occupying a big chunk of the sd card, around 60%.
> 
> When i rebooted it asked me to unlock lnd and it tootk sometime with some error that had never appeared but in the end when the blockchain synched lnd started working properlyu and this morning i have all channels back in place. Maybe i rushed to do that, but it seems to be working properly atm. I will check all that when i get home, thanks @fluidvoice

Coolio. Glad it's working for you.  @rootzoll will address this config issue in the next release v1.1

CommanderPoe, 2019-03-11:
The sd card is filling up even faster now... Yesterday only 25% of the sd was used today 60% this is madness.

LightningK0ala, 2019-03-12:
I'm also having this issue, just noticed today my 30GB SD Card is full. my /var/log/daemon.log (I have 2) are almost 20GB.

fluidvoice, 2019-03-12:
> I'm also having this issue, just noticed today my 30GB SD Card is full. My /var/log/daemon.log (I have 2) are almost 20GB.

connect via SSH, go to command line, then run:
`sudo systemctl stop bitcoind.service` 
`sudo systemctl stop lnd.service`
`sudo rm /var/log/daemon* && sudo rm /var/log/*.gz`
Edit your **/mnt/hdd/lnd/lnd.conf** file and set **debuglevel=info**
then restart.

CommanderPoe, 2019-03-12:
> connect via SSH, go to command line, then run:
> `sudo systemctl stop bitcoind.service`
> `sudo systemctl stop lnd.service`
> `sudo rm /var/log/daemon* && sudo rm /var/log/*.gz`
> Edit your [...]

@LightningK0ala did u try this? did it work for you?

fluidvoice, 2019-03-13:
> @LightningK0ala did u try this? did it work for you?
works for this guy: https://github.com/rootzoll/raspiblitz/issues/423#issuecomment-472406125

LightningK0ala, 2019-03-13:
I went the route of tweaking logrotate settings to limit the size of the logs. But I've never used it before and not sure the change I made yesterday worked as I expected.

`debuglevel=info` should do the trick, making that change now.

fluidvoice, 2019-03-13:
> I went the route of tweaking logrotate settings to limit the size of the logs. But I've never used it before and not sure the change I made yesterday worked as I expected.
> 
> `debuglevel=info` should do the trick, making that change now.

what did you set your logrotate settings for the **daemon** and **syslog**?

LightningK0ala, 2019-03-13:
Added this to `/etc/logrotate.d/rsyslog`:
```
/var/log/daemon.log
{
    size 250M
    rotate 2
    create
}
```

rootzoll, 2019-03-14:
I have done the following the measures to fix this issue for the v1.1 release ... is that enough?

1. LND logging level is set to `info`

2. Added config to `/etc/init.d/rsyslog`:
```
/var/log/daemon.log
{
        rotate 4
        size=100M
        missingok
        notifempty
        compress
        delaycompress
        sharedscripts
        postrotate
                invoke-rc.d rsyslog rotate > /dev/null
        endscript
}

3. If on boot log dir is detected to be >1GB it all gets emergency deleted:
`sudo rm -r /var/log/*`
```

So v1.1 will reduce the log load, will keep daemon.log small and has a safe guard on boot to flush the logs if they get out of control again. Please let me know if you see more things to do on this.

This is today in testing and will be released as a new image end of week for everybody to easy update.

fluidvoice, 2019-03-15:
as I mentioned elsewhere, for v1.1 it might be better just turning off logging completely and create scripts for enable and disable it.  That because 99% of the time nobody is going to use these logs - prob just dev's when debugging a problem.  And doing ZERO logging will make the SD cards last longer and could even make the system run a little faster.

rootzoll, 2019-03-15:
Zero logging is maybe a bit too hard - but I switched the services lnd, bitcoind & RTL to just send error messages to the journal logs. 

```
StandardOutput=null
StandardError=journal
```

rootzoll, 2019-03-18:
OK - logs look good after testrun. auth.log seems to grow per day around 10MB but I put it on rotation with 100MB max too. So I close this issue for v1.1 release.

closed 2019-03-18 by rootzoll

RemoRitzmann, 2022-03-05:
I had the same problem (filling up SD-Card /dev/root to 100%) 
My Problem was reindex-chainstate with the wrong user beeing logged in, so it coppied parts of the BLOCKCHAIN /root/.bitcoin/blocks/ until there was no mor space. Because I followed "Please restart with -reindex or -reindex-chainstate to recover." in a log file. 
My solution was to do it with "bitcoin" user like this:
sudo -u bitcoin bitcoind -reindex-chainstate

rootzoll, 2022-03-08:
Wrong user being logged in ... was this something raspiblitz was doing be default or more an individual error on your side?

RemoRitzmann, 2022-03-20:
my bad, but good to know now. Thanks!