raspiblitz/raspiblitz #420: Trouble logging in to pi through ssh

sa250039, 2019-03-11:
I'm sorry this is probably a stupid question and im a newb. but I am having trouble ssh'ing into the pi. I use putty to ssh and get to the log in screen but no username and password work. I use the ip address that pops onto my LCD screen as the log in, I've also used admin@myip, along with the password raspiblitz. but I have never been able to log in, it just freezes my putty and tells me connection error. the closest I ever got was to use pi as the log in but it just forwarded the screen from the LCD saying what my log in and password is and that's it. any help would be appreciated .

fluidvoice, 2019-03-11:
It should be admin@yourIP but if your node is syncing you may have to wait a while before it replies asking for you to enter your password.  Once you login, you'll have to hit ctrl-c to get out of the menu and to the command prompt. Perhaps you forgot your password?

rootzoll, 2019-03-11:
Make sure that you are on the same local network. What is `ping [ip-of-your-node]` say?

If you are in WLAN check that you dont have "IP isolation" switched on at your router (its an edge casem but can happen). Also if you have a VPN or firewall running on your laptop maybe switch that off and try again.

sa250039, 2019-03-11:
Pinging 192.168.0.32 with 32 bytes of data:
Reply from 192.168.0.32: bytes=32 time=102ms TTL=64
Reply from 192.168.0.32: bytes=32 time=5ms TTL=64
Reply from 192.168.0.32: bytes=32 time=4ms TTL=64
Reply from 192.168.0.32: bytes=32 time=6ms TTL=64

Ping statistics for 192.168.0.32:
    Packets: Sent = 4, Received = 4, Lost = 0 (0% loss),
Approximate round trip times in milli-seconds:
    Minimum = 4ms, Maximum = 102ms, Average = 29ms

and I don't have a vpn or firewall running at the moment.
 and no ip isolation.
I can connect enough to reach the log in screen but once I try the admin@myip and password raspiblitz it just tells me an error occurred. if I use log in pi and the password raspiblitz it takes me to a mirror of the screen on the little lcd but I cannot interact with it.

sa250039, 2019-03-11:
> Make sure that you are on the same local network. What is ping [ip-of-your-node] say?
> If you are in WLAN check that you dont have "IP isolation" switched on at your router (its an edge casem but can [...]

Pinging 192.168.0.32 with 32 bytes of data:
Reply from 192.168.0.32: bytes=32 time=102ms TTL=64
Reply from 192.168.0.32: bytes=32 time=5ms TTL=64
Reply from 192.168.0.32: bytes=32 time=4ms TTL=64
Reply from 192.168.0.32: bytes=32 time=6ms TTL=64
Ping statistics for 192.168.0.32:
Packets: Sent = 4, Received = 4, Lost = 0 (0% loss),
Approximate round trip times in milli-seconds:
Minimum = 4ms, Maximum = 102ms, Average = 29ms
and I don't have a vpn or firewall running at the moment.
and no ip isolation.
I can connect enough to reach the log in screen but once I try the admin@myip and password raspiblitz it just tells me access denied. if I use log in pi and the password raspiblitz it takes me to a mirror of the screen on the little lcd but I cannot interact with it.

rootzoll, 2019-03-13:
OK so you can reach your Pi - so ist not a network issue.
Can you try to logon with: bitcoin@myip ... ist to see if that works.

lolrip, 2019-03-13:
I am having the same issue. I can ping the host and receive replies but when I ssh to admin (or bitcoin@) I receive:

Connection reset by 10.10.10.57 port 22 

I attempted to mount the sd card to check SSH config, permissions, etc. but it will not allow me to mount "rootfs"

lolrip, 2019-03-13:
Sorry for the spam.

I solved this issue by mounting the SD card and removing all of the SSH keys from /etc/ssh/

`rm /mnt/rootfs/etc/ssh/*key*` 

I am now able to SSH to the pi

m8tion, 2019-03-13:
I use Linux Mint 19 and for whatever reason my ssh connections only works if I call them with root privileges ("sudo ssh admin@myip"). Maybe you can try this if you are on Linux or launch your terminal / ssh app on windows as a superuser.

Zemekis11, 2019-07-16:
Hi I'm getting the same issue as sa250039 up above. The odd thing is that after brute force trying and retrying without changing anything it let me through once. I was able to complete the torrent download and then when it asked me to ssh using Password A....right back to infinite access denied.
[image]

When I first attempt to ssh into the raspiblitz, PuTTY gives me a warning about the connection and asks if I want to add it to my cache. Should I be agreeing?

Strikeaxe, 2019-07-23:
> Pinging 192.168.0.32 with 32 bytes of data:
> Reply from 192.168.0.32: bytes=32 time=102ms TTL=64
> Reply from 192.168.0.32: bytes=32 time=5ms TTL=64
> Reply from 192.168.0.32: bytes=32 time=4ms TTL=64 [...]

I had the same issue. I was successful logging in using only admin + raspiblitz

bucko13, 2019-07-25:
Having the same issue after everything working without any problem (able to sync the node and everything). RTL doesn't work and ssh doesn't work. `ping` responds with `64 bytes from 192.168.2.137: icmp_seq=9 ttl=64 time=2.119 ms`. using `sudo` does not work either. What's strange is that on the router (I have a dd-wrt router) in the DHCP clients it does not have the device listed. I also tried to plug in a keyboard to the raspbi and reboot it but the keyboard isn't responding which seems weird.

Zemekis11, 2019-07-31:
I was able to successfully login using a different SSH software, somehow I couldn't get PuTTY to work even after 10 hours tooling around with it so I tried Terminus for Android and _voila_, it worked on my first try! If PuTTY is being a troublemaker, don't beat your head against it for too long, just try a different device or piece of software. Currently 10 days into syncing, so even if you get past this hurdle, remember that there are a lot more days of waiting ahead...

openoms, 2019-07-31:
> I was able to successfully login using a different SSH software, somehow I couldn't get PuTTY to work even after 10 hours tooling around with it so I tried Terminus for Android and _voila_, it [...]

Termius on Android is a great tool indeed. Allows you to try from a whole different Operating system.

On Windows Kitty is a good alternative to Putty: https://www.fosshub.com/KiTTY.html

Also if someone is happy to dive in a bit Windows 10 can run a whole Linux with native ssh: https://docs.microsoft.com/en-us/windows/wsl/install-win10

bucko13, 2019-08-01:
I'm on a mac so I don't think the ssh issue is related to terminal application (using iTerm fwiw)

bucko13, 2019-08-01:
I was able to fix it by unplugging the raspberry pi as a kind of "hard reset". This obviously isn't ideal since it could cause problems with the connected HDD but at least I can ssh in again now. Blocks are syncing so hopefully that's all for now. Still unclear how the problem came up in the first place though or what a preferable way to fix might be.

rootzoll, 2019-08-12:
In the future with getting the touchscreen activated there should be safer ways in the future to shutdown and restart your RaspiBlitz.

fluidvoice, 2019-08-17:
> I was able to successfully login using a different SSH software, somehow I couldn't get PuTTY to work even after 10 hours tooling around with it so I tried Terminus for Android and _voila_, it [...]

FYI. when you can ping your node but SSH connection/login is not working and you cannot figure out why. 
**Reset the SSH server daemon configuration** (and Linux client):
_there is probably similar for Windows SSH "Putty" client._

#. resets all SSH configuration
#. on the SSH server (Raspberry Pi) 
`sudo systemctl stop sshd`
`sudo rm /mnt/hdd/ssh/ssh_host*`
`sudo ssh-keygen -A`
`sudo systemctl start sshd`
#. on SSH client Linux (Windows?)
`sudo rm ~/.ssh/known_hosts`

rootzoll, 2019-08-27:
There was another report .. so when you get a "Connection reset by port 22" your ssh key on the raspiblitz could be damaged. If you have no other way make a fresh sd card, so that the raspberrypi is creating new fresh ssh keys. If you had your RaspiBlitz already configured it will go thru the recover/update mode and you have to login with password `raspiblitz` and set a new password A when its finished. But tha should fix it.

pg024, 2019-08-29:
Hello, I have updated v1.2 to v1.3 RC5.
My previous configuration was having a ssh keypair login. 
Now I cannot login anymore, neither through the key nor with password.
```

`debug2:` we sent a publickey packet, wait for reply
debug1: Authentications that can continue: publickey
debug2: we did not send a packet, disable method
debug1: No more authentication methods to try.
Permission denied `(publickey).`
```

rootzoll, 2019-09-01:
After update/recover the the password A for SSH login is "raspiblitz" again. Try that as an alternative to keypair. 

Normally `raspiblitz/home.admin/_bootstrap.provision.sh` line 49 should link back your old SSh setting. Where did you stored the keypair before the update?

operahouserat, 2019-09-29:
Booting my raspiblitz for the first time and am having the same issue. I've pinged the IP successfully, so it isn't a network issue. 

With `ssh admin@[IP address]` command I get: `Connection reset by [IP address] port 22`. With `sudo ssh admin@[IP address]` command I am prompted for a password, but "raspiblitz" doesn't work.

Any suggestions?

patfreto, 2019-10-05:
Hi, I’m having trouble connecting.
I just etched the “raspiblitz-v1.3-2019-09-05.img” on the SD. I connected the Raspi with a LAN cable to the network and am trying to connect via my laptop on wifi through ssh to the raspi. But it won’t connect.
When I try to connect it says: “Connection reset by 192.168.178.12 port 22”
When I ping the adress it responds (sent=4, received=4, lost=0)
I tried connecting through promp, putty, MobaXterm, and termius.
I am totally new at this and have no technical background, so it’s hard for me to figure out what to do.
Can anyone help?

frennkie, 2019-10-06:
I would suggest you connect a screen and a keyboard and login locally to debug this issue.

pg024, 2019-10-06:
Mount the sd somewhere and create an empty file named "ssh" on the boot partition.

`$touch ssh`

cadwgan0, 2020-01-18:
I had to login manually to enable ssh through the raspi-config to resolve this issue.

norse2, 2020-03-09:
> FYI. when you can ping your node but SSH connection/login is not working and you cannot figure out why.
> **Reset the SSH server daemon configuration** (and Linux client):
> _there is probably similar [...]

Thank you. I had re-imaged two different micro-SD cards and then tried every fix I could find. Yours did the trick.

rootzoll, 2020-03-11:
Thanks for finding a fix for that. I reopen this issue to make a repair more easy in the future for this situation:

TODO for future version: If user add a file "ssh.reset" on the sd card, the raspiblitz should on boot do that ssh reset described above automatically.

rootzoll, 2020-03-18:
OK - version v1.5 RC1 will have the feature to set a flag file on the sd card to make the RaspiBlitz reset/repair the SSH certs. See: https://github.com/rootzoll/raspiblitz/blob/v1.5/FAQ.md#how-can-i-repair-my-ssh-login

If v1.5 RC1 is out (next weeks) - further testing is welcome, but closing this issue for now.

closed 2020-03-18 by rootzoll

countfloyd66, 2020-05-24:
after a few re-image sessions

CTRL-C at first boot
sudo raspi-config
 - enable ssh
 - enable vnc
 - expand filesys

Reboot and ssh is now solid session after session as usual

chuta, 2020-06-17:
When i encountered this problem, i simply ran this one line of code on my Linux laptop, and it resolved the issue for me.

sudo systemctl stop sshd

0xjacobb, 2020-07-04:
My Raspiblitz was working and fully synched. Then I've updated some services and it got stocked in "_**Waiting for reboot...**_" screen. After two days of waiting I've cut the power (directly without disconnecting the Ethernet). This leads me to some other problems, and I've flashed my SD card again. 

Then the Raspiblitz started up (after fresh SD-Card with old SSD), and it showed the normal starting screen (with IP address).  I've tried to SSH and got this message: `ssh: connect to host [my IP]  port 22: Connection refused `

Now, I can not SSH into the Raspiblitz anymore. Any idea?

Many thanks.

rootzoll, 2020-07-04:
@0xjacobb have you tried to fix the ssh login? https://github.com/rootzoll/raspiblitz/blob/v1.5/FAQ.md#how-can-i-repair-my-ssh-login

0xjacobb, 2020-07-05:
@rootzoll Yes, actually it still says on the Blitz Screen: "waiting for reboot...". SSH is not possible, PING is possible

rootzoll, 2020-07-05:
So if even a reset of the ssh is not working anymore ... something seem broken more deeply. Do you have funding on the node and channels open? Do you have a backup of the channel.backup file on Dropbox?

rootzoll, 2020-07-05:
@0xjacobb I saw on you twitter picture that you are still on the pibow case - dou you also still have the shim-fan running? Some people reported about strange SSH problems and solved it by removing the shim-fan. Thats why we changed to passive cooling heatsink case with the v1.5 shopping lists.

0xjacobb, 2020-07-06:
@rootzoll I don’t have any funds or channels. It’s a fresh new setup therefore I am open to to what ever it needs to get the Blitz back running.

You mean, that I should first unplug the SSD and power, remove the shim-fan and try again. 

What do you think I should do, if removing shim-fan does not work? Flash SD AND formatting the SSD and beginning from scratch? Because just falsch SD doesn’t work for me as described above.

Many thanks

0xjacobb, 2020-07-06:
@rootzoll What I did today and it looks that it works:

1. Disconnected SSD
2. Cut power
3. Unmounted display and Shim Fan
4. Mounted display again
5. Connected old, synched SSD 

**Result:** I was able to SSH!

--> Looks that the Shim Fan was definitely my problem. Thanks for the hint. I will now use the Heat Sink case. 

I will test now if everything could be setup again, without Shim Fan, old SSD and new Heat Sink case.

rafaelpac, 2020-09-01:
I just tried a fresh version 1.6 and got this SSH problem.
I created a file called ssh.reset in the card and now it works.

NilByte, 2020-11-19:
Had trouble logging in on v1.6.1 using Putty. Adding ssh.reset did not fix it. Reformatted SSD exFAT. Used Windows PowerShell and ssh login was successful.

CoderBoogs, 2020-12-05:
Anyone having an issue with after entry of password the main menu doesn't load? this happened with v1.6.0 then i updated to v1.6.1 and the menu loaded up fine. Now I am upgrading to v.1.6.2 and the same issue came about.

rootzoll, 2020-12-10:
@CoderBoogs so is there any "output" after you enter the password or you hit enter and then nothing? Also does it break the connection after waiting at one point or does it just hang forever until you break it up.

At which point of the process is that happening exactly? Is that on the first ssh after you started up with a updated sd card (after LCD shows ''LOGIN FOR FINAL RECOVERY") or after that. And how did you fixed that in the past?

Have you tried resetting the SSH keys with this option mentioned in the FAQs?
https://github.com/rootzoll/raspiblitz/blob/v1.6/FAQ.md#how-can-i-repair-my-ssh-login

DocBrown101, 2021-01-06:
I now have the same problem with ssh and in addition I get a 504 gateway time-out over my http services. The update to 1.6.2 is now about 2 weeks ago and in that time everything has worked.

rootzoll, 2021-01-10:
@DocBrown101 have you tried the basics from the FAQ? https://github.com/rootzoll/raspiblitz/blob/v1.6/FAQ.md#how-can-i-repair-my-ssh-login ... also to shutdown and make a fresh raspiblitz sd card image and running a recovery update could help to bring your raspiblitz back to life.

DocBrown101, 2021-01-10:
I didn't have to do anything. After about 40 minutes, everything worked again. Even without a reboot!

billgre, 2021-04-16:
I m a newbie and  I just got ready 2 go raspiblitz from Germany and have the same ssh problem. The https://github.com/rootzoll/raspiblitz/blob/v1.5/FAQ.md#how-can-i-repair-my-ssh-login didnt work for me. 
Whats the next step?

rootzoll, 2021-04-16:
@billgre sometimes on shipping the sd card image gets corrupted .. writing a fresh image to the sd card would be a first step to do. You find download and intructions here: https://github.com/rootzoll/raspiblitz/blob/v1.6/README.md#downloading-the-software

billgre, 2021-04-19:
The new image worked for me. Thanks

deratko, 2021-06-18:
Hi everybody
Last week I started (or tried) to run my own node with raspiblitz. But I always have the issue with SSH that my connection times out on port 22. I wrote a fresh image multiple times now and it works but only for about 30 minutes than my laptop disconnetcs the SSH and I can't connect again. I also tried to turn off the firewall of my windows defender but it also doesn't work.  I also tried to plug in and out the raspberry (I now, I shouldn't do that but anyways I tried it) and it works maybe for some minutes again and than not. I also tried to put empty documents like "ssh" or "ssh.reset" into the boot but it doesn't work. Last time I succeded to log in via SSH the blockchain sync was at aorund 2%. I ordered a LCD touchscreen but it's still not here, so the only possibility to "look" whats happening is via SSH. Maybe someone can help?
Thanks a lot!

radixrat, 2021-06-18:
> Hi everybody
> Last week I started (or tried) to run my own node with raspiblitz. But I always have the issue with SSH that my connection times out on port 22. I wrote a fresh image multiple times now [...]

Is your Pi using an Ethernet cable or is it wifi? One of my pi’s drops wifi when it gets too hot.

deratko, 2021-06-18:
Wifi. I will try to cool it down a little bit and look again. Thanks!

deratko, 2021-06-18:
> Is your Pi using an Ethernet cable or is it wifi? One of my pi’s drops wifi when it gets too hot.

Does the raspberry pi 4 connect automatically to the wifi again? Sry I'm a noob...

radixrat, 2021-06-18:
> Does the raspberry pi 4 connect automatically to the wifi again? Sry I'm a noob...

I found that with my Pi that had the problem that it would disable the module until I rebooted it, if I were you I would plug it in with Ethernet temporarily and see if ssh continues to work, if it does then probably something with wifi module.

deratko, 2021-06-18:
> I found that with my Pi that had the problem that it would disable the module until I rebooted it, if I were you I would plug it in with Ethernet temporarily and see if ssh continues to work, if it [...]

To do so, I'd have to plug out the power cable and move the pi to the living room where the rooter is.. is there maybe another possibility?

jaffabaffa, 2024-11-20:
I have an issue.
When I open PuTTY, I enter the IP address of my Raspberry PI.
I than click on Load, whereafter below's menu opens:

[image]

As per the manual, I enter "ssh" after "logis as", whereafter I click enter on my keyboard.
The following line appears:

[image]

The problem now is that can't seem to enter anything as the password.
I can't even select the password field with my cursor.
They only thing I can do, is press enter a second time on my keyboard, whereafter it just says this: Access denied

[image]

What do I do now?

jaffabaffa, 2024-11-22:
Thanks!
Your're right. 

The "Login as:" should indeed be "**admin**", and **NOT "ssh"** (contrary to what the raspiblitz guide tells you).

Furthermore, the password **does not visibly appear** on screen when you enter it. You need to "blindly" enter the password (without seeing anything happen on screen), and just press enter on your keyboard afterwards.

I hope this little nugget of information might helps others in the future.
I was really stuck here for a while (I have no IT experience whatsoever).