raspiblitz/raspiblitz #422: RTL stuck after typing password

cryptomulde, 2019-03-11:
When I try to log into my Raspiblitz via RTL it loads up the page and I can insert my password.
After that RTL is stuck in that screen. Any ideas how to fix this?
[image]

cryptomulde, 2019-03-13:
Will try again with a clean image of the sd card.

rootzoll, 2019-03-13:
Working on a quick rediced v1.1 release that wil also have an updated RTL by end of week  ... maybe its worth waiting for that.

cryptomulde, 2019-03-13:
I think it had to do with Cookies. Found out it is working on Chrome. Deleted Cookies on Firefox and it was fixed.

rootzoll, 2019-03-14:
For version v1.1 the new RTL update is included - tested well on my machines. Closing Issue.

closed 2019-03-14 by rootzoll