raspiblitz/raspiblitz #424: RESEARCH: reliability issues

fluidvoice, 2019-03-12:
Broader subject is various reliability issues:

**SD Card:**
quick note I just realized some things today via the recent problem of the /var/logs folder filling up the SD card.  This brings up another source of problems... That SD cards, particularly cheap ones, can be unreliable and have a short lifespan (as far as write cycles). This can be causing a lot of "uknown" failures via SD card corruption, crashes, boot problems, etc.  
  **Perhaps the shopping list should mention the option of spending a little more for a "High Endurance" SD card** like a [Samsung Pro Endurance 32GB](https://www.amazon.com/dp/B07B98GXQT/ref=twister_B07DLWKFTM?_encoding=UTF8&psc=1) card US$13

raumi75, 2019-03-12:
Very good point. Even spending $10 more on a card would be a reasonable thing to do. I assume that most people will run mainnet and put $20 or more into lightning channels. If we ruin that experience with too cheap hardware, we'd be doing lightning a disservice.

LightningK0ala, 2019-03-12:
Couldn't we just have the log files in the external hd?

fluidvoice, 2019-03-12:
> Couldn't we just have the log files in the external hd?

You could do it for Bitcoin/LND logs but not for ALL of the system logs.  It might increase the lifespan of an SD card but would not remove the problem entirely.  It's worth considering since most of the logging is being done by LND.

rootzoll, 2019-03-12:
The basic bitcoin and LND logs are already on the HDD, but I think the problem is that they are mirrowed thru systemd into the system logs, correct?

fluidvoice, 2019-03-12:
> The basic bitcoin and LND logs are already on the HDD, but I think the problem is that they are mirrowed thru systemd into the system logs, correct?

Yes, one user said when debuglevel=debug that the daemon.log file was full of LND entries.
Basically all various daemons running their log output goes here.

LightningK0ala, 2019-03-13:
I'm going with the debuglevel=info fix. Just curious that there would be a way to prevent the daemon output to be piped into system logs.

rootzoll, 2019-03-13:
@LightningK0ala to prevent having it written to daemon logs I think the key is to set the `StandardOutput=null` in the systemd service file of bitcoind and LND. If this option is not set, it defaults to `journal`.

fluidvoice, 2019-03-13:
Or maybe this?

[Service]
StandardOutput=null
StandardError=journal

LightningK0ala, 2019-03-13:
Got it, that makes sense. Thanks @rootzoll @fluidvoice

fluidvoice, 2019-03-14:
Perhaps related to reliability is the prospect of booting Raspiblitz from a USB drive instead of SD card.
My attempts at this are over here: https://github.com/rootzoll/raspiblitz/issues/430#issue-421262771

rootzoll, 2019-03-15:
OK I updated almost all shopping lists with the samsung 32gb endurance - 16gb sd card is still ok but the "endurance" is not available smaller then 32GB.

fluidvoice, 2019-03-15:
> OK I updated almost all shopping lists with the samsung 32gb endurance - 16gb sd card is still ok but the "endurance" is not available smaller then 32GB.

@rootzoll I think also a mini-USB to USB-A cable with a built in on/off switch is also worth adding since continual plug/unplug wears out the mini-USB connector on the Pi.  It's cheap. Let me go dig out the link and add it here.

rootzoll, 2019-03-17:
Going thru the power supplies is part of another issue and we should target it after the v1.1 release. But it needs more look for detail. For example I had the first one power supply link you posted and it had undervoltage notices on my machine. Someone posted that a on/off switch cann be a source for unstable output - not sure.

This german power suply seems valid so far - big and stable - the goal would be find similar ones from the other stores. But that will be part of ticket: https://github.com/rootzoll/raspiblitz/issues/365

For the v1.1 release I will close this issue.

closed 2019-03-17 by rootzoll

fluidvoice, 2019-03-29:
Found a great article - [What's Wrong With The Raspberry Pi](https://ownyourbits.com/2019/02/02/whats-wrong-with-the-raspberry-pi/)
Turns out an inadequate power supply is a VERY common problem (as in large number of users have it)... a "silent" problem that causes all kinds of different symptoms and malfunctioning from slow/throttled performance, corrupted SD cards, hang/lockup, and probably also corrupted blockchain DB files.  So as an enhancement I think we should try to detect this situation using the `vcgencmd` to expose data showing the problem. The question is when is the appropriate time to measure this?
[image]