raspiblitz/raspiblitz #425: Fatal LevelDB error: Corruption: 87 missing files

zuardi, 2019-03-13:
Guys,
suddenly my node stopped working. I've been browsing through but did not find the exact same problem. It seems some kind of database corruption, files missing...

> *** RASPIBLITZ LOGS ***
blitzversion: 1.0
chainnetwork: bitcoin / main
 11:48:44 up  2:03,  2 users,  load average: 1,03, 1,13, 1,12
*** CHAINNETWORK SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Wed 2019-03-13 11:48:23 -03; 20s ago
  Process: 8583 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
 Main PID: 8584 (code=exited, status=1/FAILURE)
mar 13 11:48:23 ZePikeno systemd[1]: bitcoind.service: Unit entered failed state.
mar 13 11:48:23 ZePikeno systemd[1]: bitcoind.service: Failed with result 'exit-code'.
*** LAST 20 CHAINNETWORK LOGS ***
2019-03-13T14:47:35Z Loaded best chain: hashBestChain=000000000000000000132b16efbd0c357520e45a9ae6de25faa9ddb1d133cb21 height=566760 date=2019-03-12T14:24:58Z progress=0.999462
2019-03-13T14:47:35Z init message: Rewinding blocks...
2019-03-13T14:47:48Z init message: Verifying blocks...
2019-03-13T14:47:48Z Verifying last 6 blocks at level 3
2019-03-13T14:47:48Z [0%]...[16%]...[33%]...[50%]...[66%]...[83%]...[99%]...[DONE].
2019-03-13T14:48:21Z No coin database inconsistencies in last 6 blocks (14785 transactions)
2019-03-13T14:48:21Z  block index           90753ms
2019-03-13T14:48:21Z Opening LevelDB in /mnt/hdd/bitcoin/indexes/txindex
2019-03-13T14:48:22Z Fatal LevelDB error: Corruption: 87 missing files; e.g.: /mnt/hdd/bitcoin/indexes/txindex/014533.ldb
2019-03-13T14:48:22Z You can use -debug=leveldb to get more complete diagnostic messages
2019-03-13T14:48:22Z 
************************
EXCEPTION: 15dbwrapper_error       
Fatal LevelDB error: Corruption: 87 missing files; e.g.: /mnt/hdd/bitcoin/indexes/txindex/014533.ldb       
bitcoin in AppInit()       
2019-03-13T14:48:22Z Shutdown: In progress...
2019-03-13T14:48:22Z scheduler thread interrupt
2019-03-13T14:48:22Z Shutdown: done

Can you help me please?

fluidvoice, 2019-03-13:
your blockchain DB got corrupted somehow. How long was this node running before this happened?
How much free space is on your SD card?... `sudo df -h`
Not sure what you can do other than downloading it and re-sync again.  Maybe you can connect your HDD to a notebook and reindex it on that?

zuardi, 2019-03-13:
Ive been running it for a week. I have currently 30 Gb of space, did that change in the rotating file.
Probably this is due to power shortage last night...
If I download bitcoin core on windows desktop and use this HDD with it, will I be able to reindex it and use it again on Pi?

fluidvoice, 2019-03-13:
> Ive been running it for a week. I have currently 30 Gb of space, did that change in the rotating file.
> Probably this is due to power shortage last night...
> If I download bitcoin core on windows desktop and use this HDD with it, will I be able to reindex it and use it again on Pi?

I think so yes.  You might want to first check the HD for problems:
[https://www.tecmint.com/check-linux-hard-disk-bad-sectors-bad-blocks/](https://www.tecmint.com/check-linux-hard-disk-bad-sectors-bad-blocks/)

fluidvoice, 2019-03-16:
Hard to tell. Repeated blockchain corruption problems could be hardware related.  A bad power supply or unreliable hard disk?

frennkie, 2020-08-08:
@zuardi closing - let us know if this is still an issue.

closed 2020-08-08 by frennkie