raspiblitz/raspiblitz #426: Received payment got split up -  only a part of the transaction available in the Wallet

BoostMatt, 2019-03-13:
all chains been synced. i wanted to load my wallet. i logged into RTL and created a new np2wkh address. 
i sent 0.009999 BTC to 3Ko19q3ZZoeEuZfZjcPSuBbGMbAgq8DLWx
tx Hash: 51864dfb418f86e4d8d07d6a6eeadea4ec58dcd33a9b03bd747ce254cfe88e3d

on RTL i see a transaction i didnt create: 0.00606064 BTC sent to 2 addresses:
bc1qnvgmtn9avqaqvrkukhf2kx4nkr0wur60xhrcms
bc1qp98dpyp9gayq0c8gz6tfk7fu09vcpgq3g40j9uw9gele5rkfjmwq5xen6g
Tx Hash: ab6422d9dd75fd1a9d002475957c3032c137850ecb4b2dbb84c22d09bad2f676

So i only have 0.00393836 BTC left in my wallet.
Can someone explain what happened and what i did wrong?

Thanks!
Matt

rootzoll, 2019-03-19:
There is a confusion when funding a wallet sometimes with change taking time to get back to the wallet. Or the auto-pilot starting to open channels. Did you find out what was happening in your case?

BoostMatt, 2019-03-20:
Right know i can see 393836 sat in the wallet and 598637 sat in 1 Channel.
Therefore it looks good. it probably startet to open channels (Autopilot was ON).
That's the problem with the excitement. Seldom comes with patience.

closed 2019-03-20 by BoostMatt