raspiblitz/raspiblitz #428: EXT4-fs error

raumi75, 2019-03-14:
This is the second time my external harddrive is having trouble. I could not even shut down my raspi because I could not ssh into it. Plugging an external keyboard and pressing `Ctrl+C` or `Ctrl+Alt+Del `did not work either. So I had to cringe and disconnect the power.

This is what the screen said:

```
EXT4-fs error (device sda1): ext4_find_entry:1438 inode #2: comm ls: reading directory lblock 0
EXT4-fs error (device sda1): ext4_find_entry:1438 inode #23461889: comm bitcoin-cli: reading directory
EXT4-fs error (device sda1): ext4_find_entry:1438 inode #33816577: comm lncli: reading directory lblock 0
```

[image]

raumi75, 2019-03-14:
I'm aware that his might be a power issue. I'm running the pi on an anker-usb-hub that connects the pi and the harddrive with a Y-cable.

Just found this: https://www.raspberrypi.org/forums/viewtopic.php?t=80219

Not sure if `max_usb_current=1` is relevant here.

rootzoll, 2019-03-14:
I saw that once on my test machine - was gone after a reboot. But on another user report that was signaling a final HDD malfunction. Adding that `max_usb_current=1` will not hurt to try .. will add to v1.1

raumi75, 2019-03-14:
Or maybe I just have a crappy hard disk.

raumi75, 2019-03-15:
It was under voltage. I thought I was clever by using an Anker-USB-charger instead of the recommended power supply.

Looking at my logs I found this

```
cat syslog.1 | grep voltage
Mar 13 18:15:26 raumiblitz kernel: [394726.673986] Under-voltage detected! (0x00050005)
```

Looking back at past logs when I still used the proper power-supply (shopping list)

```
zcat syslog.*.gz | grep voltage
```

I had no unter-voltage.

Suggestion: How about we add an undervolt-warning to 00infoBlitz.sh and the top of 00mainMenu.sh? Not sure how to do this. Maybe we'd execute the above mentioned `grep` command and if we have a non-zero output, the user should look into this issue before it's too late. Maybe "undervoltage-detected" should be presented as "WARNING: Not enough electricity!" or something.

Now I have the Y-connector plugged into the raspberry pi. Does that even make sense?

openoms, 2019-03-15:
This is a returning issue. I had plenty of undervoltage messages. Even if the Pi is able to deliver sufficient Ampers by having the USB ports connected the Voltage can be unstable due to it's design limitations . Suggested to use an Powered USB Hub to separate the power supply of the HDD. More info here: https://github.com/rootzoll/raspiblitz/issues/368

openoms, 2019-03-15:
I am testing this one: 
USB Hub Powered, atolla 4 Ports... https://www.amazon.co.uk/dp/B07B2HJG8P?ref=ppx_pop_mob_ap_share

It still says:
Power Usage note 
For a stable connection, the devices connected to this USB 3.0 Hub with power adapter must not exceed a combined current of 5 volts 3.5Ah. External hard-drives should be connected with its own power source.

fluidvoice, 2019-03-15:
> Suggestion: How about we add an undervolt-warning to 00infoBlitz.sh and the top of 00mainMenu.sh? Not sure how to do this.

Not necessary... https://github.com/rootzoll/raspiblitz/blob/e38c48e54536a6233a0062b99858d43685635d80/FAQ.md#are-those-under-voltage-detected-warnings-a-problem

fluidvoice, 2019-03-15:
> For a stable connection, the devices connected to this USB 3.0 Hub with power adapter must not exceed a combined current of 5 volts 3.5Ah. External hard-drives should be connected with its own power source.

3.5A is a lot!  I can't see a modern HDD using that much. Maybe an old 3.5" external case/disk.

rootzoll, 2019-03-15:
A lot of error reports were data corruption related and under-voltage reports should be more dominant on the display ... so once an hour the background service will count the number of unser-voltage reports in syslog and if >0 make statement on LCD. Hopes that signals people very early to improve power-supply.

fluidvoice, 2019-03-15:
> A lot of error reports were data corruption related and under-voltage reports should be more dominant on the display ... so once an hour the background service will count the number of unser-voltage [...]

yes. Like cheap SD cards, I think cheap power supplies cause a lot of problems.  And perhaps due to the low cost, they lie a lot about the spec's and the performance under real world loads are not up to expectation and specs.

rootzoll, 2019-03-17:
OK this notice is now part of the LCD screen:

[image]

Logs will be scanned every hour. The rule should be if you see this: try a better power supply or give the HDD extra power thru a Y-cable or a USB-hub. When the number then is not growing anymore you are good. The syslog is on daily rotation and should get way if no further undervoltage warnings. 

Closing issue for the v1.1 release.

closed 2019-03-17 by rootzoll