raspiblitz/raspiblitz #432: IDEA: Add mail notification if server goes down

cryptomulde, 2019-03-15:
Hey,
is there an easy way to notify the user in case of LND or Bitcoind going down? 
I think it shoud be very easy to send out a mail if problem XY occurs.

Could this be an idea for a future Raspiblitz feature?

CommanderPoe, 2019-03-15:
this please would be very very helpful in case something -externally- goes wrong and your pi shutsdown... Many of us dont use the auto unlock for a variety of reasons so having a notification like this would be great.

cryptomulde, 2019-03-15:
The problem here is, if the Pi  is shutdown it won't send a mail ^^ 
Only works if bitcoind or lnd has issues I guess. This is also why I changed the tilte.

fluidvoice, 2019-03-15:
I think this functionality already exists in Linux. Just needs configuration: https://serverfault.com/questions/876233/how-to-send-an-email-if-a-systemd-service-is-restarted

rootzoll, 2019-03-19:
If someone gets the build in linux function working ... please report on your experience.

fluidvoice, 2019-03-19:
> If someone gets the build in linux function working ... please report on your experience.

I'm not sure what you are asking for.  If **sendmail** was built OK?  for me, yes...

admin@RaspiBlitz:/var/log $ sudo apt-get install sendmail

> Reading package lists... Done
> Building dependency tree       
> Reading state information... Done
> The following additional packages will be installed:
>   libsigsegv2 lockfile-progs m4 procmail sendmail-base sendmail-bin sendmail-cf sensible-mda
> Suggested packages:
>   m4-doc sendmail-doc rmail logcheck sasl2-bin
> The following NEW packages will be installed:
>   libsigsegv2 lockfile-progs m4 procmail sendmail sendmail-base sendmail-bin sendmail-cf sensible-mda
> 0 upgraded, 9 newly installed, 0 to remove and 13 not upgraded.
> Need to get 2,078 kB of archives.
> After this operation, 4,449 kB of additional disk space will be used.
> Do you want to continue? [Y/n] Y
> Get:1 http://raspbian.c3sl.ufpr.br/raspbian stretch/main armhf libsigsegv2 armhf 2.10-5 [28.4 kB]
> Get:2 http://mirror.nbtelecom.com.br/raspberryPI/raspbian stretch/main armhf m4 armhf 1.4.18-1 [185 kB]
> Get:3 http://raspbian.c3sl.ufpr.br/raspbian stretch/main armhf lockfile-progs armhf 0.1.17 [10.6 kB]
> Get:4 http://raspbian.c3sl.ufpr.br/raspbian stretch/main armhf sendmail-base all 8.15.2-8 [343 kB]     
> Get:5 http://mirror.nbtelecom.com.br/raspberryPI/raspbian stretch/main armhf sendmail-cf all 8.15.2-8 [291 kB]
> Get:6 http://raspbian.c3sl.ufpr.br/raspbian stretch/main armhf sendmail-bin armhf 8.15.2-8 [645 kB]    
> Get:8 http://raspbian.c3sl.ufpr.br/raspbian stretch/main armhf sensible-mda armhf 8.15.2-8 [224 kB]    
> Get:9 http://raspbian.c3sl.ufpr.br/raspbian stretch/main armhf sendmail all 8.15.2-8 [222 kB]          
> Get:7 http://mirror.ufam.edu.br/raspbian/raspbian stretch/main armhf procmail armhf 3.22-25+deb9u1 [129 kB]
> Fetched 2,078 kB in 11s (178 kB/s)                                                                     
> Selecting previously unselected package libsigsegv2:armhf.
> (Reading database ... 95719 files and directories currently installed.)
> Preparing to unpack .../0-libsigsegv2_2.10-5_armhf.deb ...
> Unpacking libsigsegv2:armhf (2.10-5) ...
> Selecting previously unselected package m4.
> Preparing to unpack .../1-m4_1.4.18-1_armhf.deb ...
> Unpacking m4 (1.4.18-1) ...
> Selecting previously unselected package lockfile-progs.
> Preparing to unpack .../2-lockfile-progs_0.1.17_armhf.deb ...
> Unpacking lockfile-progs (0.1.17) ...
> Selecting previously unselected package sendmail-base.
> Preparing to unpack .../3-sendmail-base_8.15.2-8_all.deb ...
> Unpacking sendmail-base (8.15.2-8) ...
> Selecting previously unselected package sendmail-cf.
> Preparing to unpack .../4-sendmail-cf_8.15.2-8_all.deb ...
> Unpacking sendmail-cf (8.15.2-8) ...
> Selecting previously unselected package sendmail-bin.
> Preparing to unpack .../5-sendmail-bin_8.15.2-8_armhf.deb ...
> Unpacking sendmail-bin (8.15.2-8) ...
> Selecting previously unselected package procmail.
> Preparing to unpack .../6-procmail_3.22-25+deb9u1_armhf.deb ...
> Unpacking procmail (3.22-25+deb9u1) ...
> Selecting previously unselected package sensible-mda.
> Preparing to unpack .../7-sensible-mda_8.15.2-8_armhf.deb ...
> Unpacking sensible-mda (8.15.2-8) ...
> Selecting previously unselected package sendmail.
> Preparing to unpack .../8-sendmail_8.15.2-8_all.deb ...
> Unpacking sendmail (8.15.2-8) ...
> Setting up procmail (3.22-25+deb9u1) ...
> Setting up libsigsegv2:armhf (2.10-5) ...
> Processing triggers for install-info (6.3.0.dfsg.1-1+b1) ...
> Setting up m4 (1.4.18-1) ...
> Processing triggers for libc-bin (2.24-11+deb9u4) ...
> Processing triggers for systemd (232-25+deb9u9) ...
> Setting up sendmail-cf (8.15.2-8) ...
> Processing triggers for man-db (2.7.6.1-2) ...
> Setting up lockfile-progs (0.1.17) ...
> Setting up sendmail-base (8.15.2-8) ...
> adduser: Warning: The home directory `/var/lib/sendmail' does not belong to the user you are currently creating.
> Setting up sendmail-bin (8.15.2-8) ...
> update-alternatives: using /usr/lib/sm.bin/sendmail to provide /usr/sbin/sendmail-mta (sendmail-mta) in auto mode
[... 81 lines cut ...]
> Disabling HOST statistics file(/var/lib/sendmail/host_status).
> Creating /etc/mail/sendmail.cf...
> Creating /etc/mail/submit.cf...
> Informational: confCR_FILE file empty: /etc/mail/relay-domains
> Warning: confCT_FILE source file not found: /etc/mail/trusted-users
>  it was created
> Informational: confCT_FILE file empty: /etc/mail/trusted-users
> Warning: confCW_FILE source file not found: /etc/mail/local-host-names
>  it was created
> Warning: access_db source file not found: /etc/mail/access
>  it was created
> Updating /etc/mail/access...
> Linking /etc/aliases to /etc/mail/aliases
> Informational: ALIAS_FILE file empty: /etc/mail/aliases
> Updating /etc/mail/aliases...
> WARNING: local host name (RaspiBlitz) is not qualified; see cf/README: WHO AM I?
> /etc/mail/aliases: 0 aliases, longest 0 bytes, 0 bytes total
>  
> Warning: 3 database(s) sources
> 	were not found, (but were created)
> 	please investigate.
> Setting up sensible-mda (8.15.2-8) ...
> Setting up sendmail (8.15.2-8) ...
> Processing triggers for systemd (232-25+deb9u9) ...
>

rootzoll, 2019-03-19:
@fluidvoice so its just apt-get install sendmail, set an email in the service config and its working? No other config needed?

Just can remember making email sending on a plan server was always a hassel - thats why so many today use mailgun and similar services.

azamatsmith, 2019-03-20:
A much easier solution would be to fire a webhook call out to Zapier/IFTT to trigger the email, or whichever notification type the user selects in Zapier/IFTT. 

Happy to help if there is any code I can write in Node.js 😝

BoostMatt, 2019-03-20:
would be interessting if the nodes could check each other. you install the "sendmail-service" on each of your nodes. since you probably will connect your nodes to eachother anyway, you could set an alarming, as soon one of the defined (let's call them favourites) connections breaks, it sends out an email.
So you could also cover Power- or Hardware-Issues.
Would we awesome to have this in the RTL.

fluidvoice, 2019-03-20:
> @fluidvoice so its just apt-get install sendmail, set an email in the service config and its working? No other config needed?
> 
> Just can remember making email sending on a plan server was always a hassel - thats why so many today use mailgun and similar services.

Yeah perhaps. I don't know. I tried to do what was supposedly required but it didn't work... to just run:
`sudo sendmailconfig`
`echo "test email from sendmail" | sendmail email-name@domain.com`
sendmailconfig appeared to work without errors and the sendmail test had a long pause but returned without error. But the email never arrived and I tested it with two different of my email addresses. So something must be missing.

fluidvoice, 2019-03-20:
Package: [ssmtp](https://packages.debian.org/stable/mail/ssmtp) 

> extremely simple MTA to get mail off the system to a mail hub
> A secure, effective and simple way of getting mail off a system to your mail hub. It contains no suid-binaries or other dangerous things - no mail spool to poke around in, and no daemons running in the background. Mail is simply forwarded to the configured mailhost. Extremely easy configuration.
> 
> WARNING: the above is all it does; it does not receive mail, expand aliases or manage a queue. That belongs on a mail hub with a system administrator.

[image]

rootzoll, 2019-03-22:
Looks cool - I add this to the future release package to at least give it a shot. If someone like to write a `config.script` on this - go for it :)

fluidvoice, 2019-03-22:
I r

> Looks cool - I add this to the future release package to at least give it a shot. If someone like to write a `config.script` on this - go for it :)

This page looks good too: https://blog.edmdesigner.com/send-email-from-linux-command-line/

Luigybruno, 2019-03-31:
I've been using [bitnodes.earn.com](url) to check my node status. You can put your email so when your node is down you get a notification.

cryptomulde, 2019-03-31:
Yeah I do that too :)

woeisme, 2019-09-06:
bitnodes.earn.com recommended again, they also support tor nodes