raspiblitz/raspiblitz #433: Disallow SSH from internet by default.

fluidvoice, 2019-03-15:
Think about should we allow only to manage the node via HTTP from the internet and allow SSH only from the LAN (by default). Then perhaps have a script or option to allow SSH from the internet.

One way to make it more secure is to dis-allow SSH from the Internet and allow it only from the local LAN:

> only allowing password authentication from within your network. Add the following to the end of your sshd_config:
> 
> PasswordAuthentication no
> Match Address 10.0.0.0/8,172.16.0.0/12,192.168.0.0/16
>     PasswordAuthentication yes
>

_Originally posted by @fluidvoice in https://github.com/rootzoll/raspiblitz/issues/431#issuecomment-473285667_

rootzoll, 2019-03-19:
Good point. Added for future release.

nyxnor, 2021-04-18:
pls close this. This is solved with ufw. Allowing through the internet with Tor on https://github.com/rootzoll/raspiblitz/blob/v1.6/home.admin/config.scripts/internet.hiddenservice.sh.

closed 2021-04-19 by rootzoll