raspiblitz/raspiblitz #434: Auto format of HDD when not seeing blockchain.

zer0trst, 2019-03-16:
I downloaded and synced the blockchain on one of my desktops with bitcoin-qt so it would index sync faster but when I went to setup my node and got the blockchain stage it did not detect the blockchain and formatted the drive with no warning or requirement of confirmation?

I'm trying to recover it now as I didn't make a backup :( . If I manage to recover it, have I got the folder structure correct. I had the drive named BLOCKCHAIN and /bitcoin for the blockchain folder as I just pointed bitcoin-qt to save the chain on /BLOCKCHAIN

fluidvoice, 2019-03-16:
How did you setup the drive? (partition, etc.) It must be setup so that this command 
`sudo lsblk -o UUID,NAME,FSTYPE,SIZE,LABEL,MODEL`
shows it as 
LABEL=**BLOCKCHAIN**, NAME=**sda1**, FSTYPE=**Ext4**

zer0trst, 2019-03-16:
Not sure because the drive has been formatted by raspiblitz.. possibly I accidentally formatted it to NTFS or something when I set it up initially to sync blockchain on my desktop. it was a while ago now.

Any recommended file recovery tools for Ubuntu? I'm hoping the blitz hasn't written over the data and corrupted it.. there is a chance it has because it formatted the drive and added ssh folder + raspiblitz.conf file.

Maybe for future release it would be good to have a notification saying blockchain not found and need a Y/N confirmation before it formats the drive in case we put the chain in the wrong folder or set the wrong filesystem on the drive.

fluidvoice, 2019-03-16:
> Not sure because the drive has been formatted by raspiblitz.. possibly I accidentally formatted it to NTFS or something when I set it up initially to sync blockchain on my desktop. it was a while [...]

I agree it should prompt.  @rootzoll see : https://github.com/rootzoll/raspiblitz/blob/e38c48e54536a6233a0062b99858d43685635d80/home.admin/10setupBlitz.sh#L256  It prob should not reformat the drive without prompting the user to confirm.

zer0trst, 2019-03-17:
Ok, Thank you, will download an sync again. can close this one off

closed 2019-03-17 by zer0trst