raspiblitz/raspiblitz #435: Extrem slow blockchain copy via scp

animalmother1988, 2019-03-16:
I am copying the blockchain from my mac book but the speed is extremely slow.
[image]

Any ideas about that? Or is my question to general?

Thx

rootzoll, 2019-03-16:
On my local network I get between 10-20 MB/s on transfere .. where the bottle neck is more likely the USB2.0 bus of the Raspberry. Why it is so slow on your setup its hard to say. Is your Laptop on WLAN? Is there any other process running on the Raspberry that blocks the USB2.0 bus bandwidth?  

BTW the next release v1.1 will replace SCP with rsync for MacOS and Linux: https://github.com/rootzoll/raspiblitz/issues/397

fluidvoice, 2019-03-16:
> I am copying the blockchain from my mac book but the speed is extremely slow.
> 
> Any ideas about that? Or is my question to general?
> 
> Thx

What is your storage media?  Some low quality flash drives (USB/SD) have very slow write speeds esp when copying a large amount of data where they get hot and slow down even more.  WLAN and router are possible problem sources too.

animalmother1988, 2019-03-16:
Thanks for your answers. Yeah normaley i have a speed about 15 MB/s in the LAN. The Laptop is on WLAN. The storage medias are two WD elements. Maybe its because the blockchain is on an external HDD. Maybe i try to copy it via rsync.

nyxnor, 2021-04-18:
please close this.

closed 2021-04-19 by rootzoll